linuxwacom/input-wacom #16: can not use wacom pth660 on Ubuntu buidge 17.04 only connection succeed

243750496, 2017-04-11:
[image]
[image]

with the pen moved there's nothing happened and theres no item show up in wacom settings page
so any solutions?

BTW:xserver-xorg-input-wacom=0.34
[image]

skomra, 2017-04-11:
For support of the second generation Intuos Pro you will eventually depend on kernel 4.11 or later (currently 4.11 is in rc releases). You will be better off installing input-wacom 0.35.0. (xserver-xorg-input-wacom is necessary but unrelated to your current problem.)

You will also need to install libwacom for the control panel (second screen shot) to display the device.

 Instructions for installing the drivers are available at http://linuxwacom.sourceforge.net/wiki/index.php/Downloads .

jigpu, 2018-01-24:
Closing since support for this device is already available in input-wacom.

closed 2018-01-24 by jigpu