linuxwacom/input-wacom #17: Hope can drive Wacom DTI-520

jigpu, 2017-04-12:
_From @SenlinOS on April 12, 2017 1:24_

I have a Wacom DTI-520, the old LCD tablet.
Cannot drive in Ubuntu 16.04.

[   528.461] (II) config/udev: Adding input device WACOM DTI-520-U Pen (/dev/input/event7)
[   528.461] (  ) WACOM DTI-520-U Pen: Applying InputClass "Wacom USB device class"
[   528.461] (  ) WACOM DTI-520-U Pen: Applying InputClass "Wacom class"
[   528.461] (II) Using input driver 'wacom' for 'WACOM DTI-520-U Pen'
[   528.461] (  ) WACOM DTI-520-U Pen: always reports core events
[   528.461] (  ) Option "Device" "/dev/input/event7"
[   528.528] (EE) WACOM DTI-520-U Pen: Invalid type 'stylus' for this device.
[   528.528] (EE) WACOM DTI-520-U Pen: Invalid type 'eraser' for this device.
[   528.528] (EE) WACOM DTI-520-U Pen: Invalid type 'cursor' for this device.
[   528.528] (EE) WACOM DTI-520-U Pen: Invalid type 'touch' for this device.
[   528.528] (EE) WACOM DTI-520-U Pen: Invalid type 'pad' for this device.
[   528.528] (EE) WACOM DTI-520-U Pen: No type specified
[   528.560] (EE) PreInit returned 8 for "WACOM DTI-520-U Pen"
[   528.560] (II) UnloadModule: "wacom"

A device, actually has two ID.
$ lsusb
Bus 001 Device 010: ID 056a:003a Wacom Co., Ltd 
Bus 001 Device 009: ID 056a:003b Wacom Co., Ltd 

Thank you.

_Copied from original issue: linuxwacom/xf86-input-wacom#7_

jigpu, 2017-04-12:
It looks like our kernel driver does not include support for either USB product ID that you've mentioned. I'll be upfront -- with how old the tablet is, its not likely that anyone will find the time to reverse engineer its protocol and add write the necessary protocol decoder...

That said, a quick glance at its manual both suggests that it uses the already-supported "PL" protocol and provides the tablet's hardware resolution. The libwacom library also suggests that the pen is on VID 056a:003a. My best guess about 056a:003b is that it's the device that exposes the buttons?

Please try following the [input-wacom installation instructions](http://linuxwacom.sourceforge.net/wiki/index.php/Input-wacom) from the linuxwacom wiki, only instead of downloading the driver from SourceForge as suggested, use the following command to clone an updated branch from my Github mirror:

    $ git clone https://github.com/jigpu/input-wacom -b add/dti520
    $ cd input-wacom

Afterwards, continue to follow the instructions on how to configure/make/install/load the driver. Let me know if the branch seems to allow your DTI-520 to work.

jigpu, 2017-04-12:
Just fixed a small bug in my work. If you haven't yet done the `git clone`, you'll automatically get the update. Otherwise, run `git pull` inside the input-wacom directory to download the update and then reconfigure/make/install/load the driver.

SenlinOS, 2017-04-13:
Installed.GNOME Wacom settings panel also shows. Cool.
$ xinput -list
Wacom DTI520 Pen stylus                 	id=11	[slave  pointer  (2)]
Wacom DTI520 Pen eraser                 	id=13	[slave  pointer  (2)]

But the pointer only in the bottom of the monitor movement, but also from time to time bounce.

jigpu, 2017-04-13:
Could you describe the problem a bit more? I'm not sure what you mean... Maybe a video of the issue if its difficult to explain?

The lack of a "Wacom DTI520 Pen cursor" is not an issue. A "cursor" device is used to send events from mouse-like tools (see e.g. the image [here](https://www.cdw.com/shop/products/Wacom-Intuos3-6x8-Graphics-Tablet-mouse-digitizer-stylus/770232.aspx)). Tablets with a built-in display like the DTI520 aren't supposed to have a "cursor" device.

SenlinOS, 2017-04-13:
Video download: 
https://drive.google.com/file/d/0B2AXDet3w4pwZGZNRFdveDk2c0U/view?usp=sharing

wacom DTI-520_01.mp4
The cursor is in the lower left corner of the display.
The pen moves to the right and the cursor moves.
The pen always keeps a distance from the cursor.
Pen click invalid.

wacom DTI-520_02.mp4
The cursor is like being sucked by the bottom edge.
Pen subtle click, the cursor will bounce back.
The pen always keeps a distance from the cursor.
Want to click on "Program menu",But the "click" does not work, the cursor can only be a random move.

SenlinOS, 2017-04-14:
Use "xinput-calibrator" to get the value is: Wacom DTI520 Pen eraser

`Warning: multiple calibratable devices found, calibrating last one (Wacom DTI520 Pen eraser)
use --device to select another one.
Calibrating standard Xorg driver "Wacom DTI520 Pen eraser"
current calibration values: min_x=0, max_x=6082 and min_y=0, max_y=4562`

Wacom DTI-520 pen does not exist "eraser".
Should be detected "Wacom DTI520 Pen stylus"?

SenlinOS, 2017-04-14:
http://libregraphicsworld.org/blog/entry/may-update-of-wacom-drivers-for-linux
Previous news,Mentioned DTI-520, hope to find clues.

jigpu, 2017-04-17:
I've been able to track down two linuxwacom discussion threads about the DTI-520 from 2012, [here](https://sourceforge.net/p/linuxwacom/mailman/linuxwacom-devel/thread/CAGzDe_a0D-Bs1jS_GyX-LDVC1LYh%3DNyZn%2B%3DsBJaPnHRcAHgOaQ%40mail.gmail.com/#msg28827754) and [here](https://sourceforge.net/p/linuxwacom/mailman/linuxwacom-devel/thread/4F8520F4.4070409%40shikadi.net/#msg29112593). It looks like the person was able to get DTI-520 support added to the xf86-input-wacom driver and the libwacom database, but [his upstream kernel patch](https://lkml.org/lkml/2012/4/14/20) had a few minor problems that were never resolved.

It should be possible to revive the work on his kernel patch without too much work, if someone can find the time.

SenlinOS, 2017-04-17:
DTI really very special…… Wacom only produced DTI-520! other series have a lot of models!

jigpu, 2019-11-14:
Closing for now since it seems like nobody is interested in taking this on.

closed 2019-11-14 by jigpu