linuxwacom/input-wacom #20: Unknown device_type for 'Wacom Co.,Ltd. CTL-472'.

OXiDE-dev, 2017-11-30:
I just bought a One tablet today and it doesnt work with this driver.
I thought support for this tablet was added by #18.

Here is what `cat /sys/module/wacom*/version` says:
`❯❯ cat /sys/module/wacom*/version 
v2.00-0.37.1`

Here is what journalctl says:
`Nov 30 22:25:56 arch kernel: usb 5-4.3: new full-speed USB device number 10 using xhci_hcd
Nov 30 22:26:02 arch kernel: input input59: Absolute device without dev->absinfo, refusing to register
Nov 30 22:26:02 arch kernel: wacom: probe of 0003:056A:037A.000F failed with error -22
Nov 30 22:26:02 arch kernel: wacom 0003:056A:037A.0010: Unknown device_type for 'Wacom Co.,Ltd. CTL-472'. Ignoring.`

OXiDE-dev, 2017-11-30:
never mind

OXiDE-dev, 2017-12-01:
Ok so aparently it is supported:
[image]
But when i `lsusb` it shows 037a instead of 0300:
[image]

OXiDE-dev, 2017-12-03:
I fixed it 
i just changed 0x300 to 0x37a
[image]

closed 2017-12-03 by OXiDE-dev

jigpu, 2017-12-04:
There's an older version of the "One by Wacom" which the driver supports (which has the 0x300 PID), but support for the current version still doesn't exist upstream. Patches were sent upstream, but there are a few minor problems that are blocking their acceptance: https://lkml.org/lkml/2017/10/11/543

pawelngei, 2019-04-30:
Any updates on this or manuals for less kernel-savvy users for changing 0x300 to 0x37a? Should I do that in this driver and recompile?

@OXiDE-dev any word of advice?

jigpu, 2019-04-30:
@pawelngei The Linux 4.16 kernel (used by Ubuntu 18.10, Fedora 28, etc.) supports the 0x37A device. You can run `uname -r` to see if your distribution has a new enough kernel, or try upgrading to a more up-to-date release of your distribution.

Alternatively, the current release of this driver also supports it. You should be able to follow the [instructions for building input-wacom](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) and have the device work afterwards.

pawelngei, 2019-04-30:
Thank you @jigpu , I owe you a beer