linuxwacom/input-wacom #21: problem installing Wacom One Medium

monami555, 2018-01-02:
Hello

I am pretty stuck with my tablet since several weeks already. I have been following the instructions from this Wiki, and at the end I get:

```
$ cat /sys/module/wacom*/version
v2.00-0.37.1
```

But the tablet pointer won't move, even after reboot. How can I fix it/troubleshoot it?

I have this tablet: https://www.amazon.de/gp/product/B076H8RVXR/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1.

M.

jigpu, 2018-01-03:
This particular tablet model is not supported yet. There's been some work to try and add support (see #18) but we're still waiting on the upstream kernel to merge a patch. Until then, you should be able to get the tablet working with the linked pull request.

jigpu, 2018-02-01:
A version of the linked pull request has been accepted upstream (commit c947218951da68e3fd18a25e9af19556308caf45) and is expected to arrive in Linux 4.16. The code has been imported to input-wacom (commit b12529e589dae810f0b6ef0b22f67b3860f86cde) and backported as far back as 2.6.30 (commit c6fc15ada55d54e53841e08184765b83e7911305) for the 0.39.0 release.

closed 2018-02-01 by jigpu