linuxwacom/input-wacom #23: input-wacom compiling errors

nod0ubt, 2018-03-20:
Hey guys.

I'm installing latest input-wacom drivers on  CentOS 7 machines with Wacom Intuos PTH-860 tablet connected. Got all dependencies, but when compiling I'm getting:

`/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c: In function ‘__wacom_notify_battery’:
/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c:65:7: error: used struct type value where scalar is required
   if (battery->battery.dev)
       ^
/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c: In function ‘wacom_status_irq’:
/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c:1767:7: error: wrong type argument to unary exclamation mark
   if (!wacom->battery.battery.dev &&
       ^
/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c:1773:52: error: invalid operands to binary && (have ‘int’ and ‘struct device’)
  else if ((features->quirks & WACOM_QUIRK_BATTERY) &&
                                                    ^
/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c: In function ‘wacom_mspro_device_irq’:
/root/Downloads/input-wacom-input-wacom-0.37.0/3.7/wacom_wac.c:1793:6: error: wrong type argument to unary exclamation mark
  if (!w->battery.battery.dev &&
      ^`

This happens with all new drivers down to 0.36.0 which compiles and works for me. So I'm forced to use it.

jigpu, 2018-03-21:
What version of CentOS 7 are you using? Run `cat /etc/redhat-release` if you aren't sure.

This looks very similar to an issue we've seen with the RHEL 7.5 beta, which replaced the "power_supply" API that is native to the 3.10 kernel it claims to use with a new version introduced in Linux 4.1. We're still working on a fix. I've attached a work-in-progress patch that might fix things; you can apply it by running `zcat 0001-3.7-compile-under-RHEL-7.5.patch.gz | patch -p1` from inside the input-wacom source directory and recompiling.

[0001-3.7-compile-under-RHEL-7.5.patch.gz](https://github.com/linuxwacom/input-wacom/files/1834101/0001-3.7-compile-under-RHEL-7.5.patch.gz)

nod0ubt, 2018-03-22:
Release is 7.4.1708.
Still haven't tested the patch. Once I do I'll report if its working.
Cheers

jigpu, 2018-04-06:
@raves your build problem seems to be down to the use of a third-party kernel. Our build script has special overrides necessary for RHEL7/CentOS7 that shouldn't be applied in your particular case. Not entirely sure how we'd fix that at the moment...

@nod0ubt are you also running a third-party kernel? Could you provide the output from `./configure` for us to review, along with your `uname -r`?

nod0ubt, 2018-04-16:
Hi guys.

So sorry for taking that long to reply, but I've had other stuff to do as
well.
Just got to test the patch and it seems to be working.

As requrested, I'm providing some info about the system:

*Kernel version (as per uname-r):*4.4.105-1.el7.elrepo.x86_64

*Output from ./configure:*

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers...
/lib/modules/4.4.105-1.el7.elrepo.x86_64/source
checking kernel version... 4.4.105-1.el7.elrepo.x86_64
checking RHEL7 minor release... 4
checking power supply version... v4.1+

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.30/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 3.7 3.17
      kernel source - yes /lib/modules/4.4.105-1.el7.elrepo.x86_64/source

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.

Cheers

raves, 2018-04-17:
Speaking for myself. After applying the 0001-3.7-compile-under-RHEL-7.5.patch.gz patch the build completed ok I believe and the tablet is working just fine so I'm happy with the outcome! I don't think I have a problem that needs solving but thanks for looking more closely at it.

nod0ubt, 2018-05-01:
Hi all.

Just tested out latest release which compiled successfully even without
patch, but the tablet just won't work. System is the same as previous
mentioned.

Pinglinux, 2018-05-02:
nod0ubt, can you issue "make uninstall" under the input-wacom directory then build and install the driver from the latest input-wacom again to see if it helps? Please follow the steps at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source. After a reboot, if the tablet still doesn't work, issue "rmmod hid-wacom; modprobe hid-wacom" then disconnect and connect the tablet to see what happens.

skomra, 2018-05-03:
> input-wacom/wiki/Installing-input-wacom-from-source. After a reboot, if

Hi nod0ubt and Ping,

I just tested this and I needed to issue "rmmod wacom; modprobe hid-wacom"
after installing and after the first reboot. After the second reboot I
didn't need to issue this command.

Aaron

jigpu, 2018-06-21:
Closing this issue since it seems like the updated driver works without the patch.

closed 2018-06-21 by jigpu