linuxwacom/input-wacom #25: wacom_set_report: ran out of retries

AndreaOrru, 2018-04-16:
I'm using the last version of the driver (0.39.0) on a 4.15.15 kernel (Arch Linux) on my Lenovo Yoga 920. I'm getting the following message when I load the module:
```
[ 1042.530504] input: Wacom HID 5110 Pen as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM5110:00/0018:056A:5110.0001/input/input27
[ 1042.530863] input: Wacom HID 5110 Finger as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM5110:00/0018:056A:5110.0001/input/input28
[ 1042.531119] wacom 0018:056A:5110.0001: hidraw1: I2C HID v1.00 Mouse [WCOM5110:00 056A:5110] on i2c-WCOM5110:00
[ 1043.745527] i2c_hid i2c-WCOM5110:00: failed to set a report to device.
[ 1043.745543] wacom 0018:056A:5110.0001: wacom_set_report: ran out of retries (last error = -121)
[ 1043.745704] i2c_hid i2c-WCOM5110:00: failed to set a report to device.
```
However, the touchscreen seems to work just fine. Any idea why this is happening?

jigpu, 2018-04-17:
Looks like a proper fix was never created and sent upstream (just the test patch in your link). We should have something for you to test shortly.

jigpu, 2018-04-24:
Sorry for the delay. I've created a `fix-issue-25` branch on my repository; would you mind cloning and checking that it fixes the issue for you?

`git clone https://github.com/jigpu/input-wacom -b fix-issue-25`

AndreaOrru, 2018-04-28:
Same error :(

```$ grep "" /sys/module/wacom*/version
v2.00-0.38.0.64.g65ebb48```

phileas2, 2018-05-01:
Same exact problem on my Lenovo Yoga 920. Touchpad works fine as well.
Current setup: Solus kernel 4.15.17-65. 
I believe this error is also responsible for a little system freeze during login. 

```
$ grep "" /sys/module/wacom*/version
v2.00

 $ modinfo -n wacom
/lib/modules/4.15.17-65.current/kernel/drivers/hid/wacom.ko

 $ ls -al /lib/modules/4.15.17-65.current/kernel/drivers/hid/wacom.ko
-rw-r--r-- 1 root root 194672 Apr 20 13:52 /lib/modules/4.15.17-65.current/kernel/drivers/hid/wacom.ko

```

jigpu, 2018-05-08:
I haven't been able to come up with any reason the patch shouldn't work, so I've updated my branch to have the same (completely hacky) fix in the linked Yoga 920 thread. Let's see if that at least works, and then work backwards towards something that still reliably sets up the sensor without being as hacky.

@AndreaOrru please run `git checkout fix-issue-25` and `git pull` from inside your input-wacom source directory and then rebuild/reinstall the driver.

@phileas2 Please run `git clone https://github.com/jigpu/input-wacom -b fix-issue-25` to clone a copy of the branch and then rebuild/reinstall the driver.

jigpu, 2018-05-30:
Hmm. Does what you're experiencing sound like bug #28? The "last error" in your logs is -110 which better matches that bug (as opposed to a "last error" of -121 for this bug). These two bugs could very well be related...

@AndreaOrru can you see if your system reports error number -121 while using the fix-issue-25 branch? I wonder if this branch is fixing one error (-121) but then something else is still failing later in the process (-110).

jigpu, 2018-05-30:
Oops, nevermind. I forgot that bug #28 is dealing with a system that has a USB-attached sensor, not an I2C-attached sensor like for these systems. Its less likely that they're the same or related issues...

Pinglinux, 2022-08-05:
I'd say it is a system/hardware, not a tablet/driver, issue. Those integrated systems/laptops require more integration testing, which is beyond the scope of this project.

closed 2022-08-05 by Pinglinux