linuxwacom/input-wacom #27: Wacom tablet not detected in LM 18.2 (Kernel 4.13) but detected in LM17.3 (Kernel 4.4)

cptX, 2018-05-04:
Hi guys in linux Mint 17.3 I successfully managed to recognise my tablet by installing the two files libwacom-common_0.11+dfsg-0ubuntu1_all.deb and libwacom2_0.11+dfsg-0ubuntu1_amd64.deb
Files were downloaded at that time from these links which are now dead
[(http://packages.ubuntu.com/vivid/amd64/libwacom2/download]
[http://packages.ubuntu.com/vivid/all/libwacom-common/download)]

Recently I did a clean install to 18.2 and I had no success making the same tablet working.

I started by running the command sudo dpkg -i libwacom2_0.11+dfsg-0ubuntu1_amd64.deb libwacom-common_0.11+dfsg-0ubuntu1_all.deb with the existing files I had...
which probably was a mistake as these packages were compiled for LM17. I didn't get any error though.
The tablet didn't work. I went to Synaptic and there these two packages were reported as broken. I fixed them, did an update and then they were upgraded to versions 0.22
Still the tablet is not working...

At that point I was confused: either my initial mistake to install the old debian packages is the cause or LM 18 is responsible for not making it work.
I tried to uninstall the packages manually but the dependencies do not allow me because thousand other packages (including cinnamon) will be removed. So I never uninstalled them completely! (HOW CAN I DO IT?? IS IT EVEN POSSIBLE??)

Then I found your page and I followed all the steps described here, compiling all the necessary files from scratch. Still no luck!!!

I ran out of ideas! I still have my old LM17.3 installed in the same computer so it's really annoying knowing that it is there working and not here in 18.2...

My LM17.3 had kernel 4.4 and my LM 18.2 has kernel 4.13.
The tablet has a USB ID: 172f:0047 Waltop International Corp.

Command xsetwacom --list devices returns nothing in LM18.2

Command grep "Using input driver" /var/log/Xorg.0.log returns

```
[     6.475] (II) Using input driver 'libinput' for 'Power Button'
[     6.521] (II) Using input driver 'libinput' for 'Power Button'
[     6.544] (II) Using input driver 'libinput' for 'Sleep Button'
[     6.561] (II) Using input driver 'libinput' for 'Logitech USB-PS/2 Optical Mouse'
[     6.714] (II) Using input driver 'libinput' for 'AT Translated Set 2 keyboard'
[    28.586] (II) Using input driver 'libinput' for 'WALTOP International Corp. Mars Tablet'
[    93.451] (II) Using input driver 'libinput' for 'WALTOP International Corp. Mars Tablet'
[   530.555] (II) Using input driver 'libinput' for 'WALTOP International Corp. Mars Tablet'
[   742.623] (II) Using input driver 'libinput' for 'WALTOP International Corp. Mars Tablet'
```

Command dmesg|tail returns

```
[  739.799344] usb 1-5: USB disconnect, device number 7
[  742.329912] usb 1-5: new full-speed USB device number 8 using xhci_hcd
[  742.474277] usb 1-5: New USB device found, idVendor=172f, idProduct=0047
[  742.474280] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  742.474281] usb 1-5: Product: Mars Tablet
[  742.474283] usb 1-5: Manufacturer: WALTOP International Corp.
[  742.481369] input: WALTOP International Corp. Mars Tablet as /devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.0/0003:172F:0047.0005/input/input21
[  742.538481] hid-generic 0003:172F:0047.0005: input,hidraw1: USB HID v1.10 Mouse [WALTOP International Corp. Mars Tablet] on usb-0000:00:14.0-5/input0
```

Any help please?

jigpu, 2018-05-04:
Hi! Unfortunately we can't be too much help with your particular issue. Our kernel driver is only used with Wacom tablets (or tablet PCs using Wacom sensors), not tablets from other brands like Waltop.

Looking at your log, it appears that the "hid-generic" kernel driver and the "libinput" Xorg driver are being used to power your tablet. If the pointer doesn't respond to tablet input, the issue will be a bug in one of those two drivers. I would suggest capturing a recording of the kernel events sent by your device to better understand what is going on. Install the "evemu-tools" package and then run `sudo evemu-record > /tmp/waltop.log`. You'll be prompted to select a device: type in the number for your Waltop tablet and press Enter. Next, move the stylus around on the tablet for a few seconds. Finally, press CTRL+C to stop the recording.

Next, file a bug at https://bugs.freedesktop.org against the "xorg" product and select the "Input/libinput" component on the bug form. Provide all the information you have, along with this new log file. The libinput maintainer should be able to let you know if the kernel events look sane or not. From there, the root cause of your issue can be fixed :)

***

As for the libwacom packages you've been installing... Libwacom is only a database of tablet properties which is used by programs like the Cinnamon Control Center to provide you with appropriate options for the particular tablet you have plugged in. Updating or removing it won't have any effect on the device actually working and moving the pointer. The libwacom database *does* contain data for non-Wacom devices, but it doesn't look like it has any info about your particular Waltop tablet. Once you get the issue with your tablet worked out, you're more than welcome to file an issue at https://github.com/linuxwacom/libwacom/issues requesting that your tablet be added to the database.

closed 2018-05-04 by jigpu

cptX, 2018-05-04:
Hi jigpu,
I'm really disappointed to see you closing this issue so soon.

My tablet is Waltop based on walcom chip, and it is still working using a walcom driver in linux 17.3 . So correct me if I'm wrong, you are making a driver for walcom tablets, and I have a walcom based tablet, which in 17.3 is working. So why shouldn't your driver in LM18.2 (Ubuntu 16.04) be related to my issue?

My only question actually sums up to why is your driver working in LM17.3 (Kernel 4.4) and not LM18.2 (kernel 4.13).

To understand better, who was responsible for producing the files
libwacom2_0.11+dfsg-0ubuntu1_amd64.deb
libwacom-common_0.11+dfsg-0ubuntu1_all.deb
for ubuntu 14.04 which successfully worked for my tablet in LM17.3 (Kernel 4.4)?

Are these files using your driver or not?

jigpu, 2018-05-04:
As I explained, your tablet does not use our driver. No Waltop tablet uses our driver. Your tablet uses the hid-generic driver that is developed on the LKML and the libinput driver that is developed at freedesktop.org. We do develop the libwacom library but that is not used by either of those two drivers, and has never included a definition of your tablet.

Sorry, but there really isn't anything we can do but point you to the correct developers.

cptX, 2018-05-11:
OK solution found as you can see here:
https://forums.linuxmint.com/viewtopic.php?f=49&t=268898&p=1468224#p1468224

The /usr/share/X11/xorg.conf.d/50-wacom.conf is the only correct file that should exist in that folder and includes a section for Waltop drivers, so I insist that the Waltop Tablets are walcom powered tablets...

jigpu, 2018-05-11:
Good to hear. Reading your linked thread, I think I now understand now the source of the confusion. **Both** the xf86-input-wacom and libinput drivers can be used to decode data from many different tablet manufacturers. Your logs indicated that your system was using libinput, and I thought that was intentional. Any issue with libinput would need to be filed with their bug tracker. It seems, however, that this is instead a configuration issue, with the Mint 18.2 changing the default from "use xf86-input-wacom" to "use libinput".