linuxwacom/input-wacom #29: wacom tablet unresponsive after xhci controller crash

roblem, 2018-05-19:
Maybe this should be filed against the usb subsystem in the linux kernel, but this only happens when using the wacom tablet so I thought I'd file here.  Lenovo X1 Carbon Yoga 3rd generation experiencing loss of wacom tablet functionality when optimal power management settings are used. Using stock wacom driver in Fedora (v. 2.00)

What happens:
1. Using powertop configure system for optimal power management (all settings are set to "Good" under powertop tunables, including wacom usb entries)
2. Reboot and notice that things look good in dmesg:
   ```
   [    0.000000] Linux version 4.16.7-300.fc28.x86_64 (mockbuild@bkernel02.phx2.fedoraproject.org) (gcc version 8.0.1 20180324 (Red Hat 8.0.1-0.20) (GCC)) #1 SMP Wed May 2 20:09:13 UTC 2018
   [    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.16.7-300.fc28.x86_64 root=/dev/mapper/fedora-root ro rd.lvm.lv=fedora/root rd.lvm.lv=fedora/swap rhgb quiet resume=/dev/mapper/fedora-swap acpi.ec_no_wakeup=1 psmouse.synaptics_intertouch=1 
   ```
   and the relevant parts for wacom:
   ```
   [    2.555509] usb 1-10: New USB device found, idVendor=056a, idProduct=5146
   [    2.555511] usb 1-10: New USB device strings: Mfr=1, Product=2, SerialNumber=0
   [    2.555513] usb 1-10: Product: Pen and multitouch sensor
   [    2.555514] usb 1-10: Manufacturer: Wacom Co.,Ltd.
   [    2.558357] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:056A:5146.0001/input/input7
   [    2.558719] hid-generic 0003:056A:5146.0001: input,hiddev96,hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input0
   [    2.560531] input: Wacom Co.,Ltd. Pen and multitouch sensor as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:056A:5146.0002/input/input8
   [    2.560640] hid-generic 0003:056A:5146.0002: input,hiddev97,hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input1
   [    2.607775] input: Wacom Pen and multitouch sensor Finger as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.0/0003:056A:5146.0001/input/input10
   [    2.607959] wacom 0003:056A:5146.0001: hidraw0: USB HID v1.11 Device [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input0
   [    2.608304] input: Wacom Pen and multitouch sensor Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10:1.1/0003:056A:5146.0002/input/input12
   [    2.608710] wacom 0003:056A:5146.0002: hidraw1: USB HID v1.11 Mouse [Wacom Co.,Ltd. Pen and multitouch sensor] on usb-0000:00:14.0-10/input1   
   ```
3. After approximately 5 minutes of using the tablet, it becomes unresponsive and the system becomes noticeably hotter with power use spiking to 16W.  This is what I see in dmesg:
   ```
   [   51.010836] xhci_hcd 0000:3c:00.0: remove, state 4
   [   51.010845] usb usb4: USB disconnect, device number 1
   [   51.011269] xhci_hcd 0000:3c:00.0: USB bus 4 deregistered
   [   51.011363] xhci_hcd 0000:3c:00.0: xHCI host controller not responding, assume dead
   [   51.011368] xhci_hcd 0000:3c:00.0: remove, state 4
   [   51.011375] usb usb3: USB disconnect, device number 1
   [   51.011589] xhci_hcd 0000:3c:00.0: Host halt failed, -19
   [   51.011592] xhci_hcd 0000:3c:00.0: Host not accessible, reset failed.
   [   51.011821] xhci_hcd 0000:3c:00.0: USB bus 3 deregistered
   [   51.719358] thunderbolt 0000:08:00.0: stopping RX ring 0
   [   51.719364] thunderbolt 0000:08:00.0: disabling interrupt at register 0x38200 bit 12 (0xffffffff -> 0xffffefff)
   [   51.719371] thunderbolt 0000:08:00.0: stopping TX ring 0
   [   51.719375] thunderbolt 0000:08:00.0: disabling interrupt at register 0x38200 bit 0 (0xffffffff -> 0xfffffffe)
   [   51.719378] thunderbolt 0000:08:00.0: control channel stopped
   [   51.719459] thunderbolt 0000:08:00.0: freeing RX ring 0
   [   51.719466] thunderbolt 0000:08:00.0: freeing TX ring 0
   [   51.719474] thunderbolt 0000:08:00.0: shutdown
   [   51.743396] pci_bus 0000:07: Allocating resources
   [   51.743429] pcieport 0000:07:01.0: bridge window [io  0x1000-0x0fff] to [bus 09-3b] add_size 1000
   [   51.743439] pcieport 0000:07:02.0: bridge window [io  0x1000-0x0fff] to [bus 3c] add_size 1000
   [   51.743441] pcieport 0000:07:02.0: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 3c] add_size 200000 add_align 100000
   [   51.743450] pcieport 0000:07:04.0: bridge window [io  0x1000-0x0fff] to [bus 3d-70] add_size 1000
   [   51.743459] pcieport 0000:06:00.0: bridge window [io  0x1000-0x0fff] to [bus 07-70] add_size 3000
   [   51.743463] pcieport 0000:06:00.0: BAR 13: no space for [io  size 0x3000]
   [   51.743464] pcieport 0000:06:00.0: BAR 13: failed to assign [io  size 0x3000]
   [   51.743466] pcieport 0000:06:00.0: BAR 13: no space for [io  size 0x3000]
   [   51.743467] pcieport 0000:06:00.0: BAR 13: failed to assign [io  size 0x3000]
   [   51.743473] pcieport 0000:07:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
   [   51.743474] pcieport 0000:07:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
   [   51.743475] pcieport 0000:07:01.0: BAR 13: no space for [io  size 0x1000]
   [   51.743476] pcieport 0000:07:01.0: BAR 13: failed to assign [io  size 0x1000]
   [   51.743477] pcieport 0000:07:02.0: BAR 13: no space for [io  size 0x1000]
   [   51.743479] pcieport 0000:07:02.0: BAR 13: failed to assign [io  size 0x1000]
   [   51.743480] pcieport 0000:07:04.0: BAR 13: no space for [io  size 0x1000]
   [   51.743482] pcieport 0000:07:04.0: BAR 13: failed to assign [io  size 0x1000]
   [   51.743484] pcieport 0000:07:04.0: BAR 13: no space for [io  size 0x1000]
   [   51.743485] pcieport 0000:07:04.0: BAR 13: failed to assign [io  size 0x1000]
   [   51.743487] pcieport 0000:07:02.0: BAR 15: no space for [mem size 0x00200000 64bit pref]
   [   51.743488] pcieport 0000:07:02.0: BAR 15: failed to assign [mem size 0x00200000 64bit pref]
   [   51.743490] pcieport 0000:07:02.0: BAR 13: no space for [io  size 0x1000]
   [   51.743491] pcieport 0000:07:02.0: BAR 13: failed to assign [io  size 0x1000]
   [   51.743492] pcieport 0000:07:01.0: BAR 13: no space for [io  size 0x1000]
   [   51.743493] pcieport 0000:07:01.0: BAR 13: failed to assign [io  size 0x1000]
   [   56.038973] pcieport 0000:07:00.0: Refused to change power state, currently in D3
   [   56.040866] pci_bus 0000:08: busn_res: [bus 08] is released
   [   56.040931] pci_bus 0000:09: busn_res: [bus 09-3b] is released
   [   56.040977] pci_bus 0000:3c: busn_res: [bus 3c] is released
   [   56.041076] pci_bus 0000:3d: busn_res: [bus 3d-70] is released
   [   56.041144] pci_bus 0000:07: busn_res: [bus 07-70] is released
   [  633.322847] xhci_hcd 0000:00:14.0: xHC is not running.
   [  701.099735] xhci_hcd 0000:00:14.0: xHC is not running.
   [  722.939901] xhci_hcd 0000:00:14.0: xHC is not running.
   [  732.892947] xhci_hcd 0000:00:14.0: xHC is not running.
   [  819.554547] xhci_hcd 0000:00:14.0: xHC is not running.
   [  863.556178] xhci_hcd 0000:00:14.0: xHC is not running.
   [  901.909278] xhci_hcd 0000:00:14.0: xHC is not running.
   [  944.272820] xhci_hcd 0000:00:14.0: xHC is not running.
   [ 1032.719054] xhci_hcd 0000:00:14.0: xHC is not running.
   [ 1045.293000] xhci_hcd 0000:00:14.0: xHC is not running.
   [ 1069.748934] xhci_hcd 0000:00:14.0: xHC is not running.
   [ 1077.114486] xhci_hcd 0000:00:14.0: xHC is not running.
   [ 1088.723426] xhci_hcd 0000:00:14.0: xHC is not running.
   [ 1088.726020] xhci_hcd 0000:00:14.0: xHCI host controller not responding, assume dead
   [ 1088.726036] xhci_hcd 0000:00:14.0: HC died; cleaning up
   [ 1088.801794] usb 1-5: USB disconnect, device number 2
   [ 1088.803044] usb 1-8: USB disconnect, device number 4
   [   1088.803668] usb 1-10: USB disconnect, device number 5 
   ```
5. The other usb devices and the xhci appears to come back online but not the tablet. Unloading/reloading wacom kernel module fails to bring the tablet back.  Have to restart at this point to recover functionality.  

I can avoid the problem 2 ways:
1. Use computer in notebook mode and avoid the touchscreen, or
2. Turn off power management on the xHCI controller:
  ```
  # Runtime PM for PCI Device Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller 
  /sys/bus/pci/devices/0000:00:14.0/power/control=on
  ```
   with the downside to this second approach being that system power usage goes up quite a bit (about 1Whr), which is a concern because the machine is already not the most battery efficient.

jigpu, 2018-05-21:
Have you tried only disabling power management on the Wacom device (i.e. `/sys/module/wacom/drivers/hid:wacom/0003:056A:5146.*/power/control=on`?) With the controller dying like that its hard to say why our hardware isn't coming back up. We could potentially provide you with a version of input-wacom which more thoroughly debugs the `probe()` process to see if the kernel tries to reset the device afterwards and if so where exactly the driver has to fail.

roblem, 2018-05-21:
Just tried it with power control set to on for wacom, and the xhci_hcd set to auto.   Crashes don't happen with this configuration either but it is approximately 3 W/hr less efficient which is like a 50% increase in power usage.  I didn't test this config for too long because it was so energy inefficient, but it was crash-free.

If you think it is worth debugging the driver I could help out.

roblem, 2018-06-01:
I am also seeing "usb: port power management may be unreliable" in dmesg.  I found this [description of the issue](https://unix.stackexchange.com/questions/323692/what-does-usb-port-power-management-may-be-unreliable-actually-mean) which happens because the "kernel was unable to determine a port's peer".  Seems like it could be related.

jigpu, 2018-06-08:
Sorry about the delay in responding... Its interesting that both you and https://github.com/linuxwacom/input-wacom/issues/28 appear to be having issues with USB autosuspend. I'm not sure if something recently changed in the kernel, or if these particular devices have some kind of firmware bug.

I've added additional logging to our `probe` function that is called by the kernel when setting up a tablet. It should let us know if the kernel is even attempting to have us reconnect to the tablet after a crash, and if so roughly where its failing. Please clone a copy of my branch by running `git clone https://github.com/jigpu/input-wacom.git -b issue-29` and then compiling/installing it like normal. Hopefully the dmesg log will contain a bit more information after the crash that will help us out.

roblem, 2018-06-09:
Using this driver with all powertop settings to "good" and even though dmesg reports xHCI controller crashes:
```
[Sat Jun  9 12:03:33 2018] xhci_hcd 0000:3c:00.0: xHCI host controller not responding, assume dead
[Sat Jun  9 12:03:33 2018] xhci_hcd 0000:3c:00.0: remove, state 4
[Sat Jun  9 12:03:33 2018] usb usb3: USB disconnect, device number 1
```
The touchscreen keeps working unlike before.  Note: I am on kernel 4.16.14-300.fc28.x86_64 (different from above).

Fedora by default is using wacom driver 2.00.36.1.  I haven't tried using the stock driver with Fedora's more recent kernels, so it is possible this problem has been fixed by the kernel updates as opposed to the more recent wacom driver from your branch, will investigate further.

One thing- I can't permanently use this (or any more recent wacom driver that I compile myself) even after following instructions [image].  Using `dracut --force` completes without error, but Fedora persists with the stock wacom kernel. I have to `rmmod` and `modprobe` (as described in troubleshooting) to load it.

roblem, 2018-06-11:
I've used your github branch as well as the current stock driver in Fedora all weekend, and everything is working now even with all power saving features enabled. I am also not experiencing the lags described in #28.  Closing.  Happy to keep monitoring the logs if it helps in any way.  Let me know.

closed 2018-06-11 by roblem