linuxwacom/input-wacom #30: Wacom one medium no more recognized

chdioux, 2018-06-07:
Hello,
(Moved from libwacom where i posted it first it mistake)
My Wacom one medium tablet was not recognized on both of my computers under Ubuntu 16.04.
Although I'm not a developer and very afraid of doing mistakes, I tried to fix that by building the drivers from the sources you are providing because i can't wait any longer a built-in version in Ubuntu. Your explanations were very helpful for a non professional like me, thank you so much for that.
Unfortunately, on both computers, my tablet is still not recognized. BUT more than that, on my old Packard Bell EasyNote, it actually DID work for a while yesterday at night, just after the upgrading of the drivers, but it is not working anymore this morning. I've done nothing in the meantime, except to shutdown the computer.
I'm attaching the logs you asked for on another similar topic.
Regards
[sysinfo.LKtj9xwr3T.tar.gz](https://github.com/linuxwacom/input-wacom/files/2079513/sysinfo.LKtj9xwr3T.tar.gz)

skomra, 2018-06-07:
Hi chdioux,

Could you clone and test the following branch to see if this patch resolves the issue for you:

$ git clone -b wacom_one_3_17 https://github.com/skomra/input-wacom.git
$ cd input-wacom
$ ./autogen.sh
$ make
$ sudo make install

then reboot

chdioux, 2018-06-07:
Hi Aaron!
Great job! Everything works perfectly now. Thank you so much!
Have a nice day.

skomra, 2018-06-07:
Hi chdioux,

Glad that worked for you, and thanks for confirming. I'm actually having trouble reproducing this issue now, so I think this issue might be more complicated on our end. 

Also, we generally leave the issues open until the fixes are in a release, so I'm going to re-open this.

Best,
Aaron

chdioux, 2018-06-08:
Hi Aaron,

If it can help, i had a similar problem on my other computer (different computer but very similar configuration, as i use both mostly the same way).

So yesterday, i tried to update input-wacom the same way on the second computer, with a terrible result first:
After a reboot, my tablet was correctly working, but mouse and keyboard did'nt work any more!

Then, using a virtual keyboard, the tablet, and a Teamviewer connection, i tried first to "apt remove" some wacom packets with no result. Then i had the idea of installing xfce4 and it worked!
Not only my mouse and keyboard were working again but at my great surprise, my tablet was working too! Everything is working perfectly now, keyboard, mouse and tablet, both with Unity and xfce.

My humble ideas:
1) Remember i'm a grassroots user. I've been using Linux yor 20 years now, so i have some basis, but only as a simple end user. So I probably made some beginner's mistakes when building the drivers. And most probably my environment when building the drivers contains lacks or errors.
2) I remember i had several warnings during the building about a "m4" thing, possibly around a line 160 in a "main" file. But all this by memory only. Anyway, the scripts ended normally.
3) Please find here the sysinfo file when it was all messed up:
[sysinfo.huuHXAEUGr.tar.gz](https://github.com/linuxwacom/input-wacom/files/2083482/sysinfo.huuHXAEUGr.tar.gz)
4) And the new one now that everything is working fine, after wfce4 installing:
[sysinfo.ZaBQEku9i4.tar.gz](https://github.com/linuxwacom/input-wacom/files/2083501/sysinfo.ZaBQEku9i4.tar.gz)

Hope this helps and thanks again to you and everybody in the linuxwacom team.

jigpu, 2018-06-08:
Hi chdioux,

I'm trying to figure out exactly what version of the code the "standard input-wacom-master" you built is. Aaron has applied the fix to his copy of "master", but the fix is not yet applied to the project's official version. Your sysinfo logs would normally tell us the exact version of the driver, but for some reason its not listed in the working sysinfo.ZaBQEku9i4.tar.gz...

Do you still have a copy of the "standard input-wacom-master" code that you compiled? If you do, please open a terminal, navigate to that directory, run the following commands, and reply back with what the commands output.
```
$ cat version
$ grep -i -A2 0x37A 4.5/wacom_wac.c
```

chdioux, 2018-06-08:
Hi jigpu

I was not that much at ease with git, so i used the zipfile i found here:
https://github.com/linuxwacom/input-wacom
and which is:
https://github.com/linuxwacom/input-wacom/archive/master.zip

Actualy, i first compiled (and installed) the version skomra proposed yesterday, and which has been perfect on my laptop, but as my desktop computer froze with it, i tried to go back to a previous version of input-wacom by trying to install this older version (https://github.com/linuxwacom/input-wacom/archive/master.zip)

If i navigate to that directory (once unziped), i can't find any "version" file to cat.

For your second question, i get this:
```
christophe@GIGABYTE:~/Bureau/input-wacom-master$ grep -i -A2 0x37A 4.5/wacom_wac.c

static const struct wacom_features wacom_features_0x37A =
	{ "Wacom One by Wacom S", 15200, 9500, 2047, 63,
	  BAMBOO_PEN, WACOM_INTUOS_RES, WACOM_INTUOS_RES };
--
	{ USB_DEVICE_WACOM(0x37A) },
	{ USB_DEVICE_WACOM(0x37B) },
	{ USB_DEVICE_WACOM(0x4001) },
```
Hope this helps.

jigpu, 2018-06-08:
Thanks, that's exactly what I needed to know. Does the "linuxwacom" version work on both of your computers? I wonder if there was just some kind of configuration issue that was making it not work at first...

chdioux, 2018-06-08:
Well i'm not very sure what to answer.

On my laptop, i was in the sysinfo.LKtj9xwr3T.tar.gz state. I installed Skomra's version (https://github.com/skomra/input-wacom.git) and everything worked fine then. So i think it is still the version which is working there.

On my desktop computer, i tried to do the same things:
- Installing Skomra's version -> Tablet ok, but keyboard and mouse out of order ! (sysinfo.huuHXAEUGr.tar.gz)
- Tried to "get remove" different wacom packets -> reboot -> no difference
- Tried to install archive/master.zip -> reboot -> no difference. Worst, actually even Unity went wrong (i lost the dock and menu bar). This gave me the idea of installing xfce4 as a last chance.
-> reboot -> everything went well then! (sysinfo.ZaBQEku9i4.tar.gz)
So i imagine that on the desktop, the Skomra's version fixed the tablet problem but created some other configuration issue which has been fixed then by returning back to archive/master.zip PLUS installing xfce4.
But you know, all that is so much out of my depth...

jigpu, 2018-06-08:
Thanks so much for your detailed replies :) Would you be willing to try the archive/master.zip on your laptop to see if it works there as well?

chdioux, 2018-06-08:
Well....
I would be glad to help, because you guys helped me a lot. But now that my laptop is working perfectly fine... Well, OK! I'll do that immediately. See you! :-)

chdioux, 2018-06-08:
OK. It was a little scary, because you know... "if it ain't broke...", but everything went allright.
I took a sysinfo first, which is:
[sysinfo.xQVzkgnv5M.tar.gz](https://github.com/linuxwacom/input-wacom/files/2085835/sysinfo.xQVzkgnv5M.tar.gz)

then i built input wacom from the zip archive. Here is a log, with the same errors i had on the other computer: line 21 (m4) and 110 (SSL)

[build_log.txt](https://github.com/linuxwacom/input-wacom/files/2085856/build_log.txt)

But beside that, everything went allright. After a reboot, keyboard, mouse and tablet are still working.
Sysinfo nows is:
[sysinfo.QgUT6T1usq.tar.gz](https://github.com/linuxwacom/input-wacom/files/2085851/sysinfo.QgUT6T1usq.tar.gz)

jigpu, 2018-06-08:
*phew!* Glad to hear that nothing broke after installing :) The warning is innocuous (we should see if we can get rid of it to avoid scaring people...) and the error only matters on systems which are using secure boot (it's complaining that it can't sign the modules with the secure boot key).

Since the unmodified driver seems to be working, I'm going to tenatively guess that it was just some kind of temporary configuration issue. Do let us know if anything stops working though...

chdioux, 2018-06-08:
:-D Thanks again for your help. Have a good day.

closed 2018-06-19 by jigpu