linuxwacom/input-wacom #33: Linux delay when 2 finger scroll or zoom

damarusama, 2018-06-13:
I'm experiencing a lag when I use a double finger touch scroll or zoom. 
I am using 0.36.1-1 on Arch Linux. The lag happens on scroll and on zoom. I am using Wacom Intuos PT S Finger touch. It's exactly as described in this previous bug:

https://sourceforge.net/p/linuxwacom/bugs/218/

There is an early delay then it 'kicks in' (scroll or zoom) so you can then continue to scroll or zoom while you still have your 2 fingers on the tablet, but the behavior is reset once you release your fingers.

jigpu, 2018-06-15:
Thanks for re-filing that over here. It looks like the scroll delay is caused by the driver trying to determine if you're performing a two-finger scroll or a pinch-zoom gesture. You have to move your fingers a minimum distance so the driver can tell if your fingers are moving vertically, horizontally, together, or apart. The distance at which the driver will classify a scroll is apparently just over four times the normal scroll distance. Zoom gestures on the other hand require movement of just under _six_ times the normal zoom distance.

I'll have to experiment with a few different values to see if there's a better number that has a less noticible delay while still working well on older devices with low-resolution touch sensors.

damarusama, 2018-06-15:
There is no way to have that as a setting so it could be modified for every tablet?

jigpu, 2018-06-15:
I'd like to focus on fixing the driver defaults since that's what everyone will use. There is an argument to be made for adding a setting (especially since it the driver lets you control other gesture distances) but since almost nobody will be aware of the option's exisistence the more important thing is to improve the default feel.

jigpu, 2018-06-18:
The scroll recognition code is a little more complex than I initially realized, and I now see that there's actually also a ~250 millisecond delay which is being imposed on scrolling recognition. I've uploaded a version to my github that should be a bit more responsive. Please try it out and let me know what you think. You can get a copy of the branch by running the following command, and then build/install using the normal instructions.

```
$ git clone https://github.com/jigpu/input-wacom.git -b issue-33
```

damarusama, 2018-06-18:
Definitely a good improvement. For me with the previous delay, my eyes (and brain) would start scrolling the page when I would scroll with my finger which made the 2 finger scrolling unworkable. With this smaller delay, it's a lot better. Maybe a tad faster, but at this point, it's already usable.

damarusama, 2018-06-18:
Well, I am not sure if I'm really testing the right thing here. 

I've removed xf86-input-wacom from my system, then downloaded your source code and installed it from source:

if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"

I rebooted and the delay is gone. But I don't have access to 2 finger zoom. So I checked the xsetwacom --list devices and nothing comes up. 

lsusb gives me: 

Bus 001 Device 002: ID 056a:0302 Wacom Co., Ltd CTH-480 [Intuos Pen & Touch (S)]

grep "" /sys/module/wacom*/version
v2.00

I didn't uninstall libwacom as it would send me through a dependency of libinput with then most of the system is dependent on. 

Anything else to check?

I'm on Archlinux.

jigpu, 2018-06-18:
Are you sure you're running an Xorg session? Its not likely to have switched to Wayland between reboots, but that's pretty often the case when the tablet is working but not showing up in xsetwacom. You can also run `xsetwacom -V` to get the installed driver version number.

damarusama, 2018-06-18:
I boot in console and startx from the command line so I'm pretty sure it's Xorg. 

> xsetwacom -V
0.35.99.1.12.g25d0e8a

jigpu, 2018-06-18:
Definitely Xorg then, and the reported version number is correct as well. Could you upload a copy of your Xorg log for review? It may have some hints about what is going on.

damarusama, 2018-06-18:
This is the part of the log that pertains to waccom. 

http://ponnuki.me/XorgLog.txt

jigpu, 2018-06-19:
Thanks. Looks like Xorg isn't able to find the driver so its falling back to using xf86-input-libinput instead. Although the xsetwacom binary seems to have been installed in the right location, the driver itself may not have. Please run the following commands:
```
$ cd <path to input-wacom source>
$ sudo make uninstall
$ make clean
$ ./configure --prefix=/usr --libdir=/usr/lib
$ make
$ sudo make install
```
Next check to make sure that the file `/usr/lib/xorg/modules/input/wacom_drv.so` exists. If it does, try restarting Xorg. It should find and load the driver without issue.

damarusama, 2018-06-19:
Ok that work, it gets listed. 
The delay is a bit slower than the xf86-input-libinput and it doesn't have circular motion with 2 fingers, but I've regained the zoom and xsetwacom list the device.
 
The delay still feels unnatural although it's almost there, can it be reduced even more?

jigpu, 2018-06-19:
I'll see what I can do :)

damarusama, 2018-06-19:
Thanks ;) Is the circular motion with 2 finger something possible or that need to be enabled?

jigpu, 2018-06-25:
Sorry about the delay... Working with the gesture code has lead me down a bit of a rabbit hole that I'm only starting to surface from. Not too much progress I'm afraid, but in case you're interested:

The remaining delay appears to be the result of the code introduced in commit 8f85692a19bd34dae77071a296a14c01a491cf4a. That commit initializes the root coordinates of the gesture to the location where it was recognized rather than the location it was actually started. The reason for this is that using the starting location will result in a sudden batch of catch-up events being sent the moment the gesture is recognized. This can cause a large jump in the scrolled position or zoom level. If we could reliably distinguish gestures in the same amount of distance as a single event, we could get rid of that code, but after playing around a bit I haven't had too much luck striking the right balance.

I also found that the default scroll/zoom distances for some tablets was wrong, which ended up causing some tablets to be much less usable out of the box than they should have been. There's a bug both in the kernel and our X driver, so I'm working on those fixes as well.

> Thanks ;) Is the circular motion with 2 finger something possible or that need to be enabled?

Unfortunately, the circular scroll isn't a gesture that our driver recognizes.

moaxey, 2018-11-06:
I'm not sure that 'enhancement' is the right label for this issue.

As a user, the 250ms delay makes the touch features pretty much unusable. As someone who uses the touch features to reduce strain, the delay feels like a physical blockage, causing muscles to tense up etc.

I got the issue-33 branch installed, and it mitigates the issue to be usable, but there still is a perceptible lag.