linuxwacom/input-wacom #34: Wacom Cintiq 24HD Pen stopped working

mnlipp, 2018-06-19:
My Wacom Cintiq 24HD Pen stopped working when I upgraded the kernel to 4.16.x. Booting fc26 (X Server 1.19.6) with kernel 4.15.17 (from fc25) makes it work again.

"Working" in this case means that the cursor does no longer follow the pen movements. Actually, nothing that I do with the pen has any effect. The X server log files don't show any problems. The device is detected and registered.

Any hints?

jigpu, 2018-06-19:
I've just tried using a Cintiq 24HD with a fully-updated Fedora 27 install (4.16.15-200 kernel) and don't see any problems. I've occasionally had issues in the past with the Cintiq randomly stopping until I turn it off and then on again; perhaps you're experiencing something similar?

mnlipp, 2018-10-08:
After using the old kernel for a long time, I finally upgraded to Fedora 28 today. The situation is the same as initially described. `xinput` shows the wacom "pen" device, `xsetwacom` lists it, but it doesn't work.

It *does* start working after unpluging and re-plugging the USB connection to the tablet. As the plug isn't easily reachable in my case ("built-in" device), I've come up with this reset script:

```
#!/bin/bash

for d in /sys/bus/usb/devices/*; do 
  if [ -r "$d/idVendor" ]; then
    idVendor=`cat $d/idVendor` 
    idProduct=`cat $d/idProduct` 
    if [ "$idVendor" = "056a" -a "$idProduct" = "00fa" ]; then
      echo 0 > "$d/authorized"
      echo 1 > "$d/authorized"
    fi
  fi
done
```
Executing this on boot fixes things, but this is, of course, only a hack, not a solution.

jigpu, 2018-10-08:
What kernel were you using previously that worked for you? That might provide a hint of where to start looking. Also, does the tablet always require the reset after boot-up? Do you ever have to reset after the machine is up and running? Does turning the Cintiq off and then on also fix the issue, or does only unplugging (or resetting) the USB work?

mnlipp, 2018-10-08:
Kernel 4.15.17 was the last that worked (see above). Yes, it always requires the reset. 
> Do you ever have to reset after the machine is up and running?

I've only run the upgraded machine for a short time today (about 2 hours) during which it was stable. I'll have to supply this information later. 
> Does turning the Cintiq off and then on also fix the issue, or does only unplugging (or resetting) the USB work?

Turning off (unplugging the power) fixes the problem. In fact, this is what I found out first. It was this  observation that made me try to unplug the USB connection as a more gentle approach (and eventually the software reset as the mildest "solution"). I'll have to check if switching to standby and reactivating helps, I didn't try that.

mnlipp, 2018-10-19:
> Does turning the Cintiq off and then on also fix the issue, or does only unplugging (or resetting) the USB work?

Finally remembered to check, yes, it does.

jigpu, 2018-10-24:
Looks like there's no USB traffic being sent by the 24HD at all after the issue is triggered. Still not sure of the actual trigger of if its a driver or firmware bug...

mnlipp, 2018-10-25:
As the behavior is related to booting a newer kernel, it should be the driver in that newer kernel, shouldn't it? Or does the driver include some vendor blob that has been updated somewhere between kernel 4.15.17 and 4.16.x?

jigpu, 2018-10-25:
That would make sense, though I'm somewhat skeptical that a kernel change is the actual cause of the problem (as opposed to merely triggering an existing problem). My 24HD has been acting up for a long time. I can roll back my own kernel to double-check, but I suspect it was having issues even before 4.15 was released. Also, none of the four changes to our driver in that period (kernel commits torvalds/linux@99acedadde, torvalds/linux@8341720642, torvalds/linux@791ae27373, torvalds/linux@403c0f681c, and torvalds/linux@c947218951) would plausibly impact the 24HD. A change somewhere else in the kernel (e.g. the underlying USB subsystem) could be responsible, but isolating the offending commit is going to require quite a bit of compiling and testing ("git bisect" estimates 13 compile/test rounds).

Pshemas, 2019-02-19:
also suffer from this issue. On my end it is definitely triggered by kernel change. When I boot OpenSUSE LEAP 15 with stock kernel (4.12) all works as expected. As soon as I boot with 4.20 immediately I get this problem (and as with others unplugging and plugging the device again makes the device start working).

Pinglinux, 2019-02-20:
Looks like something changed after kernel 4.15 which affected the tablet. Can one of you bisects the code to figure out which patch is the root cause?

Thank you for reporting the issue!

Pinglinux, 2022-08-05:
Do they work with newer distributions now?

mnlipp, 2022-08-05:
I currently don't have access to the rooms with the Cintiq. Will be a while until I can provide the information.

Pinglinux, 2022-08-12:
The root cause of those Cintiq related issues are not in Wacom driver, especially since the issue disappear after a power/USB cycle. It should be in the kernel/system configuration for kernels later than 4.15.17+. Feel free to open a new request if systems running kernel 5.18+ still have the issue.

closed 2022-08-12 by Pinglinux