linuxwacom/input-wacom #35: Build Failed

bmw2621, 2018-07-04:
I cannot figure out why the build is failing,  I have the correct kernel and headers.  I cannot find a resource that says what config file the build is looking for or how to get it.  Please help.

ben@archlabs ~/Downloads/input-wacom-0.41.0 %  if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... not found
configure: WARNING: Unable to find build config in any of:
	  [/lib/modules/4.14.52-1-lts/source],
	  [/lib/modules/4.14.52-1-lts/build],
	  [/usr/src/linux],
	  [/usr/src/linux-4.14.52-1-lts],
	  [/usr/src/linux-2.6]
	

configure: WARNING: Kernel directory does not appear to have needed config files
checking kernel version... 
configure: WARNING: Spcified '4.14.52-1-lts' kernel, but found '' instead
configure: error: We could not find the development environment to build modules for the '4.14.52-1-lts' kernel within the '' directory. Please install the kernel source or the kernel development package and try again.
Build Failed
ben@archlabs ~/Downloads/input-wacom-0.41.0 % uname -or
4.14.52-1-lts GNU/Linux
ben@archlabs ~/Downloads/input-wacom-0.41.0 %

jigpu, 2018-07-05:
One common reason for this happening is that a kernel update has been installed but you haven't yet restarted your system. The configure script will be looking for the kernel headers associated with the currently-running kernel (4.14.52-1-lts), but if an upgrade replaces the kernel headers with a new version then the script will fail. In this case, the easiest solution is to simply reboot and then retry. Let us know if that does not work.

bmw2621, 2018-07-07:
That did it. Youre a saint!

closed 2018-07-07 by bmw2621