linuxwacom/input-wacom #36: Wacom PTH-660 not starting on boot Centos 7.4

spikyness, 2018-07-25:
I'm seeing a strange issue with the newer tablets not working when plugged into a machine at boot. unplugging and plugging the tablet in when it's in the state does not start the tablet. Disconnecting the tablet and turning it off, then restarting the machine and connecting it when the machine has fully booted in the os causes it work as intended.

Pinglinux, 2018-07-25:
I assume you updated Wacom driver from input-Wacom, otherwise the device would not even work when system booted. Did you follow the steps at https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source? Which input-Wacom version did you use? Can you update the driver again with the latest version to see what happens? Please post your dmesg here if the latest input-wacom doesn't resolve your issue.

Pinglinux, 2018-07-25:
Sorry, accidently hit "Close" instead of "Comment"

spikyness, 2018-08-24:
Hi ping, 

sorry about the very long reply.

This is what I'm seeing from dmesg relating to wacom when it doesn't work. Unplugging and replugging the device does not work. 

# Wacom Version 

This is the version that we are using

\# modinfo hid_wacom|grep version
version:        v2.00-0.41.0
rhelversion:    7.4
srcversion:     4D5B6BAC88F56E785B959D5
vermagic:       3.10.0-693.21.1.el7.x86_64 SMP mod_unload modversions 

# Fix once machine is up 

I've found doing the following will get the device to start working again once it is in this state.

remove and readd kernel module

  rmmod /lib/modules/$(uname -r)/extra/hid-wacom.ko
  insmod /lib/modules/$(uname -r)/extra/hid-wacom.ko

Determine which usb port the device is attached to

  usbport=$(ls -lah /sys/bus/hid/devices/|grep 0003:056A:0357|head -n 1|cut -d \/ -f 8)

Simulate unplug and replug on port

  echo 0 > /sys/bus/usb/devices/${usbport}/authorized
  echo 1 > /sys/bus/usb/devices/${usbport}/authorized

# Dmesg

[    3.743268] usb 3-4: New USB device found, idVendor=056a, idProduct=0357
[    3.743277] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.743283] usb 3-4: Product: Wacom Intuos Pro M
[    3.743288] usb 3-4: Manufacturer: Wacom Co.,Ltd.
[    3.743293] usb 3-4: SerialNumber: 6LQ00A1020985
[    3.755940] input: Wacom Co.,Ltd. Wacom Intuos Pro M Pen as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:056A:0357.0001/input/input2
[    3.756030] wacom 0003:056A:0357.0001: hidraw0: USB HID v1.10 Mouse [Wacom Co.,Ltd. Wacom Intuos Pro M] on usb-0000:00:14.0-4/input0
[    3.756351] wacom 0003:056A:0357.0002: Unknown device_type for 'Wacom Co.,Ltd. Wacom Intuos Pro M'. Ignoring.

Pinglinux, 2018-08-27:
Your system may have an older driver left from your last update. Can you run "sudo make uninstall" inside your input-wacom and then build and install the driver again. Please make sure you run "sudo dracut --force" as suggested.

Thank you,
Ping

jigpu, 2019-11-14:
Closing as out-of-date. Please feel free to re-open if still an issue.

closed 2019-11-14 by jigpu