linuxwacom/input-wacom #37: Touchscreen shows 0% battery

goobliata, 2018-08-07:
I've got a Dell Inspiron 5379 with an active touchscreen. I'm not sure the details of the device but if you tell me how I'd be happy to get it.

As soon as I use the stylus for the first time, I notice there's a keyboard in battery settings with 0% battery and I think it's related to the touchscreen/stylus. A message pops up saying the battery is low. 

The real problem is after that the screen starts dimming after just a few seconds, presumably to conserve battery power. I need to reboot to fix it so consequently I never bother using the stylus.

If you need help debugging this just let me know what you want me to do.

Thanks.

jigpu, 2018-08-08:
Probably the best way to start would be to install the "[hid-replay](https://bentiss.github.io/hid-replay-docs/)" tools. Once installed, run `sudo hid-recorder > /tmp/pen-$(date +"%s").log`. When prompted, type in the device number for your pen sensor and then press Enter. Next, bring the pen into proximity and move it around for about 10 seconds. Finally, press CTRL+C and then upload the log file for review (you should be able to attach it directly to this bug report).

goobliata, 2018-08-09:
[pen-1533799710.log](https://github.com/linuxwacom/input-wacom/files/2273206/pen-1533799710.log)

There it is. Pretty sure I got the right device but there was another DELLXXX device I didn't recognise. Thanks.

jigpu, 2018-08-09:
That looks like the right device. Unfortunately, it doesn't look like a driver issue: the hardware is reporting that the battery is at 1%. Assuming that you're sure the battery is fresh, the only other possibilities I can think of are:

- Needing to tell the hardware to e.g. switch to a different mode
- There being some kind of firmware bug
- There being some kind of incompatibility between the pen and the sensor

If you still have Windows in a partition on the machine, it would be interesting to know if the battery level is reported correctly. If it works there, it might be possible to reverse engineer why.

goobliata, 2018-08-09:
I pulled the original drive with Windows on it from the computer when I got it and still have it intact. Not really an easy solution but I'll let you know what I find out if I try putting it back in. Thanks.

goobliata, 2018-08-24:
Was just thinking, does it really make sense to dim the screen even if the pen battery really is at 1%? So the only way to stop it is rebooting or getting a new battery?

jigpu, 2018-08-24:
The screen dimming is almost certainly a bug in your desktop environment. I know that when we first added AES battery reporting to the driver, GNOME started behaving like you describe. Their code wasn't paying attention to what the battery was powering (i.e., the entire system or just a peripheral) and incorrectly enacted power-saving features like screen dimming when the battery level dropped low.

I thought this bug had been fixed in GNOME for a while, but I'm not 100% sure which version... You could potentially try booting your system from a USB stick with the latest Fedora live image to see if it also does screen dimming. The latest fedora will be running a minimally-modified version of the latest GNOME. If the screen dims there, a bug will need to be filed on the GNOME bug tracker. If not, a bug should be filed on the bug tracker for either your desktop environment (if not GNOME) or your distribution (if shipping an older version of GNOME).

Out of curiosity, what distro are you using?

goobliata, 2018-08-24:
Thanks. I'm running Ubuntu 18.04. I'll download Fedora and see what happens.

goobliata, 2018-08-25:
The screen still dims in Fedora 28, so I guess they haven't fixed it or it's something different. Would you happen to know which GNOME package I should file a bug against?

goobliata, 2018-08-25:
So I booted Windows on this laptop. I can't find anywhere on the system a battery level indicator for the stylus nor can I find any info about it on Google, other than for Microsoft Surface which this is not. Could it be that this device doesn't support it but the Wacom driver thinks it does?

goobliata, 2018-08-25:
If you need any testing done while I'm running Windows let me know.

jigpu, 2018-08-30:
Sorry about the delay in replying...

I'm not 100% sure which GNOME package is correct, but I'd recommend [reporting it against gnome-control-center](https://gitlab.gnome.org/GNOME/gnome-settings-daemon/issues). If it turns out to be incorrect, they should be able to move the bug or let you know where to re-open it.

As for the Windows testing, my understanding is that you'll need to install a driver from Wacom in order to see the battery information. See [this link](https://www.howtogeek.com/236195/how-to-find-out-which-build-and-version-of-windows-10-you-have/) for information on how to find out what version of Windows is installed. Then, use the appropriate link below:

* [Driver for Windows 10 versions 1803 & 1709](http://www.wacomeng.com/component/get.php?dlid=81533199783122657)
* [Driver for all other versions of Windows](http://www.wacomeng.com/component/get.php?dlid=0263003025)

I haven't installed these drivers myself, but you should afterwards have a control panel similar to what is shown at the bottom of [this page](https://support.hp.com/us-en/document/c04874553) and which will let you see your battery status. Let me know approximately what value it is reporting for your pen.

goobliata, 2018-08-30:
I don't think my touchscreen/pen are Wacom products but I'll give it a try. It will be delayed a bit though since I've switched back to Linux and swapping drives is a bit of a pain on this system. I'll post back when I have results.

jigpu, 2018-08-30:
Ah, indeed. I somehow forgot about that detail before replying. I'm not sure if there's a similar control panel in Windows that would report the battery status from the Elan sensor your Dell seems to have. The Wacom sensors, for what its worth, report *very* granular battery information, so I wouldn't be surprised for a "low" battery to report 1% even if it has e.g. 20% life left. There's a chance that replacing the battery would work around the issue, but GNOME would definitely need fixing regardless.

goobliata, 2018-08-30:
I just got a battery tester the other day and it looks fine, but to be sure I've tried 3 batteries, the one that came with it should be new and I tried another never used one plus one from another stylus so I don't think the battery is really low. No idea what the problem is though. Could be a faulty stylus I suppose. I'll report the bug to GNOME though.

goobliata, 2018-09-11:
The worst part of this has been fixed in this issue: https://gitlab.gnome.org/GNOME/gnome-settings-daemon/issues/83

The battery showing 1% still isn't fixed, but I've got another stylus on the way to double check if it's a hardware issue. I've found some info about the battery using lsusb -vd after unbinding the device. Attached is the whole thing with the battery strength part inline. Don't know if this is useful (I haven't figured it out yet) but there it is.

[usbdump.txt](https://github.com/linuxwacom/input-wacom/files/2369590/usbdump.txt)

                            Battery Strength
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Logical Maximum, data= [ 0x64 ] 100
            Item(Main  ): Input, data= [ 0x42 ] 66
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State Null_State Non_Volatile Bitfield
            Item(Global): Usage Page, data= [ 0x00 0xff ] 65280
                            (null)
            Item(Global): Report Count, data= [ 0x02 ] 2
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Input, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
            Item(Main  ): End Collection, data=none

jigpu, 2018-09-11:
Good to hear that GNOME has a fix in place now :)

The output you've provided is a snippet of the HID descriptor which describes how/where battery information is contained in the reports sent by the device. The hid-recorder logs I had you gather actually contain the entire descriptor (albeit as a blob of hex data that has to be decoded with e.g. `hidrd-convert`) which is the only reason I was able to understand the log at all and see that your hardware was reporting 1% battery. This part of the descriptor looks normal and shouldn't be a problem for the driver.

Hopefully the new hardware is better behaved...

goobliata, 2018-09-12:
I get the same behaviour with the new pen, so it must be some quirk with the hardware. I noticed in the kernel driver there is an quirk to ignore the battery of these things altogether, so maybe that's an option, although it has been 20 years since I compiled a kernel so I have no idea how to test it or get the proper ids for the device.

jigpu, 2018-09-19:
Could you point out where you see the quirk? I'm not very familiar with the Elan kernel drivers, but some of the quirks can be applied at runtime rather than needing to recompile a kernel.

goobliata, 2018-09-19:
drivers/hid/hid-input.c around line 300 you should see:

    #define HID_BATTERY_QUIRK_PERCENT	(1 << 0) /* always reports percent */
    #define HID_BATTERY_QUIRK_FEATURE	(1 << 1) /* ask for feature report */
    #define HID_BATTERY_QUIRK_IGNORE	(1 << 2) /* completely ignore the battery */

goobliata, 2018-09-19:
Here's a patch that should be close, if I can figure out how to compile a kernel on Ubuntu.

[elan.txt](https://github.com/linuxwacom/input-wacom/files/2397982/elan.txt)

jigpu, 2018-09-19:
Looks like there's no good way to inject the quirk at runtime, so your patch is probably necessary. It looks like it should do the trick to me...

I'm not too familiar with the process of compiling an Ubuntu kernel, but there should be some good guides out there that can help. If you do manage to get the patch working and tested, the next step for it would be to go to the [linux-input kernel mailinglist](http://vger.kernel.org/vger-lists.html#linux-input) where the HID driver is developed.

goobliata, 2018-09-19:
I was able to compile a kernel and the patch works. The stylus still works but there is no "1% battery" notification and GNOME Control Center doesn't show a battery. So I'll post the patch on linux-input. I already brought the issue up there.

Thanks a lot for your help!