linuxwacom/input-wacom #38: One by Wacom medium doesn't work on Ubuntu 18.04

PashaWNN, 2018-08-16:
It was working before update, after xorg package was updated it wasn't working. I did rollback via Timeshift and disabled system updates, but now it's broken again. I tried to recompile input-wacom, but compilation was failed
[config.log](https://github.com/linuxwacom/input-wacom/files/2293085/config.log)

PashaWNN, 2018-08-16:
I'm so sorry, I've downloaded a new version and it works

PashaWNN, 2018-10-17:
Now it not working again, with same error, with newest version

PashaWNN, 2018-10-17:
Getting `sign-file: certs/signing_key.pem: No such file or directory` on compile time

Secure boot is disabled, but module still not loading

varishtsg, 2018-11-04:
Facing the same issue. @jigpu here is the output of the command you mentioned above:

`<pre>if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi &amp;&amp; make &amp;&amp; sudo make install || echo &quot;Build Failed&quot;
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/4.15.0-38-generic/build
checking kernel version... 4.15.0-38-generic
checking RHEL7 minor release... 0
checking power supply version... v4.1+
checking abs accessor version... v2.6.36+

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/4.15.0-38-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run &apos;make &amp;&amp; make install&apos;.
make  all-recursive
make[1]: Entering directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
Making all in 4.5
make[2]: Entering directory &apos;/home/senpai/Downloads/input-wacom-0.42.0/4.5&apos;
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/4.15.0-38-generic/build M=/home/senpai/Downloads/input-wacom-0.42.0/4.5
make[3]: Entering directory &apos;/usr/src/linux-headers-4.15.0-38-generic&apos;
  Building modules, stage 2.
  MODPOST 2 modules
make[3]: Leaving directory &apos;/usr/src/linux-headers-4.15.0-38-generic&apos;
make[2]: Leaving directory &apos;/home/senpai/Downloads/input-wacom-0.42.0/4.5&apos;
make[2]: Entering directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
make[2]: Leaving directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
make[1]: Leaving directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
Making install in 4.5
make[1]: Entering directory &apos;/home/senpai/Downloads/input-wacom-0.42.0/4.5&apos;
make -C /lib/modules/4.15.0-38-generic/build M=/home/senpai/Downloads/input-wacom-0.42.0/4.5 modules_install
make[2]: Entering directory &apos;/usr/src/linux-headers-4.15.0-38-generic&apos;
  INSTALL /home/senpai/Downloads/input-wacom-0.42.0/4.5/wacom.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/senpai/Downloads/input-wacom-0.42.0/4.5/wacom_w8001.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-38-generic
make[2]: Leaving directory &apos;/usr/src/linux-headers-4.15.0-38-generic&apos;
mkdir -p /etc/depmod.d
echo &quot;override wacom * extra&quot; &gt; /etc/depmod.d/input-wacom.conf
PATH=&quot;/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin&quot; depmod -a 4.15.0-38-generic
/usr/sbin/update-initramfs -u -k 4.15.0-38-generic
update-initramfs: Generating /boot/initrd.img-4.15.0-38-generic
make[1]: Leaving directory &apos;/home/senpai/Downloads/input-wacom-0.42.0/4.5&apos;
make[1]: Entering directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
make[2]: Entering directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
make[2]: Nothing to be done for &apos;install-exec-am&apos;.
make[2]: Nothing to be done for &apos;install-data-am&apos;.
make[2]: Leaving directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;
make[1]: Leaving directory &apos;/home/senpai/Downloads/input-wacom-0.42.0&apos;</pre>`

jigpu, 2018-11-07:
Please try running `sudo rmmod wacom` and then `sudo insmod 4.5/wacom.ko` from inside the input-wacom source directory. If you get an error from the first command, try the second one anyway. Report back all errors either way.

Also, please run `md5sum 4.5/wacom.ko` from the input-wacom source directory, as well as `find /lib/modules/$(uname -r)/extra -type f -print0 | xargs -0 md5sum | grep wacom`

PashaWNN, 2018-11-08:
New version(0.42.0) works for me

varishtsg, 2018-11-10:
> Please try running `sudo rmmod wacom` and then `sudo insmod 4.5/wacom.ko` from inside the input-wacom source directory. If you get an error from the first command, try the second one anyway. Report back all errors either way.
> 
> Also, please run (and post the output of) `md5sum 4.5/wacom.ko` from the input-wacom source directory, as well as `find /lib/modules/$(uname -r)/extra -type f -print0 | xargs -0 md5sum | grep wacom`

This is what I get:
<pre>sudo rmmod wacom
[sudo] password for senpai: 
rmmod: ERROR: Module wacom is not currently loaded
<font color="#8AE234"><b>senpai@senpai-laptop</b></font>:<font color="#729FCF"><b>~/Downloads/input-wacom-0.42.0</b></font>$ sudo insmod 4.5/wacom.
wacom.ko     wacom.mod.o  wacom.o      
<font color="#8AE234"><b>senpai@senpai-laptop</b></font>:<font color="#729FCF"><b>~/Downloads/input-wacom-0.42.0</b></font>$ sudo insmod 4.5/wacom.ko 
insmod: ERROR: could not insert module 4.5/wacom.ko: Required key not available
<font color="#8AE234"><b>senpai@senpai-laptop</b></font>:<font color="#729FCF"><b>~/Downloads/input-wacom-0.42.0</b></font>$ md5sum 4.5/wacom.ko 
24420f91d4e57e26dff2f9ecaca1cf1b  4.5/wacom.ko
<font color="#8AE234"><b>senpai@senpai-laptop</b></font>:<font color="#729FCF"><b>~/Downloads/input-wacom-0.42.0</b></font>$ find /lib/modules/$(uname -r)/extra -type f -print0 | xargs -0 md5sum | grep wacom
40b52a3fc538f91358d024319c2bf41c  /lib/modules/4.15.0-38-generic/extra/<font color="#EF2929"><b>wacom</b></font>_w8001.ko
24420f91d4e57e26dff2f9ecaca1cf1b  /lib/modules/4.15.0-38-generic/extra/<font color="#EF2929"><b>wacom</b></font>.ko</pre>