linuxwacom/input-wacom #40: 'sticky' stylus, won't click or drag

MaanasArora, 2018-09-04:
I have installed the input-wacom and xf86-input-wacom drivers from both source and the repositories, however, my stylus would not click and not move when pressed down. I'm running Ubuntu 18.04.1 LTS. It has been this way since the beginning.

These are the files produced when I ran sysinfo.sh:
[sysinfo.7oadtDMbOF.tar.gz](https://github.com/linuxwacom/input-wacom/files/2349184/sysinfo.7oadtDMbOF.tar.gz)

Any hints?

jigpu, 2018-09-04:
Confirmed on a fresh Ubuntu 18.04 installation. The device works fine on my workstation though, so there's definitely something odd going on. The data stream from the kernel device seems to continue when the pen goes down, but I see it stop when monitoring for X11 events. I still get "RawMotion" events over XI2, but not regular Motion events, so I suspect that something is going wrong in the X driver. I'll need to dig around some more...

jigpu, 2018-09-05:
I've done a little more testing and am still finding really weird results. When I turn on the X driver logging to see what's going on, it appears that the events are still being processed as expected even when the pen tip is down. This means that it's not a bug with something like the palm rejection code that would be disabling the stream.

I'll have to add additional logging right at the point we send out events to the X server to make sure that they aren't being derailed at the last second. If they are, it shouldn't take too long to figure out the cause. If the events are making it all the way to our call to the X server though, it'll be a bit more complicated to debug since that means the server itself is eating our events for some reason.

MaanasArora, 2018-09-08:
Did you make it work? If not, any way I can for now?

jigpu, 2018-09-12:
I don't have a real fix, but I've attached a workaround that should get the pen working for the moment. If you cloned the input-wacom driver from git, you can run `git am 0001-HACK-Rename-Bamboo-PAD-for-Ubuntu-18.04-compatibilit.patch.txt` to apply the fix; otherwise you'll need to run `patch -p1 < 0001-HACK-Rename-Bamboo-PAD-for-Ubuntu-18.04-compatibilit.patch.txt` from inside the "input-wacom" source directory. Afterwards, recompile/reinstall/reboot.

[0001-HACK-Rename-Bamboo-PAD-for-Ubuntu-18.04-compatibilit.patch.txt](https://github.com/linuxwacom/input-wacom/files/2376273/0001-HACK-Rename-Bamboo-PAD-for-Ubuntu-18.04-compatibilit.patch.txt)

whot, 2018-09-27:
mutter's repo has `clutter/clutter/x11/clutter-device-manager-xi2.c:create_device()` with this bit:
```
      gchar *name;

      name = g_ascii_strdown (info->name, -1);

      if (strstr (name, "eraser") != NULL)
        source = CLUTTER_ERASER_DEVICE;
      else if (strstr (name, "cursor") != NULL)
        source = CLUTTER_CURSOR_DEVICE;
      else if (strstr (name, " pad") != NULL)
        source = CLUTTER_PAD_DEVICE;
```
And yeah, re-using "PAD" in the name of the device is probably going to cause issues with the libwacom udev rules too, isn't it?

whot, 2018-10-04:
> re-using "PAD" in the name of the device is probably going to cause issues with the libwacom udev rules too, isn't it?

good news here, it won't because we match on "Pad" and udev rule matching is case-sensitive. So libwacom's side is sorted in that regard afaict.

whot, 2018-10-04:
https://gitlab.gnome.org/GNOME/mutter/merge_requests/250 is the mutter fix, I hope.

whot, 2018-10-12:
Closing, the fix for mutter above assigns the right device type now, so this isn't something we need to track in libwacom.

closed 2018-10-12 by whot

boxbotteries, 2018-11-16:
This may need to be reopened, with the latest update my Bamboo pad stylus is no longer clicking or moving on press.

jigpu, 2018-11-19:
The fix hasn't yet been integrated into the GNOME codebase. Until it is, you will need to re-compile and install the input-wacom driver with the [above "HACK" patch](https://github.com/linuxwacom/input-wacom/issues/40#issuecomment-420740042) whenever your system installs a kernel update.

boxbotteries, 2018-11-19:
Did exactly that and lost all response to my tablet; it won't even acknowledge the stylus' movement now. Got a new system update just now so will give it another try.

boxbotteries, 2018-11-21:
Found out I was recompiling an older version of the input-wacom driver. Recompiled the latest version with the hack patch and everything is back to normal! Thanks @jigpu .

MaanasArora, 2018-11-21:
Oh, me too @boxbotteries. Thanks!