linuxwacom/input-wacom #43: Wacom One CTL-672 is not detected anymore in Ubuntu 18.04

olegkqa, 2018-10-06:
Hello.
I have Wacom CTL-672 and Ubuntu 18.04.1 LTS.
 I installed the driver following the instruction here: https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source
And it worked fine just one day, but then (apparently after sleep-mode or turning off) it stopped working. 
In system settings there's message "No tablet detected", but it is detected as a usb device when I type "lsusb" I get "Bus 004 Device 004: ID 056a:037b Wacom Co., Ltd".
I tried to reinstall driver by using the same instruction, but still no result.
When I type command "grep "" /sys/module/wacom*/version" I get the message "/sys/module/wacom*/version: No such file or directory".
Please, help.

Here is the installation log with a couple of errors:

if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/4.15.0-36-generic/build
checking kernel version... 4.15.0-36-generic
checking RHEL7 minor release... 0
checking power supply version... rm: cannot remove '.autoconf/build/conftest.mod.o': Permission denied
rm: cannot remove '.autoconf/build/conftest.c': Permission denied
rm: cannot remove '.autoconf/build/modules.order': Permission denied
rm: cannot remove '.autoconf/build/conftest.o': Permission denied
rm: cannot remove '.autoconf/build/conftest.ko': Permission denied
rm: cannot remove '.autoconf/build/.conftest.ko.cmd': Permission denied
rm: cannot remove '.autoconf/build/.conftest.o.cmd': Permission denied
rm: cannot remove '.autoconf/build/conftest.mod.c': Permission denied
rm: cannot remove '.autoconf/build/Module.symvers': Permission denied
rm: cannot remove '.autoconf/build/.cache.mk': Permission denied
rm: cannot remove '.autoconf/build/Makefile': Permission denied
rm: cannot remove '.autoconf/build/.conftest.mod.o.cmd': Permission denied
rm: cannot remove '.autoconf/build/.tmp_versions/conftest.mod': Permission denied
./configure: line 3682: .autoconf/build/Makefile: Permission denied
pre-v4.1
checking abs accessor version... rm: cannot remove '.autoconf/build/conftest.mod.o': Permission denied
rm: cannot remove '.autoconf/build/conftest.c': Permission denied
rm: cannot remove '.autoconf/build/modules.order': Permission denied
rm: cannot remove '.autoconf/build/conftest.o': Permission denied
rm: cannot remove '.autoconf/build/conftest.ko': Permission denied
rm: cannot remove '.autoconf/build/.conftest.ko.cmd': Permission denied
rm: cannot remove '.autoconf/build/.conftest.o.cmd': Permission denied
rm: cannot remove '.autoconf/build/conftest.mod.c': Permission denied
rm: cannot remove '.autoconf/build/Module.symvers': Permission denied
rm: cannot remove '.autoconf/build/.cache.mk': Permission denied
rm: cannot remove '.autoconf/build/Makefile': Permission denied
rm: cannot remove '.autoconf/build/.conftest.mod.o.cmd': Permission denied
rm: cannot remove '.autoconf/build/.tmp_versions/conftest.mod': Permission denied
./configure: line 3743: .autoconf/build/Makefile: Permission denied
pre-v2.6.36

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 4.5
      kernel source - yes /lib/modules/4.15.0-36-generic/build

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory '/home/user/Downloads/input-wacom-0.41.0'
Making all in 4.5
make[2]: Entering directory '/home/user/Downloads/input-wacom-0.41.0/4.5'
    Building input-wacom drivers for 4.5 kernel.
make -C /lib/modules/4.15.0-36-generic/build M=/home/user/Downloads/input-wacom-0.41.0/4.5
make[3]: Entering directory '/usr/src/linux-headers-4.15.0-36-generic'
Makefile:975: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  CC [M]  /home/user/Downloads/input-wacom-0.41.0/4.5/wacom_wac.o
/home/user/Downloads/input-wacom-0.41.0/4.5/wacom_wac.c: In function ‘wacom_setup_pen_input_capabilities’:
/home/user/Downloads/input-wacom-0.41.0/4.5/wacom_wac.c:3517:3: warning: this statement may fall through [-Wimplicit-fallthrough=]
   __clear_bit(ABS_MISC, input_dev->absbit);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/Downloads/input-wacom-0.41.0/4.5/wacom_wac.c:3519:2: note: here
  case WACOM_MO:
  ^~~~
  CC [M]  /home/user/Downloads/input-wacom-0.41.0/4.5/wacom_sys.o
  LD [M]  /home/user/Downloads/input-wacom-0.41.0/4.5/wacom.o
  Building modules, stage 2.
  MODPOST 2 modules
  LD [M]  /home/user/Downloads/input-wacom-0.41.0/4.5/wacom.ko
make[3]: Leaving directory '/usr/src/linux-headers-4.15.0-36-generic'
make[2]: Leaving directory '/home/user/Downloads/input-wacom-0.41.0/4.5'
make[2]: Entering directory '/home/user/Downloads/input-wacom-0.41.0'
make[2]: Leaving directory '/home/user/Downloads/input-wacom-0.41.0'
make[1]: Leaving directory '/home/user/Downloads/input-wacom-0.41.0'
[sudo] password for user: 
Making install in 4.5
make[1]: Entering directory '/home/user/Downloads/input-wacom-0.41.0/4.5'
make -C /lib/modules/4.15.0-36-generic/build M=/home/user/Downloads/input-wacom-0.41.0/4.5 modules_install
make[2]: Entering directory '/usr/src/linux-headers-4.15.0-36-generic'
Makefile:975: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  INSTALL /home/user/Downloads/input-wacom-0.41.0/4.5/wacom.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/user/Downloads/input-wacom-0.41.0/4.5/wacom_w8001.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-36-generic
make[2]: Leaving directory '/usr/src/linux-headers-4.15.0-36-generic'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin:/bin:/sbin" depmod -a 4.15.0-36-generic
/usr/sbin/update-initramfs -u -k 4.15.0-36-generic
update-initramfs: Generating /boot/initrd.img-4.15.0-36-generic
make[1]: Leaving directory '/home/user/Downloads/input-wacom-0.41.0/4.5'
make[1]: Entering directory '/home/user/Downloads/input-wacom-0.41.0'
make[2]: Entering directory '/home/user/Downloads/input-wacom-0.41.0'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/user/Downloads/input-wacom-0.41.0'
make[1]: Leaving directory '/home/user/Downloads/input-wacom-0.41.0'

Zeredan, 2018-10-09:
If you are using the 4.16 kernel, you just have to unplug it and plug it back in, and after that, it should work.

uzabon, 2018-10-11:
maybe you should try to install the dependent component first as it indicates.
> Makefile:975: "Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel"

 As for the SSL error, it seems that the module is not signed, so either validating the module with signed keys or turning off the module signing will help.
BUT, the these won't influence the installation and use in fact.
> sign-file: certs/signing_key.pem: No such file or directory

jigpu, 2018-10-31:
Did any of the above suggestions help?

jigpu, 2018-11-07:
Please try running our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attaching the generated tarball to this issue. It will gather a number of logs and other system information that should give us a better idea of what's going on...

varishtsg, 2018-11-10:
> Please try running our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attaching the generated tarball to this issue. It will [...]

[sysinfo.HIPhUh736S.tar.gz](https://github.com/linuxwacom/input-wacom/files/2568174/sysinfo.HIPhUh736S.tar.gz)

@jigpu Here is the tarball you requested.

jigpu, 2018-11-13:
It looks like the kernel is refusing to load the wacom.ko kernel module that you've compiled and installed. Specifically, the dmesg log states `Lockdown: Loading of unsigned modules is restricted; see man kernel_lockdown.7` right after detecting the device and when it would normally load the module.

You could try temporarily disabling Secure Boot to see if the device starts working. If it does (and you want to continue using Secure Boot), then you'll need to sign the kernel module and ensure that the key is registered with your system. The following two links look helpful in this regard:

* "[How can I do non-automated signing of drivers?](https://wiki.ubuntu.com/UEFI/SecureBoot#How_can_I_do_non-automated_signing_of_drivers.3F)"
* "[How to sign your own UEFI binaries for Secure Boot](https://wiki.ubuntu.com/UEFI/SecureBoot/Signing)"

Read through the above links to be sure, but I think the following three commands should do the trick:
~~~
$ sudo update-secureboot-policy --new-key
$ sudo update-secureboot-policy --enroll-key
$ sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der /lib/modules/$(uname -r)/extra/wacom.ko
~~~
After generating and enrolling a key with the first two commands its possible that modules will be automatically signed in the future. If not, the last command should be all that's necessary to sign updated versions of the module.

varishtsg, 2019-01-05:
> It looks like the kernel is refusing to load the wacom.ko kernel module that you've compiled and installed. Specifically, the dmesg log states `Lockdown: Loading of unsigned modules is restricted; [...]

Sorry for the late update. The above method works. I was able to manually sign and use the module.

closed 2019-01-09 by jigpu

jameswilddev, 2019-01-27:
I can't get this to work with secure boot enabled.  

```
$ sudo update-secureboot-policy --new-key
$ sudo update-secureboot-policy --enroll-key
No DKMS modules installed.
$ sudo kmodsign sha512 /var/lib/shim-signed/mok/MOK.priv /var/lib/shim-signed/mok/MOK.der /lib/modules/$(uname -r)/extra/wacom.ko
$ sudo insmod 4.5/wacom.ko
insmod: ERROR: could not insert module 4.5/wacom.ko: Required key not available
```

Any ideas what I might be doing wrong?

jigpu, 2019-01-28:
Did the second command ask you for a temporary password and then instruct you to reboot? If not, then its likely that the key wasn't actually enrolled and was never loaded into the kernel. Try running the following command, which should ask you for a temporary password and then instruct you to reboot:

~~~
$ sudo mokutil --enable-validation
~~~

After rebooting, the blue-colored "MokManager" utility should start up instead of Linux. You'll need to tell it to `Enroll MOK` and then you can either `View Key 0` or `Continue`. When asked to enroll the key, select `Yes` and enter the temporary password. Afterwards tell it to `Reboot` and the system should restart into Linux. At this point the kernel *should* have the key loaded into memory and be able to load the kernel module (and indeed, if your tablet was connected, already have done so without you needing to run `insmod` manually).

jameswilddev, 2019-01-30:
Aha yeah I've since found the rest of the instructions and got it installed.  Thanks!

shashank-16, 2021-07-20:
Yes also facing the same problem. 
Setting is detecting the device name  but unable to move pointer with pen and xsetwacom commands are not working in the terminal and a only the xsetwacom --help is working .
I think that the driver is not detecting it properly . plz help