linuxwacom/input-wacom #44: I can't change settings on my Wacom CTL-672

Zeredan, 2018-10-09:
I couldn't change it with Manjaro's built-in GUI settings.

`xsetwacom` doesn't seem to recognize my tablet.
```
 $ xsetwacom --list devices
 $ 
```

`xinput` sees it as a generic mouse, because I can't change the mode of it to "absolute".
```
$ xinput --set-mode 9 ABSOLUTE
X Error of failed request:  XI_BadMode (invalid Mode parameter)
  Major opcode of failed request:  131 (XInputExtension)
  Minor opcode of failed request:  5 (X_SetDeviceMode)
  Mode id in failed request: 0x17
  Serial number of failed request:  18
  Current serial number in output stream:  18

```

jigpu, 2018-10-09:
Do you know if you're running a Wayland or Xorg session? If you don't know, post the output of `xinput list` for me.

Zeredan, 2018-10-09:
```
$ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder 2013             	id=14	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder 2013             	id=13	[slave  pointer  (2)]
⎜   ↳ Wacom Co.,Ltd. CTL-672                  	id=9	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Microsoft® LifeCam VX-2000: Mi         	id=10	[slave  keyboard (3)]
    ↳   USB Keyboard                          	id=11	[slave  keyboard (3)]
    ↳   USB Keyboard                          	id=12	[slave  keyboard (3)]
    ↳ Razer Razer DeathAdder 2013             	id=15	[slave  keyboard (3)]
    ↳ Power Button 
```
I don't think I use them.

jigpu, 2018-10-12:
It almost looks like the device isn't being correctly picked up the kernel. The name should be different than "Wacom Co.,Ltd. CTL-672"... If the wrong kernel driver is being used that could also explain why it doesn't show up in xsetwacom. Please try rebooting, running our [sysinfo.sh script](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh), and uploading the generated tarball to this bug. It should hopefully contain enough information to figure out why the tablet isn't being detected as expected.

jigpu, 2019-11-14:
Closing as out-of-date. Please feel free to re-open if it is still an issue.

closed 2019-11-14 by jigpu