linuxwacom/input-wacom #48: HP Elite x2 1013 G3: i2c_hid errors on boot

RussianNeuroMancer, 2018-11-01:
Hello!

On HP Elite x2 1013 G3 I see error messages on related to loading driver for Wacom 4870 input device: 

> **[    2.519741] i2c_hid i2c-WCOM4870:00: i2c-WCOM4870:00 supply vdd not found, using dummy regulator
> [    2.519755] i2c_hid i2c-WCOM4870:00: Linked as a consumer to regulator.0
> [    2.519757] i2c_hid i2c-WCOM4870:00: i2c-WCOM4870:00 supply vddl not found, using dummy regulator**
> [    2.860577] input: WCOM4870:00 056A:4870 Touchscreen as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input12
> [    2.860937] input: WCOM4870:00 056A:4870 as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input13
> [    2.861120] input: WCOM4870:00 056A:4870 as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input14
> [    2.861437] input: WCOM4870:00 056A:4870 as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input15
> [    2.861636] input: WCOM4870:00 056A:4870 Mouse as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input16
> [    2.861717] hid-generic 0018:056A:4870.0001: input,hidraw0: I2C HID v1.00 Mouse [WCOM4870:00 056A:4870] on i2c-WCOM4870:00
> [    3.149446] input: Wacom HID 4870 Pen as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input20
> [    3.149837] input: Wacom HID 4870 Finger as /devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-WCOM4870:00/0018:056A:4870.0001/input/input21
> [    3.150314] wacom 0018:056A:4870.0001: hidraw0: I2C HID v1.00 Mouse [WCOM4870:00 056A:4870] on i2c-WCOM4870:00
> **[    4.283793] i2c_hid i2c-WCOM4870:00: failed to set a report to device.
> [    4.285058] i2c_hid i2c-WCOM4870:00: failed to set a report to device.**

Sysinfo and complete dmesg is here: https://github.com/linuxwacom/wacom-hid-descriptors/issues/42

Pinglinux, 2018-11-02:
With those “failed” message, does your pen and touch still work?

RussianNeuroMancer, 2018-11-02:
Yes, both of touchscreen and pen works.

jigpu, 2018-11-07:
Still trying to understand what's causing these errors. We have [logs from a similar system](https://github.com/linuxwacom/wacom-hid-descriptors/tree/master/Dell%20XPS%2015%209575/sysinfo.QW6usrrbMq) that doesn't show these errors. Information about the Dell XPS 15 9575 (whose HID descriptor is identical aside from min/max values) was received from a user with Linux 4.19-rc0 and Fedora 28. There is no indication of problems on that system.

The error returned from 'wacom_set_device_mode' is -EREMOTEIO. The i2c_designware driver that your system has should only return that error when the transfer is aborted due to a "noack" condition (https://elixir.bootlin.com/linux/latest/source/drivers/i2c/busses/i2c-designware-common.c#L308). This could be due to a bus error, the sensor not responding, or the sensor not liking the commands that we're sending it.

The "supply vdd not found, using dummy regulator" errors may be unrelated, but could also point to possible errors in the system's ACPI tables...

Please try following the instructions to [temporarily add a kernel parameter for testing](https://wiki.ubuntu.com/Kernel/KernelBootParameters#Temporarily_Add_a_Kernel_Boot_Parameter_for_Testing) and add the parameter `i2c_hid.debug=1` to the end of the kernel commandline. When the system starts up, run `journalctl -b0 | grep -Ei -C5 "WCOM|wacom" > ~/wacom-i2c.log` and then reboot again to disable the logging. On the second startup, attach the wacom-i2c.log that is in your home directory to this bug.

RussianNeuroMancer, 2018-11-09:
Thank you for looking into this issue!

wacom-i2c.log is here: https://paste.fedoraproject.org/paste/3naM-FsaApf-f-NcNBlkcw

jigpu, 2018-11-15:
The commands being sent to the sensor do look a little suspect, but I'll need more time to fully understand if the data is normal or not...

Pinglinux, 2022-08-05:
We have no idea what those set commands were doing. Since the device worked and the commands were from i2c_hid, it could be something specific in the common i2c HID layer. Just ignore them.

closed 2022-08-05 by Pinglinux