linuxwacom/input-wacom #53: failed to link up the device via "Wireless Accessory Kit" - "unknown PID"

zerobikappa, 2018-11-09:
device: wacom cth-690 art
kernel: 3.10.0-862.14.4.el7.x86_64

system log as below:
Nov  9 14:44:24 kappas kernel: usb 4-2: new full-speed USB device number 3 using uhci_hcd
Nov  9 14:44:25 kappas kernel: usb 4-2: New USB device found, idVendor=056a, idProduct=0084
Nov  9 14:44:25 kappas kernel: usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Nov  9 14:44:25 kappas kernel: usb 4-2: Product: Wacom Wireless Receiver
Nov  9 14:44:25 kappas kernel: usb 4-2: Manufacturer: Wacom Co.,Ltd.
Nov  9 14:44:25 kappas mtp-probe: checking bus 4, device 3: "/sys/devices/pci0000:00/0000:00:1a.1/usb4/4-2"
Nov  9 14:44:25 kappas mtp-probe: bus: 4, device: 3 was not an MTP device
Nov  9 14:44:45 kappas kernel: wacom 4-2:1.0: wireless tablet connected with PID 33e
Nov  9 14:44:45 kappas kernel: wacom 4-2:1.0: ignoring unknown PID.

jigpu, 2018-11-15:
The CTH-690 is not supported by the Linux 3.10 kernel. It is supported by the current version of this input-wacom driver, however. If you compile and install it the error should go away.

closed 2018-11-15 by jigpu

zerobikappa, 2018-11-15:
I tried to compile and install input-wacom driver but failed. The commend line showed "Can't read private key" when installing hid-wacom.ko and wacom_w8001.ko.
How could I deal with this problem?

$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"

checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... none
checking for kernel type... Linux
checking for linux-based kernel... yes
checking for kernel source/headers... /lib/modules/3.10.0-862.14.4.el7.x86_64/source
checking kernel version... 3.10.0-862.14.4.el7.x86_64
checking RHEL7 minor release... 4
checking power supply version... v4.1+
checking abs accessor version... v2.6.36+

checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating 2.6.32/Makefile
config.status: creating 2.6.38/Makefile
config.status: creating 3.7/Makefile
config.status: creating 3.17/Makefile
config.status: creating 4.5/Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure:
----------------------------------------
  BUILD ENVIRONMENT:
       linux kernel - yes 3.17
      kernel source - yes /lib/modules/3.10.0-862.14.4.el7.x86_64/source

NOTE: The kernel drivers included in this package are only tested with the
X Wacom driver built from xf86-input-wacom. If you are running an X server
version older than 1.7, please use the drivers provided by the linuxwacom
package.

Please run 'make && make install'.
make  all-recursive
make[1]: Entering directory `/home/myusername/downloads/input-wacom-0.42.0'
Making all in 3.17
make[2]: Entering directory `/home/myusername/downloads/input-wacom-0.42.0/3.17'
    Building input-wacom drivers for 3.17 kernel.
make -C /lib/modules/3.10.0-862.14.4.el7.x86_64/source M=/home/myusername/downloads/input-wacom-0.42.0/3.17
make[3]: Entering directory `/usr/src/kernels/3.10.0-862.14.4.el7.x86_64'
  Building modules, stage 2.
  MODPOST 2 modules
make[3]: Leaving directory `/usr/src/kernels/3.10.0-862.14.4.el7.x86_64'
make[2]: Leaving directory `/home/myusername/downloads/input-wacom-0.42.0/3.17'
make[2]: Entering directory `/home/myusername/downloads/input-wacom-0.42.0'
make[2]: Leaving directory `/home/myusername/downloads/input-wacom-0.42.0'
make[1]: Leaving directory `/home/myusername/downloads/input-wacom-0.42.0'
Making install in 3.17
make[1]: Entering directory `/home/myusername/downloads/input-wacom-0.42.0/3.17'
make -C /lib/modules/3.10.0-862.14.4.el7.x86_64/source M=/home/myusername/downloads/input-wacom-0.42.0/3.17 modules_install
make[2]: Entering directory `/usr/src/kernels/3.10.0-862.14.4.el7.x86_64'
  INSTALL /home/myusername/downloads/input-wacom-0.42.0/3.17/hid-wacom.ko
Can't read private key
  INSTALL /home/myusername/downloads/input-wacom-0.42.0/3.17/wacom_w8001.ko
Can't read private key
  DEPMOD  3.10.0-862.14.4.el7.x86_64
make[2]: Leaving directory `/usr/src/kernels/3.10.0-862.14.4.el7.x86_64'
mkdir -p /etc/depmod.d
echo "override wacom * extra" > /etc/depmod.d/input-wacom.conf
PATH="/sbin:/bin:/usr/sbin:/usr/bin:/bin:/sbin" depmod -a 3.10.0-862.14.4.el7.x86_64
make[1]: Leaving directory `/home/myusername/downloads/input-wacom-0.42.0/3.17'
make[1]: Entering directory `/home/myusername/downloads/input-wacom-0.42.0'
make[2]: Entering directory `/home/myusername/downloads/input-wacom-0.42.0'
make[2]: Nothing to be done for `install-exec-am'.
make[2]: Nothing to be done for `install-data-am'.
make[2]: Leaving directory `/home/myusername/downloads/input-wacom-0.42.0'
make[1]: Leaving directory `/home/myusername/downloads/input-wacom-0.42.0'