linuxwacom/input-wacom #55: Wacom Cintiq Pro 16 Pad working only a few seconds on Ubuntu 18.10

blenderli, 2018-11-24:
The Wacom Cintiq Pro 16 Pad is working for a view seconds on Xorg until it the pen and touch disappears, while the screen is still working and it remains as USB device.

It seams that Xorg is removing the device because the pen resolution is unknown:
`Wacom Cintiq Pro 16 Pad: libinput bug: missing tablet capabilities: pen resolution. Ignoring this device.`

The device is not seen as xinput device:
`
$ xinput list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Rapoo V200 gaming mouse Mouse           	id=9	[slave  pointer  (2)]
⎜   ↳ Rapoo V200 gaming mouse Consumer Control	id=10	[slave  pointer  (2)]
⎜   ↳ Razer Razer Tarantula Keyboard Consumer Control	id=12	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ Rapoo V200 gaming mouse                 	id=8	[slave  keyboard (3)]
    ↳ Razer Razer Tarantula Keyboard          	id=11	[slave  keyboard (3)]
    ↳ Razer Razer Tarantula Keyboard System Control	id=13	[slave  keyboard (3)]
    ↳ Rapoo V200 gaming mouse Consumer Control	id=16	[slave  keyboard (3)]
    ↳ Razer Razer Tarantula Keyboard Consumer Control	id=17	[slave  keyboard (3)]
`

It is still visible as USB device:
`
$ lsusb
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 044: ID 0403:6014 Future Technology Devices International, Ltd FT232H Single HS USB-UART/FIFO IC
Bus 007 Device 043: ID 05e3:0749 Genesys Logic, Inc. 
Bus 007 Device 042: ID 056a:0354 Wacom Co., Ltd 
Bus 007 Device 045: ID 056a:0350 Wacom Co., Ltd 
Bus 007 Device 041: ID 056a:0365 Wacom Co., Ltd 
Bus 007 Device 040: ID 056a:036c Wacom Co., Ltd 
Bus 007 Device 004: ID 1532:0102 Razer USA, Ltd Tarantula Keyboard
Bus 007 Device 002: ID 05e3:0604 Genesys Logic, Inc. USB 1.1 Hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 24ae:4100  
Bus 001 Device 002: ID 1e71:170e NZXT 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

$ xsetwacom --list devices
Does not show anything.

Package versions:
libwacom-bin; 0.31-1
libwacom-common; 0.31-1
libwacom; 0.31-1
xserver-xorg-input-wacom; 0.36.1-0ubuntu1

Full output of the Xorg log:
`
[    11.152] (--) Log file renamed from "/var/log/Xorg.pid-1855.log" to "/var/log/Xorg.0.log"
[    11.153] 
X.Org X Server 1.20.1
X Protocol Version 11, Revision 0
[    11.153] Build Operating System: Linux 4.4.0-138-generic x86_64 Ubuntu
[    11.153] Current Operating System: Linux linux-MS-7B09 4.18.0-11-generic #12-Ubuntu SMP Tue Oct 23 19:22:37 UTC 2018 x86_64
[    11.153] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.18.0-11-generic root=UUID=25e89904-760d-4c25-be26-d1eeadd326c6 ro quiet splash vt.handoff=1
[    11.153] Build Date: 25 October 2018  02:53:34PM
[    11.153] xorg-server 2:1.20.1-3ubuntu2.1 (For technical support please see http://www.ubuntu.com/support) 
[    11.153] Current version of pixman: 0.34.0
[    11.153] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[... 870 lines cut ...]
[    40.700] (II) No input driver specified, ignoring this device.
[    40.700] (II) This device may have been added with another device file.
[    40.703] (II) config/udev: Adding input device Wacom Cintiq Pro 16 Pad (/dev/input/js0)
[    40.703] (II) No input driver specified, ignoring this device.
[    40.703] (II) This device may have been added with another device file.
[    40.704] (II) config/udev: Adding input device Wacom Cintiq Pro 16 Pen (/dev/input/mouse2)
[    40.704] (II) No input driver specified, ignoring this device.
[    40.704] (II) This device may have been added with another device file.
[    40.881] (II) config/udev: Adding input device Wacom Cintiq Pro 16 Pen (/dev/input/event9)
[    40.881] (**) Wacom Cintiq Pro 16 Pen: Applying InputClass "libinput tablet catchall"
[    40.881] (II) Using input driver 'libinput' for 'Wacom Cintiq Pro 16 Pen'
[    40.961] (II) systemd-logind: got fd for /dev/input/event9 13:73 fd 29 paused 1
[    40.961] (II) systemd-logind: releasing fd for 13:73
[    41.056] (II) config/udev: Adding input device Wacom Cintiq Pro 16 Pad (/dev/input/event10)
[    41.056] (**) Wacom Cintiq Pro 16 Pad: Applying InputClass "libinput keyboard catchall"
[    41.056] (**) Wacom Cintiq Pro 16 Pad: Applying InputClass "libinput tablet catchall"
[    41.056] (II) Using input driver 'libinput' for 'Wacom Cintiq Pro 16 Pad'
[    41.129] (II) systemd-logind: got fd for /dev/input/event10 13:74 fd 29 paused 1
[    41.129] (II) systemd-logind: releasing fd for 13:74
[    41.196] (II) config/udev: Adding input device Wacom Cintiq Pro 16 Touch Finger (/dev/input/event8)
[    41.197] (**) Wacom Cintiq Pro 16 Touch Finger: Applying InputClass "libinput touchscreen catchall"
[    41.197] (II) Using input driver 'libinput' for 'Wacom Cintiq Pro 16 Touch Finger'
[    41.212] (II) systemd-logind: got fd for /dev/input/event8 13:72 fd 29 paused 1
[    41.212] (II) systemd-logind: releasing fd for 13:72
`

blenderli, 2018-11-29:
Hello jigpu
Thanks  
[sysinfo.pZGbE5tGNj.zip](https://github.com/linuxwacom/input-wacom/files/2631166/sysinfo.pZGbE5tGNj.zip)

jigpu, 2018-11-30:
Did you mean to close this issue? Has it resolved itself?

blenderli, 2018-11-30:
Hello jigpu

Sorry i press the wrong button. Can i send you the hole zip file from the sysinfo?

jigpu, 2018-12-03:
It looks like the sysinfo file is still uploaded to Github as [sysinfo.pZGbE5tGNj.zip](https://github.com/linuxwacom/input-wacom/files/2631166/sysinfo.pZGbE5tGNj.zip).

I've been able to replicate the issue locally and it seems like the problem is triggered by the "fwupd" service. If you run `sudo systemctl mask fwupd` and reboot the tablet should start working again. Unfortunately, disabling fwupd like this is not a great solution. I'll get in touch with the fwupd developers to let them know that something is going wrong...

superm1, 2018-12-04:
In the interim to this being fixed in fwupd, the "easy" solution is to blacklist the `wacomhid` plugin from fwupd loading.  Just modify `/etc/fwupd/daemon.conf` and add `wacomhid` to the `BlacklistPlugins` and restart your daemon.

jigpu, 2019-01-14:
Should be fixed now by fwupd-1.2.2

closed 2019-01-14 by jigpu