linuxwacom/input-wacom #57: Cannot change Wacom Intuos S cursor proximity

dmitry-kulikov, 2018-12-04:
Cannot change Wacom Intuos S cursor proximity. Exact model name is Wacom Intuos Basic Pen S CTL-4100K-N.
Default value:
```sh
xsetwacom get 'Wacom Intuos S Pen cursor' CursorProximity
10
```
Physically it is about 1 cm. I would like to move cursor only when pen touches pad, so I do:
```sh
xsetwacom set 'Wacom Intuos S Pen cursor' CursorProximity 1
```
(or 0 instead of 1, there should not be much difference)
Command itself works, I can check result with get command and it shows 1 for CursorProximity.
But actual behavior was not changed, proximity is still about 1 cm.
```sh
xsetwacom --version
0.36.1
```
I was able to reach desired behavior on Windows using GUI settings tool, so it is not problem of device itself.

dmitry-kulikov, 2018-12-04:
I want to do it because I would like to use Wacom as full mouse replacement. Default Wacom behavior may lead to occasional clicks which not acceptable in many cases.
Whole script which I use to simulate mouse:
```sh
# "xsetwacom list" shows:
DEVICE_STYLUS='Wacom Intuos S Pen stylus'
DEVICE_PAD='Wacom Intuos S Pad pad'
DEVICE_CURSOR='Wacom Intuos S Pen cursor'
DEVICE_ERASER='Wacom Intuos S Pen eraser'

xsetwacom set "$DEVICE_STYLUS" Mode "Relative" # mouse mode
xsetwacom set "$DEVICE_STYLUS" Button 1 "button +0" # nothing on touch
xsetwacom set "$DEVICE_STYLUS" Button 2 "button +1" # left click on lower button
xsetwacom set "$DEVICE_STYLUS" Button 3 "button +3" # right click on upper button

xsetwacom set "$DEVICE_PAD" Button 1 "key +Super_L" # "Super" key on 1st pad button
xsetwacom set "$DEVICE_PAD" Button 2 "button +2" # middle click on 2nd pad button
xsetwacom set "$DEVICE_PAD" Button 3 "button +9" # "forth" on 3rd pad button
xsetwacom set "$DEVICE_PAD" Button 8 "button +8" # "back" on 4th pad button

xsetwacom set "$DEVICE_CURSOR" Mode "Relative" # mouse mode
xsetwacom set "$DEVICE_CURSOR" CursorProximity 1 # cursor should be moved only if pen has contact with pad; unfortunately it does not work

xsetwacom set "$DEVICE_ERASER" Mode "Relative" # mouse mode
```
Everything works well except CursorProximity.

Pinglinux, 2018-12-05:
> I want to do it because I would like to use Wacom as full mouse
> replacement. Default Wacom behavior may lead to occasional clicks which not
> acceptable in many cases.
> Whole script which I use to simulate mouse:
>
> # "xsetwacom list" shows:
> DEVICE_STYLUS='Wacom Intuos S Pen stylus'
> DEVICE_PAD='Wacom Intuos S Pad pad'
> DEVICE_CURSOR='Wacom Intuos S Pen cursor'
> DEVICE_ERASER='Wacom Intuos S Pen eraser'
>
> xsetwacom set "$DEVICE_STYLUS" Mode "Relative" # mouse mode
> xsetwacom set "$DEVICE_STYLUS" Button 1 "button +0" # nothing on touch
> xsetwacom set "$DEVICE_STYLUS" Button 2 "button +1" # left click on lower button
> xsetwacom set "$DEVICE_STYLUS" Button 3 "button +3" # right click on upper button
>
> xsetwacom set "$DEVICE_PAD" Button 1 "key +Super_L" # "Super" key on 1st pad button
> xsetwacom set "$DEVICE_PAD" Button 2 "button +2" # middle click on 2nd pad button
> xsetwacom set "$DEVICE_PAD" Button 3 "button +9" # "forth" on 3rd pad button
> xsetwacom set "$DEVICE_PAD" Button 8 "button +8" # "back" on 4th pad button
>
> xsetwacom set "$DEVICE_CURSOR" Mode "Relative" # mouse mode
> xsetwacom set "$DEVICE_CURSOR" CursorProximity 1 # cursor should be moved only if pen has contact with pad; unfortunately it does not work
>
> xsetwacom set "$DEVICE_ERASER" Mode "Relative" # mouse mode
>
> Everything works well except CursorProximity.
I think what you wanted was Threshold, which decides at which pressure
level a click is triggered.

CursorProximity is for out of Proximity event.

Cheers,
Ping

dmitry-kulikov, 2018-12-05:
Thank you for the tip. Threshold is interesting possibility and works as described. However I would like to change distance at which tablet starts to send "cursor moved" event. Now cursor starts moving when distance between pad and stylus is about 1 cm. I would like to decrease this value, high value provides some discomfort with Mode "Relative". Is CursorProximity for that? If yes, it does not work for me. If I misunderstood CursorProximity then this issue should be closed.

Pinglinux, 2018-12-05:
Hi Dmitry,

In your case, CursorProximity is the one you need. However, the option
should be applied to your stylus, instead of cursor. Cursor is the physical
puck of the tablet, which is not supported by your Intuos. Stylus is the
tip of the pen. Eraser is the eraser side of the pen.

To make things a bit complicated is that CursorProximity is not supported
for pen since, in theory, no one would buy a Wacom device as a mouse
replacement. Hence, no one would change pen to relative mode. If you have
coding experience, you should be able to enable CursorProximity option for
xsetwacom yourself. We'll be very happy to review and accept your patches.
If you can't or don't want to make patches yourself, you'll need to wait
until one of us gets a chance to make it work for you.

Thank you for using Wacom devices, even if it is just a replacement of your
mouse :).
Ping