linuxwacom/input-wacom #59: Touch

MZc6MH, 2018-12-23:


MZc6MH, 2018-12-23:
I'm not sure if this is even the right one of the three projects, I'm not very experienced in these things. Please redirect me If I'm not in the right place

MZc6MH, 2018-12-23:
Another development (and this is reliably reproducible): when I reboot from the install manjaro into the live USB (without turning the machine off between, that is) the touchscreen doesn't work either and dmesg has the same message. Turning the machine off and then booting into the Live USB it works again...

jigpu, 2018-12-26:
We've occasionally had problems in the past where the pen/touch sensor was not being reset for some reason, leading to very similar symptoms. Does it seem like the sensor works more often when doing a full shutdown compared to a restart?

MZc6MH, 2018-12-26:
> pen/touch sensor was not being reset for some reason

That sounds exactly like what I'm experiencing (it _feels_ that way, if you know what I mean)

> Does it seem like the sensor works more often when doing a full shutdown compared to a restart?

Yes, that's right. I never had it work after a restart, but I did sometimes have it work after a full shutdown. Thanks for getting back to me, this doesn't really help my case for getting my parents to switch to Linux ;).

So what should I do to further debug (or solve it if you already know how) ?

MZc6MH, 2018-12-26:
This is what I get when running hid-recorder:
I don't really know what any of this means though :laughing: 

```
[src] $ sudo hid-recorder
Available devices:
/dev/hidraw0:  Wacom Co.,Ltd. Pen and multitouch sensor
/dev/hidraw1:  Wacom Co.,Ltd. Pen and multitouch sensor
Select the device event number [0-1]: 0
D: 0
R: 522 05 0d 09 04 a1 01 85 0c 95 01 75 08 26 ff 00 15 00 81 03 09 54 81 02 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 27 ff ff 00 00 75 10 95 01 09 56 81 02 85 0c 09 55 75 08 95 01 26 ff 00 b1 02 85 0a 06 00 ff 09 c5 96 00 01 b1 02 c0 06 11 ff 09 11 a1 01 85 03 a1 02 09 00 75 08 15 00 26 ff 00 95 27 81 02 c0 85 02 09 00 95 01 b1 02 85 03 09 00 95 3f b1 02 85 04 09 00 95 0f b1 02 85 07 09 00 96 00 01 b1 02 85 08 09 00 96 87 00 b1 02 85 09 09 00 96 3f 00 b1 02 85 0d 09 00 95 07 b1 02 c0 05 0d 09 0e a1 01 85 0e 09 23 a1 02 09 52 09 53 15 00 25 0a 75 08 95 02 b1 02 c0 c0
N: Wacom Co.,Ltd. Pen and multitouch sensor
P: usb-0000:00:14.0-5/input0
I: 3 056a 501b
^C
No events where recorded.
You may need to add the option "--debugfs" to get more recordings.
[src]$ sudo hid-recorder
Available devices:
/dev/hidraw0:  Wacom Co.,Ltd. Pen and multitouch sensor
/dev/hidraw1:  Wacom Co.,Ltd. Pen and multitouch sensor
Select the device event number [0-1]: 1
D: 0
R: 438 05 0d 09 02 a1 01 85 06 09 20 a1 00 09 42 09 44 09 45 09 3c 09 5a 09 32 15 00 25 01 75 01 95 06 81 02 95 02 81 03 05 01 09 30 27 a7 6b 00 00 47 a7 6b 00 00 65 11 55 0d 75 10 95 01 81 02 09 31 27 86 3c 00 00 47 86 3c 00 00 81 02 45 00 65 00 55 00 05 0d 09 30 26 ff 07 75 10 81 02 06 00 ff 09 5b 16 00 80 26 ff 7f 75 10 81 02 05 0d 09 5b 17 00 00 00 80 27 ff ff ff 7f 75 20 95 01 81 02 06 00 ff 09 00 75 08 95 01 81 02 05 0d 09 3b 75 08 95 01 26 ff 00 15 00 81 02 c0 c0 06 11 ff 09 02 a1 01 85 0b 09 20 a1 00 09 42 09 44 09 45 09 3c 09 5a 09 32 15 00 25 01 75 01 95 06 81 02 95 02 81 03 05 01 09 30 27 a7 6b 00 00 47 a7 6b 00 00 65 11 55 0d 75 10 95 01 81 02 09 31 27 86 3c 00 00 47 86 3c 00 00 81 02 45 00 65 00 55 00 05 0d 09 30 26 ff 07 75 10 81 02 06 00 ff 09 5b 16 00 80 26 ff 7f 75 10 81 02 05 0d 09 5b 17 00 00 00 80 27 ff ff ff 7f 75 20 95 01 81 02 06 00 ff 09 00 75 08 95 01 81 02 05 0d 09 3b 75 08 95 01 26 ff 00 15 00 81 02 c0 85 05 09 00 95 17 81 02 85 0f 09 00 95 27 81 02 85 0f 09 00 95 07 b1 02 85 11 09 00 95 09 b1 02 85 05 09 00 95 08 b1 02 85 10 09 00 96 3f 00 b1 02 85 0b 09 00 96 3f 00 b1 02 85 12 09 00 75 08 15 00 26 ff 00 96 05 01 b1 02 c0 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 95 02 75 01 81 02 95 01 75 06 81 03 05 01 09 30 09 31 26 ff 7f 75 10 95 02 81 02 c0 c0
N: Wacom Co.,Ltd. Pen and multitouch sensor
P: usb-0000:00:14.0-5/input1
I: 3 056a 501b
```

MZc6MH, 2018-12-26:
When I run the script in python/report.py I get this:
[hid-recorder.txt](https://github.com/linuxwacom/input-wacom/files/2710803/hid-recorder.txt)

MZc6MH, 2018-12-26:
While running these, I get two lines like this in dmesg for each time I run it:
```
[  991.117125] wacom 0003:056A:501B.0019: wacom_set_report: ran out of retries (last error = -110)
[  991.117130] wacom 0003:056A:501B.001A: wacom_set_report: ran out of retries (last error = -110)
```

MZc6MH, 2018-12-26:
In #28 the issue is solved with a modified power setting if I understand that correctly? How would I go about trying that out (if it's relevant)?

MZc6MH, 2019-01-09:
@jigpu Does the data I sent help in any way?

jigpu, 2019-01-14:
Sorry about the delay in getting back to you...

It doesn't look like the hid-recorder runs managed to capture any interesting data. Would you mind trying again, but while it is running touching either the pen or you finger to the screen. In theory one device should print out additional information in response to one kind of input; the other should respond to the other. If additional information is printed when touching the screen, please copy it back here for review. If not, then the sensor isn't talking to our kernel driver for some reason...

MZc6MH, 2019-01-18:
Would it be helpful if I ran hid-recorder while using the live-usb? Where touch _is_ working?

jigpu, 2019-02-13:
Sure, go ahead and run hid-recorder from the live USB and post the results here. I don't know if they'll be very useful, but the more data we have to work with the better. If you're able to momentarily get the touchscreen working from an installed version of Linux it would be good to fire up hid-recorder to see what kind of output is produced at those times too.

I haven't been able to come up with any plausible explanation for the Manjaro live USB would consistently work well but the installed version would fail. If you want to try toggling the autosuspend bit, you can do the following:

1. Use my [usbpath.sh script](https://gist.github.com/jigpu/180ca9cefcea64e3dc9b197db363857c) to get the USB path of your tablet. Just run `bash /path/to/usbpath.sh 056a:501b`

2. Use the path returned in step 1 to find the autosuspend node at `/sys/bus/usb/devices/<usbpath>/power/autosuspend`

3. Run `echo on > <autosuspend_path>` to disable autosuspend

MZc6MH, 2019-02-22:
sudo ./hid-recorder 
Available devices:
/dev/hidraw0:	Wacom Co.,Ltd. Pen and multitouch sensor
/dev/hidraw1:	Wacom Co.,Ltd. Pen and multitouch sensor
Select the device event number [0-1]: 0
D: 0
R: 522 05 0d 09 04 a1 01 85 0c 95 01 75 08 26 ff 00 15 00 81 03 09 54 81 02 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 09 22 a1 02 09 42 15 00 25 01 75 01 95 01 81 02 81 03 09 47 81 02 95 05 81 03 75 10 09 51 95 01 81 02 05 01 75 10 95 01 55 0e 65 11 09 30 26 10 2b 35 00 46 c4 0a 81 02 46 0d 06 09 31 26 34 18 81 02 c0 05 0d 27 ff ff 00 00 75 10 95 01 09 56 81 02 85 0c 09 55 75 08 95 01 26 ff 00 b1 02 85 0a 06 00 ff 09 c5 96 00 01 b1 02 c0 06 11 ff 09 11 a1 01 85 03 a1 02 09 00 75 08 15 00 26 ff 00 95 27 81 02 c0 85 02 09 00 95 01 b1 02 85 03 09 00 95 3f b1 02 85 04 09 00 95 0f b1 02 85 07 09 00 96 00 01 b1 02 85 08 09 00 96 87 00 b1 02 85 09 09 00 96 3f 00 b1 02 85 0d 09 00 95 07 b1 02 c0 05 0d 09 0e a1 01 85 0e 09 23 a1 02 09 52 09 53 15 00 25 0a 75 08 95 02 b1 02 c0 c0
N: Wacom Co.,Ltd. Pen and multitouch sensor
P: usb-0000:00:14.0-5/input0
I: 3 056a 501b
D: 0
E: 0.000000 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 11
E: 0.009902 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b8 11
E: 0.019890 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1c 12
E: 0.029888 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 12
E: 0.039907 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e4 12
E: 0.049926 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 48 13
E: 0.059907 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ac 13
E: 0.071903 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 10 14
E: 0.081905 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 14
E: 0.091846 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d8 14
E: 0.101847 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3c 15
E: 0.111892 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 a0 15
E: 0.121927 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 16
E: 0.131908 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 68 16
E: 0.143901 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 cc 16
E: 0.153894 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 30 17
E: 0.163878 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 94 17
E: 0.173879 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f8 17
E: 0.183883 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5c 18
E: 0.193874 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c0 18
E: 0.203871 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 24 19
E: 0.215882 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 88 19
E: 0.225885 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ec 19
E: 0.233884 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 1a
E: 0.243878 40 0c 00 01 04 01 00 c9 22 c1 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 b4 1a
E: 0.975908 40 0c 00 01 05 01 00 80 25 4b 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 70 36
[... cut ...]
E: 11.560188 40 0c 00 01 05 01 00 a2 07 b2 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 64 c8
E: 11.570189 40 0c 00 01 05 01 00 a2 07 b2 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 c8 c8
E: 11.578247 40 0c 00 01 05 01 00 85 07 88 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2c c9
E: 11.590233 40 0c 00 01 04 01 00 85 07 88 0a 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 90 c9

MZc6MH, 2019-02-22:
I get this with `journalctl -b | grep -i wacom` : 
[wacomlog.txt](https://github.com/linuxwacom/input-wacom/files/2894862/wacomlog.txt)

jigpu, 2019-03-07:
The data from the tablet is a bit weird... The rough translation of `E: 0.000000 40 0c 00 01 05 01 00 c9 22 c1 01 00 00 00 [...] 54 11` is:
* E: → Event recorded
* 0.000000 → Elapsed time = 0.000000
* 0x40 → Packet length: 0x40 bytes
* 0x0c → Report ID: 0x0c
* 0x00 → (padding)
* 0x01 → Total contacts present: 1
* 0x05 → First touch present and valid (0x01 + 0x04)
* 0x01 0x00 → Touch ID: 1
* 0xc9 0x22 → X location: 8905
* 0xc1 0x01 → Y location: 449
* 0x00 → Second touch not present and not valid
* 0x00 0x00 → (n/a because not present)
* [...]
* 0x54 0x11 → Sensor scan timestamp = 4436

The weird thing is that the values don't actually change like I'd expect. The first touch (from t=0.000000 to t=0.243878) is detected, doesn't move, and then leaves. The same is true for all subsequent touches as well. The scan timestamps are being updated though, weirdly enough. I'm not sure I've ever seen a sequence quite like that...

The HID descriptor itself is pretty normal (several other Lenovo devices use the same descriptor, with the only change being to the sensor's dimensions), so its unlikely that the sensor in your Yoga 12 is some kind of hardware we haven't seen before that requires special communication protocols.

Since you didn't have any luck changing the autosuspend value, try running `echo on > control` as root (won't work through sudo) in the path returned by the usbpath.sh script instead. It should succeed without an error, but I'm not sure if it will make the sensor return more data or not...

MZc6MH, 2019-03-09:
[log.txt](https://github.com/linuxwacom/input-wacom/files/2948297/log.txt)
[log2.txt](https://github.com/linuxwacom/input-wacom/files/2948298/log2.txt)

Tried various things, including the `echo on > control`. It seems like some movement is detected? I'm not sure I managed to read it correctly.

What about the gnome-shell stuff at the end of the `journalctl -b | grep -i wacom` output? `Could not get tablet information for 'Wacom Co.,Ltd. Pen and multitouch sensor Touchscreen pad': Device '/dev/input/event6' is not a tablet`

I might have to switch to windows, although it pains my heart :cry:

jigpu, 2019-03-12:
Thanks for the logs. It looks like the data is intermittent; sometimes there are good flows of data (that's probably when movement actually works) and other times it is just "frozen" and repeating the same data over and over. Adjusting the power settings doesn't seem to be a sure-fire fix, though it might have slightly improved things?

The "Could not get tablet information" message is a warning from the GNOME compositor that it doesn't recognize the device as a tablet and so won't apply settings. The cause is that libwacom wasn't able to find the expected udev tags on the device, which itself is because libwacom doesn't have a definition for your particular device. Fixing this is highly unlikely to affect the ability to use touch since even the Live version of Manjaro won't have the device definition and should be printing the same error message.

Have you tried any other distributions (live and/or installed) by any chance?

MZc6MH, 2019-04-11:
I actually did try a bunch of distributions, both installed and live, none worked other than manjaro (live). I tried Ubuntu (18.04 and 18.10) and debian for example IIRC. I've kind of abandoned getting it to work for the moment, might play with it some more in the next two weeks. Thank you for your help, and please do tell if you have any more things I can try or ways I can help you debug it

bjgroom, 2019-11-27:
> Have you tried any other distributions (live and/or installed) by any chance?

Hi, I have the same issue on Ubuntu 19.10, Yoga 12 Lenovo tablet. The touchscreen works perfectly on live session, but does not work after having installed Ubuntu. I'll work my way through the troubleshooting in this thread and report back if I find anything new.

MZc6MH, 2019-11-27:
Oh yes, please do. Maybe we can get it to work with another data point.
In the meantime, I found a more reliable way to have a touchscreen on an installed device. I installed antergos, and that makes the touchscreen work almost all the time when booting (not when rebooting or after resuming from sleep though). Sometimes it needs a couple of reboots. Sometimes it just doesn't work for a while.

I really have no clue why. I won't be able to test more since I will have access to the device only in the end of January.

@jigpu would you by any chance come to FOSDEM? If yes and if it interest you (and only then, I don't expect you to do work during such a fun event :wink:) I could take the machine with me so you could take a look at what is happening

MZc6MH, 2019-11-27:
This feels like a timing issue where changes to configuration make it so some race condition doesn't happen or happens less often. The live boot would maybe be slower and thus make some initialization happen at a better time or whatever?

bjgroom, 2019-11-29:
Well, I've run the hid-recorder on the installed Ubuntu 19.10 system, and it recorded zero events on hidraw0 and hidraw1. I tapped the display several times while the hid-recorder was running.

I have also run the hid-recorder on the Unbuntu live session, and it detected events on the hidraw0 device only. I don't have a stylus, so I presume the hidraw1 device would detect stylus input. 
I'll attach the output from the working hid-recorder session in a sec.

bjgroom, 2019-11-29:
> I installed antergos, and that makes the touchscreen work almost all the time when booting

This is really interesting! I found a reply in a related bug thread where someone indicated that touch screen worked from a cold boot on xfce:

https://github.com/linuxwacom/input-wacom/issues/110

I'm going to install Xubuntu 19.10 shortly and I'll report back with my findings.

bjgroom, 2019-11-29:
I'm also curious to maybe look at the code for wacom_set_report, and see the mechanism for the "retries" that ultimately max out. Any idea where this code is located?

bjgroom, 2019-11-29:
In case there is any interest in my hid_recorder output from the live Ubuntu session

[hid_recorded_0.log](https://github.com/linuxwacom/input-wacom/files/3906254/hid_recorded_0.log)

bjgroom, 2019-11-29:
Alright, so the touch screen works on Xubuntu live session only. Installed results in the same error, ran out of retries -110. I've also tried Ubuntu Budgie, and interestingly the touchscreen fails with this same error in the live session! I may try it installed just for curiosity, but I'm going to try Arch next for fun.

GRbit, 2020-01-21:
I've got Thinkpad Yoga 12.

Touchscreen stops working after sleep, but starts working after cold boot to xfce (xubuntu 18.04.3)

If there is need for some tests or log records - I'm here to help

bjgroom, 2020-03-04:
I've gotten the touchscreen working on Solus under kernel 5.5!

The touchscreen stops working after suspend, even with the Whisprin sleep fix patch applied. The filesystem structure is slightly different than Ubuntu releases, so I've been toying with the systemd service files from Whisprin's fix collection and trying to get the services to run correctly. 

The only way I've solved the touchscreen-after-suspend issue is to manually **sudo rmmod wacom** before sleep, and then **sudo modprobe wacom** after sleep.

I've also tried kernel 5.5 under Ubuntu, and the touchscreen still fails to work on an installed OS. I consistently receive **wacom_set_report: ran out of retries (last error: -110)** in dmesg.

GRbit, 2020-03-04:
> The only way I've solved the touchscreen-after-suspend issue is to manually **sudo rmmod wacom** before sleep, and then **sudo modprobe wacom** after sleep.

Wow! Sounds great actually! I'll try to write scrpt to automatically `rmmod` before sleep and modpronbe after.

caleb-allen, 2020-03-10:
@GRbit any luck?

GRbit, 2020-03-10:
@caleb-allen thanks for reminder)

I'm still can't get access to my Yoga 12. Borroweed it to friend of mine. I'll touch it for a while after two days and will test modprobe hack, I'll write about the result.
Later on weekends I'll have much more time to write and test script.

bjgroom, 2020-03-10:
> I'll try to write scrpt to automatically `rmmod` before sleep and `modpronbe` after.

@GRbit I believe this is what [Whisprin](https://github.com/Whisprin) has implemented in his [thinkpad-yoga-scripts](https://github.com/Whisprin/thinkpad-yoga-scripts) fix, using systemd. I haven't yet gotten this fix working under Solus, but I am very close. Good luck!

GRbit, 2020-03-15:
@caleb-allen finally tried it)

Well, in my case it was unusual.

After I run `sudo rmmod wacom` touchscreen was still working. Than I put laptop to sleep and when I resume work I didn't need to run `sudo modprobe wacom` because touchscreen was still working. So I put `rmmod wacom` to autorun and now feel just fine =)

I don't have wacom touch pen so can't test screen with that.

@bjgroom thanks for Whisprin link :+1:

Pinglinux, 2022-08-05:
This issue belongs to system integration, which is beyond this driver project's scope.

closed 2022-08-05 by Pinglinux