linuxwacom/input-wacom #61: Installing input wacom from source Build / Install "Build Failed" Error

BlobOfAwe, 2018-12-26:
I'm trying to set up my Wacom Intuos S with my Linux Mint computer.

OS: Linux Mint 18.1 Cinamon 64-bit
Linux Kernel: 4.4.0-53-generic

Followed the steps to install the 'input-wacom' pack, reached the Build/Install command, it returned 'Build Failed'

The tutorial says to ask the devs for help, but I'm not sure who to ask, so I'm here.

I'm new to programming, so please don't dump a bunch of techy lingo on me in the solution. The simpler the better.

jigpu, 2018-12-26:
Can you please copy/paste the full output from the terminal when running the install steps? We'll need this to understand what part failed.

BlobOfAwe, 2018-12-27:
$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
bash: ./configure: No such file or directory
Build Failed
`

jigpu, 2018-12-27:
Thanks! Did you run the command inside the extracted input-wacom directory? Could you run `ls -l` inside the directory for me so I can see what files are missing? Either the configure or autogen.sh script should be there...

BlobOfAwe, 2018-12-27:
Oh pfft. I forgot to download and extract the directory, that's why it wasn't working.

I extracted it and re-ran the program inside the directory and it works fine now. I just didn't read the instructions properly. Thanks for helping me though!

jigpu, 2018-12-27:
Glad it was something simple :) Hope the rest is smooth sailing.

closed 2018-12-27 by jigpu