linuxwacom/input-wacom #62: Pen stays in the "clicked" state as soon as it's connected, unless.I change the threshold

EternityForest, 2019-01-04:
I'm using a Wacom Bamboo, which works flawlessly if I set the threshold to 170 in the xorg conf, which was just a guess. I haven't tested to see what the lowest working value is. 

A few others have reported this problem, and I had a similar issue on an XP-Pen that's now packed away somewhere.

If this is common enough, would it make sense to either increase the default threshold, or run a self calibration where the threshold is set just above whatever the lowest value ever detected is, so that just hovering the pen won't be seen as a click?

I had the same issue, with the same fix on Mint, and Kubuntu. My Wacom was used when I got it, so there may be some kind of calibration drift.

Thanks a bunch guys!

jigpu, 2019-01-09:
I believe the driver is supposed to compensate for this automatically. If it enters proximity with a non-zero pressure value, it should take that value to be the real zero point so that the threshold works properly. I'm not sure why this wouldn't be working in your case... I'll have to do some digging...

EternityForest, 2019-01-09:
It could be that there's enough noise in the pressure reading in high EMI
areas that the noise pushes it above the zero point.

jigpu, 2019-01-14:
Please try running the [capture.sh script] to record what's going on. When you first run it, you'll probably be asked to install a few of its dependencies. Once they've been installed you should be asked to select a device -- choose the "stylus" device. Next, bring the pen in and out of prox a few times (without touching the pen to the surface) to be sure that the issue has been captured. Finally, hit CTRL+C and attach the generated tarball to this issue.

This should let us see the events that are sent by the device and our driver to narrow down where it comes from. E.g., a hardware issue, EMI (which has been known to cause occasional problems), or one of the software layers.

EternityForest, 2019-01-14:
Allright, here ya go! I don't know if it matters, but this is *with* the xorg.conf and the modifed threshold.
[record_1547507083.tar.gz](https://github.com/linuxwacom/input-wacom/files/2757391/record_1547507083.tar.gz)

jigpu, 2019-01-15:
Thanks! Those logs indicate that the hardware reported zero pressure at all times, aside from a brief moment (t=9.60 to t=9.71). The pressure values reported at this time were much higher than I would normally expect for interference (nearly 5% of full-scale). I don't see any indication that the pen would have been clicked at any other time in the log.

Could you re-run the capture script without the custom threshold set?

Pinglinux, 2022-08-09:
xf86-input-wcom or libinput, whichever is used, takes case of initial non-zero pressure issue. The issue can be fixed by replacing the worn-out pen tip with a new one.

closed 2022-08-09 by Pinglinux