linuxwacom/input-wacom #63: Touchscreen, Stylus, and Fingerprint sensor not working on Lenovo Ideapad Flex 5

danhab99, 2019-01-07:
I recently upgraded my Lenovo Flex 5 with Ubuntu and I need help linking up the touchscreen, stylus and fingerprint sensor. According to `hwinfo` they all exist. What didn't I do.

Ubuntu 18.10

```bash
dan@dan-Lenovo-ideapad-FLEX-5-1470:~$ xsetwacom --list
Wacom HID 50DC Pen stylus       	id: 12	type: STYLUS    
Wacom HID 50DC Finger touch     	id: 13	type: TOUCH     
Wacom HID 50DC Pen eraser       	id: 16	type: ERASER  

dan@dan-Lenovo-ideapad-FLEX-5-1470:~$ sudo xinput test-xi2 --root "Wacom HID 50DC Finger touch"
[sudo] password for dan: 
Wacom HID 50DC Finger touch             	id=13	[slave  pointer  (2)]
	Reporting 9 classes:
		Class originated from: 13. Type: XIButtonClass
		Buttons supported: 8
		Button labels: None None None None None None None None
		Button state:
		Class originated from: 13. Type: XIKeyClass
		Keycodes supported: 248
		Class originated from: 13. Type: XIValuatorClass
		Detail for Valuator 0:
		  Label: Abs X
		  Range: 0.000000 - 12368.000000
		  Resolution: 40000 units/m
		  Mode: absolute
		  Current value: 1974.000000
		Class originated from: 13. Type: XIValuatorClass
		Detail for Valuator 1:
		  Label: Abs Y
		  Range: 0.000000 - 6956.000000
		  Resolution: 40000 units/m
		  Mode: absolute
		  Current value: 4850.000000
		Class originated from: 13. Type: XIValuatorClass
		Detail for Valuator 2:
		  Label: Abs Pressure
		  Range: 0.000000 - 65536.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 13. Type: XIValuatorClass
		Detail for Valuator 3:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 13. Type: XIValuatorClass
		Detail for Valuator 4:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 13. Type: XIValuatorClass
		Detail for Valuator 5:
		  Label: None
		  Range: 0.000000 - 1.000000
		  Resolution: 1 units/m
		  Mode: absolute
		  Current value: 0.000000
		Class originated from: 13. Type: XITouchClass
		Touch mode: direct
		Max number of touches: 10

^C

dan@dan-Lenovo-ideapad-FLEX-5-1470:~$
```

jigpu, 2019-01-09:
What exactly is the problem? As you point out, the devices are listed in xsetwacom which means that they should work...

jdloft, 2019-10-31:
I have a Flex 5 1570 and the touchscreen and wacom devices work fine (with the exception of some weird interaction between palm rejection and the wacom device). As for the fingerprint reader, that is outside the scope of the input-wacom project. There is a project (https://github.com/nmikhailov/Validity90) that is/was working on reverse engineering the sensor.

danhab99, 2019-11-03:
Hmm, ya upon further work I did fix the wacom tablet (and the disabled it because who the hell needs a drawing tablet on a programmers computer), but I still need help with the fingerprint sensor so I'll take myself over to https://github.com/nmikhailov/Validity90 . 

Closing

closed 2019-11-03 by danhab99

jasonday, 2022-02-05:
@danhab99 How did you get the touch screen working?

danhab99, 2022-02-05:
Don't remember, I already sold that laptop