linuxwacom/input-wacom #64: new budget Cintiq 16 (DTK1660K0B)

paint-rulz, 2019-01-10:
Hello :-)
May I ask if the existing drivers in the kernel will cover the new Cintiq 16 (DTK1660K0B)?

skomra, 2019-01-14:
Hi paint-rulz,

Sorry for the delay in responding.

The new Cintiq 16 is supported by our generic kernel driver upstream. If you let me know which kernel you are running I can tell you what support on that kernel looks like (if it uses our generic driver or will need our backports driver.)

Best,
Aaron

paint-rulz, 2019-01-14:
Hi Aaron and thank you very much for your response; that's great news. I'm running kernel 4.19.13-1 on Manjaro KDE. 
Best regards, Cat

skomra, 2019-01-14:
Great, the kernel driver should work out of the box on 4.19.
-Aaron

closed 2019-01-14 by skomra

paint-rulz, 2019-01-19:
Hello again :-) I've received the Cintiq 16, which installs and works perfectly; thank you for work on the drivers. A real issue is that the screen isn't bright/contrasted enough, so much so that I may need to return it. May I ask if there is any way to calibrate that?

jigpu, 2019-01-21:
Your question doesn't seem to be in the Github issue for some reason, so
I'll reply to it here.

We don't yet have a replacement for the screen controls that are available
on Windows. If you have access to a Windows machine, it would probably be
best to get the calibration set up there and then reconnect the Cintiq to
your Linux machine; the calibration should be stored in the Cintiq from
what I understand.

If you don't have access to a Windows machine, you might take a look at
https://github.com/linuxwacom/xf86-input-wacom/issues/30 where someone has
had success getting the "ddcutil" command to change brightness and contrast
of the Cintiq 27QHD. I haven't tried it with the Cintiq 16, but maybe it
would also work.

paint-rulz, 2019-01-21:
Thanks jigpu :-) Sorry, I removed the question. Much appreciate advice. Great that there's the ddcutil option.