linuxwacom/input-wacom #65: Ubuntu 18.04 v0.42.0 pen jumps

blauertee, 2019-01-11:
My HP stylus pen jumps weirdly between it's last and it next position (draws straight lines in multiple drawing apps). 
I upgraded from v0.36.1 to v0.42.0 the bug persists. 
Also the pen is not showing up in the gnome preferences and neither in `xsetwacom --list`

But it does in xinput --list:
`⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ ELAN0732:00 04F3:264C                   	id=11	[slave  pointer  (2)]
⎜   ↳ SynPS/2 Synaptics TouchPad              	id=15	[slave  pointer  (2)]
⎜   ↳ ELAN0732:00 04F3:264C Pen Pen (0)       	id=18	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Video Bus                               	id=8	[slave  keyboard (3)]
    ↳ Power Button                            	id=9	[slave  keyboard (3)]
    ↳ HP Wide Vision HD Camera: HP Wi         	id=10	[slave  keyboard (3)]
    ↳ ELAN0732:00 04F3:264C Pen               	id=12	[slave  keyboard (3)]
    ↳ Intel Virtual Button driver             	id=13	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=14	[slave  keyboard (3)]
    ↳ HP WMI hotkeys                          	id=16	[slave  keyboard (3)]
    ↳ HP Wireless hotkeys                     	id=17	[slave  keyboard (3)]
`
Device: hp envy x360 15-cn0008ng
[Image showing jumps](http://jakob.zwoggel.net/Data/stylus_jumps.png)

Thanks!

petergthatsme, 2019-01-12:
I was going to submit a separate report about this (which i might still), but it might be related. 
I ve had a similar issue on 18.04 - basically every now and then click events get registered even when the pen is hovering (but not touching the screen). 
What helps, is installing the latest input-wacom (v2.00-0.42.0), however, it only helps with kernel 4.15.0-39-generic - when I install the latest input-wacom with the later kernels (say 4.15.0-42-generic), i see the same jumpy behavior again (it's intermittent, not always present). 
So you might want to try 4.15.0-39-generic and latest input-wacom.

And to the developers, just fyi, the fix that happened in the recent input-wacom, does not seem to work with latest ubuntu kernels. 
Thanks.

blauertee, 2019-01-14:
> I ve had a similar issue on 18.04 - basically every now and then click events get registered even when the pen is hovering (but not touching the screen).
Hm, I did not experience anything like that on my device.

> What helps, is installing the latest input-wacom (v2.00-0.42.0), however, it only helps with kernel 4.15.0-39-generic - when I install the latest input-wacom with the later kernels (say [...]

I tried to install 4.15.0-39-generic but I couldn't get my touch screen to work at all and neither my Stylus. But I installed wacom-input v2.00-0.42.0 on 4.15.0-29-generic, where I experience exactly the same jumping behavior. 

So I'm not sure weather our issues are related ...

jigpu, 2019-01-14:
Would you mind running our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to gather some additional information about your system? You should attach the generated tarball to a [new wacom-hid-descriptors issue](https://github.com/linuxwacom/wacom-hid-descriptors/issues). This info may be useful in the future if others have questions about the 15cn-0008ng...

Your device shows up as being from Elan, which our drivers aren't used for. I believe the kernel driver for your device is "hid-generic" and the Xorg driver is "xf86-input-libinput" (the sysinfo.sh output would confirm). I would create a new issue about the skipping against the [libinput project](https://gitlab.freedesktop.org/libinput/libinput/issues/). They should be able to help you narrow down the cause of the jumps and hopefully get things fixed.

closed 2019-01-14 by jigpu