linuxwacom/input-wacom #66: ubuntu 18.04 - registering ghost touch events

petergthatsme, 2019-01-16:
I am running ubuntu 18.04 on a Lenovo X1 yoga. Basically every now and then, the pen "touch events" (i.e. as if the pen were touching the screen) get registered even when the pen is hovering (but not actually touching the screen). When this problem is active, every time the pen comes gets "close" to the screen, a touch event gets registered. I then take the pen away, and bring it back close again, another touch event gets registered. 
What helps, is installing the latest input-wacom (v2.00-0.42.0), however, it only helps with kernel 4.15.0-39-generic - when I install the latest input-wacom with the later kernels (say 4.15.0-42-generic), i occasionally see the same behavior again (it's intermittent, not always present).

Any ideas why that would be and what it is about 4.15.0-39-generic that helps. 
(As a side note, I use fvwm as my window manager, but would imagine this would not be relevant to this behavior)

Thanks!

jigpu, 2019-02-13:
There shouldn't be any difference in behavior when using the same input-wacom release with two "patch" releases of the same kernel. I suspect that the intermittent nature of the problem just makes it seem like one kernel works better than the other.

I also suspect that the pen's pressure sensor is just a little too sensitive and periodically sends values large enough to be interpreted as a touch by our driver. You can change the click threshold by running `xsetwacom set <id> threshold` to a higher number as a workaround and once you find a good threshold may want to add the command to a login script.

If you want, you can try running our [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh) to record the device, kernel, and X events that are generated when bringing the pen into and out of proximity. If you're careful to never touch the tablet yet see a touch event generated while recording, you can press CTRL+C and attach the generated tarball to this bug.

therahedwig, 2019-02-13:
Hi, I have a virtually same issue, except my device is the Yoga 920, am running KDE neon(also based on ubuntu 18.04) and I didn't change anything about the kernel. I ran the script, and the following is a sample where the stylus never touches the screen, I just bring it into proximity several times.

[record_1550088058.tar.gz](https://github.com/linuxwacom/input-wacom/files/2861973/record_1550088058.tar.gz)

Interestingly, halfway through this, the driver somehow decides to stop interpreting proximity events as mouse presses. This is the 'every now and then' that peter is talking about: Sometimes there's a period the proximity events work as expected, sometimes they get interpreted as mouse clicks for a period of time.

Please tell me if you need more information to fix this.

OberstFredward, 2019-05-20:
Hey, I have the exactly same problem with Kubuntu 18.04 on the Thinkpad X380 Yoga.
I tried so much to solve this annoying problem. To increase the threshold didn't help. I also tried to set the pressure curve insensitive as possible, but also without any success.
I can confirm that the buggy-period is coming and going randomly. Sometimes I help myself with pressing the lower button on my stylus to use the rubber for a short period of time. After this the problem is gone until it appears randomly again. 

All click-events on this log are false/ghost-clicks. I didn't touch the display while I was recording.
[record_1558349024.tar.gz](https://github.com/linuxwacom/input-wacom/files/3197520/record_1558349024.tar.gz)

petergthatsme, 2019-05-20:
For me, the situation is solved by running the particular kernel: 4.15.0-39 along with the input-wacom v2.00-0.42.0. You could try that to see if it helps. While I haven't tried the absolute latest kernel, I've tried a few newer ones (with the updated input wacom) and the situation was present.

jigpu, 2019-05-20:
Thanks @OberstFredward. The logs show no evidence of a click event or pressure at the HID or kernel layers, only at the X11 layer. This would seem to indicate some kind of bug with xf86-input-wacom. I don't see the problem when replaying the logs locally, however, so isolating the exact issue will take a bit more work.

Would you mind running [our sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) to gather some additional information about your hardware and software?

OberstFredward, 2019-05-20:
Updating to the newest version fixed it so far (Build from source). It seems that this is a bug in the kernel-integrated version (v2.00). I really hope this fix is staying. I will continue testing.
I'm using v2.00-0.42.0.33.g00b2d78 with Kernel 4.15.0-48-generic right now.

Here is my result of sysinfo.sh script. Maybe you find out what the reason of this bug was.
[sysinfo.h9vmhBuIxl.tar.gz](https://github.com/linuxwacom/input-wacom/files/3198445/sysinfo.h9vmhBuIxl.tar.gz)

Good job developers!

jigpu, 2019-05-20:
Its very interesting that an update to the kernel driver seems to have worked for you as well... The kernel log looked sane enough but maybe I need to to take a second closer look.

petergthatsme, 2019-05-20:
@OberstFredward  
`It seems that this is a bug in the kernel-integrated version (v2.00). `
i think it has to be more than that, because, at least for me, the updated input-wacom and some of the older kernels that you're using (for example: 4.15.0-42, etc) would still show the same behavior with an input-wacom update. (but yet things work for 4.15.0-39, which is what i've been on for a long time now). I will try to see what I end up with when switching to 4.15.0-48-generic.

jigpu, 2019-05-30:
If either of you have time to gather some additional logs, would you mind trying the following? You'll want to reboot to clear the existing logs, and then save the following script and run it with `sudo bash <filename>`.

The script will try to temporarily switch back to the stock kernel driver and then set up the X driver to log all the events that it receives and sends. Once its ready, try to reproduce the issue. Aftewards, press CTRL+C to restore the normal input-wacom driver and produce a tarball of the logs. You can then attach the generated log here for review.

```bash
#!/bin/bash

if [[ $EUID -ne 0 || -z "$SUDO_UID" || -z "$SUDO_USER" ]]; then
    echo "This script must be run with sudo, and not as root."
    exit 1
fi

OUTFILE_TXZ=$(mktemp --suffix=".tar.xz")
OUTFILE_TXGZ=$(mktemp --suffix=".tar.xz.gz")
STOCK_DRIVER=$(ls /lib/modules/$(uname -r)/kernel/drivers/hid/wacom.ko* | head -n1)

HOME=$(awk -vFS=: -vUID=$SUDO_UID '{if($3==UID){print $6}}' /etc/passwd)
XZ_OPT="-T0 --lzma2=preset=6e,dict=1MiB"
export HOME
export XZ_OPT

finish-debug() {
    echo "Restoring default driver..."
    rmmod wacom && modprobe wacom
    echo "Compressing logs (this may take a few minutes)..."
    (cd $HOME/.local/share/xorg; tar cJf "$OUTFILE_TXZ" *.log)
    gzip -1c "$OUTFILE_TXZ" > "$OUTFILE_TXGZ"
    rm "$OUTFILE_TXZ"
    chown "$SUDO_USER" "$OUTFILE_TXGZ"

    echo
    echo "Done! Please attach $OUTFILE_TXGZ to the bug report for review."
}

echo "Loading stock driver..."
rmmod wacom && insmod "$STOCK_DRIVER" || { echo "Failed to swap driver"; modprobe wacom; exit 1; }

echo "Waiting for devices to settle..."
sleep 3

echo "Configuring devices for debugging..."
xsetwacom list | cut -f1 | while read -r DEV; do
    xsetwacom set "$DEV" ToolDebugLevel 10
    xsetwacom set "$DEV" TabletDebugLevel 10
done

echo
echo "Reproduce bug and press CTRL+C when done"
echo
trap finish-debug EXIT
sleep infinity
```

petergthatsme, 2019-05-30:
i will try to do this on the weekend. 
BTW, I also tried the latest kernel 4.15.0-50-generic with self-built input-wacom v2.00-0.42.0, and I still see this problem occasionally. So, for me anyway, it's really only 4.15.0-39 kernel that never exhibits the issue. 
@OberstFredward after using 4.15.0-48 for a while now, is the problem still gone for you?

petergthatsme, 2019-06-02:
Hey @jigpu attaching the log after running the script:
[logs.tar.xz.gz](https://github.com/linuxwacom/input-wacom/files/3245349/logs.tar.xz.gz)

Luckily the problem showed almost right away, so I didn't have to wait log. 
I also made a video in case it could help:
https://streamable.com/d7md3

jigpu, 2019-06-03:
Thanks for the logs and video! I accidentally didn't have the logging level set quite high enough in my script to dump everything I wanted to see, but between what *was* captured and your video, I'm getting a better idea of what might be going on.

It looks like this might be related to some of the issues we've had in the past with AES pens not immediately sending their serial number. This causes the xf86-input-wacom driver to think that there are two pens present (one with no serial number and a second with a serial number) and not properly track the various state changes that occur (like if the pen is touching the display or not).

We've (theoretically) already fixed this in the input-wacom driver, and the fix shouldn't rely on anything outside of the driver. Now the question is why the behavior still occasionally occurs for you when using any kernel except 4.15.0-39.

Here's a modified version of the script above which increases the log level and which leaves the input-wacom driver loaded. Please try switching "back" to 4.15.0-50-generic, re-installing input-wacom, rebooting to clear out logs, and then finally try to reproduce the issue while the below script is running.

```bash
#!/bin/bash

if [[ $EUID -ne 0 || -z "$SUDO_UID" || -z "$SUDO_USER" ]]; then
    echo "This script must be run with sudo, and not as root."
    exit 1
fi

OUTFILE_TXZ=$(mktemp --suffix=".tar.xz")
OUTFILE_TXGZ=$(mktemp --suffix=".tar.xz.gz")

HOME=$(awk -vFS=: -vUID=$SUDO_UID '{if($3==UID){print $6}}' /etc/passwd)
XZ_OPT="-T0 --lzma2=preset=6e,dict=1MiB"
export HOME
export XZ_OPT

finish-debug() {
    echo "Restoring default driver..."
    rmmod wacom && modprobe wacom
    echo "Compressing logs (this may take a few minutes)..."
    (cd $HOME/.local/share/xorg; tar cJf "$OUTFILE_TXZ" *.log)
    gzip -1c "$OUTFILE_TXZ" > "$OUTFILE_TXGZ"
    rm "$OUTFILE_TXZ"
    chown "$SUDO_USER" "$OUTFILE_TXGZ"

    echo
    echo "Done! Please attach $OUTFILE_TXGZ to the bug report for review."
}

echo "Configuring devices for debugging..."
xsetwacom list | cut -f1 | while read -r DEV; do
    xsetwacom set "$DEV" ToolDebugLevel 11
    xsetwacom set "$DEV" TabletDebugLevel 11
done

echo
echo "Reproduce bug and press CTRL+C when done"
echo
trap finish-debug EXIT
sleep infinity
```

OberstFredward, 2019-06-03:
I used 4.15.0-48 around a week without any problems. A few days ago I updated to 4.15.0-50 and immediately loaded the newest version of input-wacom into the kernel. (I was afraid that this problem will come back, now it's gone I use my Laptop regularly for writing stuff) 
This way the problem stayed fixed. 
Mysteriously the problem also does not occur with your newest script. I tried around 10min to produce the issue. I will try to reinstall 4.15.0-50 and cause the issue without installing the newest version of input-wacom from source the next days.

petergthatsme, 2019-06-03:
@jigpu sure, will try the latest script; although that might take a few days. 

@OberstFredward It's interesting that you're not seeing the issue with 4.15.0-50.
BTW, are you using the pen/stylus that came with your laptop? I use the bigger IBM one that I had to buy separately. Next time i see the problem, will also try to observe if it happens with both pens.

jigpu, 2019-06-04:
@OberstFredward Linux 4.15.0-50 without input-wacom should definitely be buggy since the kernel fix didn't land until 4.16. The fix is also present in input-wacom 0.38.0 and later though, so if you install it with any kernel the issue "should" go away.

I'm only interested in logs with input-wacom installed (or Linux 4.16 in use) since that's the only time I wouldn't expect the bug to occur.

petergthatsme, 2019-06-23:
just a quick update; I've tried reproducing the problem a few times after a restart, but have not been able to. Every now and then the problem happens again but as soon as I restart to clear the logs the problem is gone. BTW, i've tried things with the kernel 4.15.0-52 and the problem is still there. In fact there seems to be a new issue where random lines are now drawn all over the place sometimes, but I'm not sure if this is a driver/kernel problem, or if it could be related to the newest xournalpp app i'm using.

jigpu, 2019-06-27:
Thanks for the update. If the issue goes away after rebooting, lets try running the script immediately after you notice issues. Hopefully the pre-existing logs won't be too confusing to deal with.

petergthatsme, 2019-07-23:
So finally i had some time to capture this again. This time it's with the latest ubuntu 18.04 kernel:
`uname -a
Linux laptop6 4.15.0-55-generic #60-Ubuntu SMP Tue Jul 2 18:22:20 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
and the latest wacom driver:
`modinfo wacom | grep version 
version:        v2.00-0.43.0`

also you can see the video of how things started to go wrong here: 
https://streamable.com/y6ldi

here are the logs:
[bug_report_2019_07_23.tar.xz.gz](https://github.com/linuxwacom/input-wacom/files/3424376/bug_report_2019_07_23.tar.xz.gz)

thanks for looking at this!

jigpu, 2019-07-24:
At first glance it looks like there is an issue with the kernel failing to give a consistent pen serial number (MSC_SERIAL) for the duration of the pen being in prox. This results in the X driver thinking that multiple pens are in prox and can lead to weird behavior like the ghost touches. We wrote a fix for this over a year ago (https://github.com/linuxwacom/input-wacom/commit/650a13c902a297e6e3e2aceafe572aa2be719688), but clearly it isn't working properly in some cases.

I don't see much that could be going wrong after a quick evaluation. About the only thing that seems plausible is the hardware having the "in-range" bit clear for the first few packets (which will cause us to immediately send them out to userspace) and then setting the bit once the serial number is ready, causing the serial number to suddenly appear in the stream. It could be something else entirely though; I'll have to do a bit more investigation...

petergthatsme, 2019-07-25:
ok thans. Let us know once you have an idea how to sort this out - I can test any code updates.

jdchristensen, 2020-05-25:
I also have an X1 Yoga running Ubuntu 18.04 with Gnome, and I can consistently make this problem happen and make it go away.  When freshly started, there is no problem.  To make the problem start, I hold the stylus far from the screen, press the button on the stylus furthest from the tip, and approach the screen.  From this point on, every time the stylus approaches the screen, a spurious touch event happens once I get within range.  To make the problem go away, I hold the stylus far from the screen, press the button on the stylus closest to the tip, and approach the screen.  Now the problem is gone until I start it again with the method described above.

Not sure if this is the same issue.  It seemed random to me at first until I finally figured out how to cause it, and then it took me some time to notice a way to stop it.  If this is not the same issue, should I file a new bug?

```
Linux yogi 4.15.0-101-generic #102-Ubuntu SMP Mon May 11 10:07:26 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

ii  libwacom-bin               0.29-1                 amd64     Wacom model feature query library -- binaries
ii  libwacom-common            0.29-1                 all       Wacom model feature query library (common files)
ii  libwacom2:amd64            0.29-1                 amd64     Wacom model feature query library
ii  xserver-xorg-input-wacom   1:0.36.1-0ubuntu1.1    amd64     X.Org X server -- Wacom input driver
```

jigpu, 2020-06-02:
@jdchristensen would you mind using the [hid-recorder tool](https://bentiss.github.io/hid-replay-docs/) to record the associated raw hardware events? Once you have the tool installed, run `sudo hid-recorder` to get a list of available hidraw devices, and then press CTRL+C and run `sudo hid-recorder <device> [device [...] ] | tee recording.log` with the path to each Wacom device node.

Once the recording is started, trigger the issue as described above and then remove the pen. Bring the pen into prox a second time to exhibit the ghost touch issue, and then finally use your described method to fix the issue. Press CTRL+C when done and attach the recording.log to this bug (you can just drag & drop it onto the comment box).

Hopefully the recording contains some hints about how we can filter these events...

jdchristensen, 2020-06-02:
I just upgraded from Ubuntu 18.04 to 20.04 on this machine, and so far I haven't been able to reproduce the problem, although I haven't used the stylus very much yet.  I'll follow your instructions if I am able to reproduce it.  For now, I'll just report some of the versions I currently have installed.

```
Linux yogi 5.4.0-33-generic #37-Ubuntu SMP Thu May 21 12:53:59 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

ii  libwacom-bin               1.3-2ubuntu1        amd64   Wacom model feature query library -- binaries
ii  libwacom-common            1.3-2ubuntu1        all     Wacom model feature query library (common files)
ii  libwacom2:amd64            1.3-2ubuntu1        amd64   Wacom model feature query library
ii  xserver-xorg-input-wacom   1:0.39.0-0ubuntu1   amd64   X.Org X server -- Wacom input driver
```

jigpu, 2020-06-02:
Thanks. Be sure that the recording captures the issue, since it won't be helpful otherwise. If you have to make a few attempts at triggering/exhibiting/fixing the issue within the recording, that's fine. We should still be able to correlate the touch and pen events.

jcbradfield, 2020-10-03:
I have just (I very rarely use the stylus) run into this, exactly as described by jdchristensen. Thinkpad X1 Yoga, Xubuntu 18.04, 
Linux vis 4.15.0-115-generic #116-Ubuntu SMP Wed Aug 26 14:04:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
<pre>
ii  libwacom-bin   0.29-1       amd64        Wacom model feature query library
ii  libwacom-commo 0.29-1       all          Wacom model feature query library
un  libwacom0      <none>       <none>       (no description available)
ii  libwacom2:amd6 0.29-1       amd64        Wacom model feature query library
un  wacom-tools    <none>       <none>       (no description available)
ii  xserver-xorg-i 1:0.36.1-0ub amd64        X.Org X server -- Wacom input dri
</pre>
Let me know if you still want me to do some of the things you asked jdchristensen to do.

Pinglinux, 2022-08-10:
The kernel driver fix should be in the newer kernel release. Please either update your Linux to a newer version or update input-wacom to the latest version.

closed 2022-08-10 by Pinglinux