linuxwacom/input-wacom #67: Wacom Intuos S BT not in /dev/input/event

danneboom, 2019-01-25:
I access the device in my application using open(..) from fcntl.h on all events in /dev/input/ and check for a wacom device. It works perfectly when connected via USB using the last input-wacom and a udev-rules  file containing:

     ATTRS{idProduct}=="0376",ATTRS{idVendor}=="056a",MODE:="0666",GROUP="plugdev"

But, when connected via Bluetooth, it works in Ubuntu, but open(..) returns -1 for every event. 
The dmesg output and xsetwacom list devices is the same as described here:
https://forum.manjaro.org/t/wacom-intuos-bt-s-wrongly-detected-type-over-bluetooth/61288/3

Any ideas why? is it a bug in the driver or a problem of fcntl.h? i use

     int dev = open("/dev/input/eventX", O_RDONLY|O_NONBLOCK)

in a loop where X is the event number.

jigpu, 2019-01-25:
I believe that tablet should work fine with the current version of input-wacom. Have you checked the `errno` value that gets set when `open` fails? That may give more information about why the call is failing.

In addition to the udev rule you posted for `056a:0376`, do you also have a rule in place for `056a:0377`? The device has a different PID when connecting over Bluetooth and if you don't have a similar udev rule in place then the permissions won't be what you expect.

danneboom, 2019-01-26:
```errno``` is ```EACCES``` resp. "permission denied" for every event. The creation of an additional udev rule or adding a line with ```065a:0377``` does not change anything. My file now looks like

    ATTRS{idProduct}=="0376",ATTRS{idVendor}=="056a",MODE:="0666",GROUP="plugdev"
    ATTRS{idProduct}=="0377",ATTRS{idVendor}=="056a",MODE:="0666",GROUP="plugdev"

but I'm not very familiar with these strange udev rules, so maybe I'm wrong here?
I also find out that on your Wiki page the BT ID ```0377``` requires Kernel 4.17 - can that be the problem? I'm a bit afraid of an update, thats why I still haven't tested it.

jigpu, 2019-01-28:
If you're using the latest input-wacom then it shouldn't be necessary to update to the 4.17 kernel. The input-wacom driver is just a backport of the latest kernel driver for people who can't or don't want to update the whole kernel :)

The udev syntax looks reasonable (the use of `:=` instead of plain `=` is a little weird but should be fine). As long as `ls -l /dev/input/event/*` shows that the mode for the tablet has indeed been updated to 666 and that its in the `plugdev` group then the rule isn't your problem.

I'm not sure what would cause an EACCES error when accessing over Bluetooth rather than USB. Are there any hints in `dmesg` or `journalctl` when you try to run your application?

danneboom, 2019-01-29:
Thanks for pointing this out! It seems like the rule is not applied correctly, as the ```ls -l /dev/input/event* ``` output is for USB the mode 0666:

     crw-rw-rw- 1 root input 13, 78 gen 29 11:34 /dev/input/event14
     crw-rw-rw- 1 root input 13, 79 gen 29 11:34 /dev/input/event15

but for Bluetooth the mode 0660:

    crw-rw---- 1 root input 13, 78 gen 29 11:36 /dev/input/event14
    crw-rw---- 1 root input 13, 79 gen 29 11:36 /dev/input/event15

(the events 14 and 15 are the tablet events). So it seems that the mode has not been updated. Btw, the group change to plugdev seems to fail in both cases, but my program works nevertheless, so it seems not to be necessary.

Any ideas why the rule is not applied? The dmesg output for USB connection:

    [  918.236192] usb 1-1: new full-speed USB device number 11 using xhci_hcd
    [  918.386524] usb 1-1: New USB device found, idVendor=056a, idProduct=0376
    [  918.386530] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [  918.386534] usb 1-1: Product: Intuos BT S
    [  918.386539] usb 1-1: Manufacturer: Wacom Co.,Ltd.
    [  918.386542] usb 1-1: SerialNumber: 8FH00S2007284
    [  918.392206] input: Wacom Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0376.000D/input/input39
    [  918.392690] input: Wacom Intuos BT S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:056A:0376.000D/input/input41
    [  918.393172] wacom 0003:056A:0376.000D: hidraw4: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos BT S] on usb-0000:00:14.0-1/input0

and for Bluetooth:

    [  787.874853] wacom 0005:056A:0377.000C: unknown main item tag 0x0
    [  787.875120] input: Wacom Intuos BT S Pen as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.000C/input/input36
    [  787.875681] input: Wacom Intuos BT S Pad as /devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.000C/input/input38
    [  787.876271] wacom 0005:056A:0377.000C: hidraw4: BLUETOOTH HID v0.00 Device [Intuos BT S] on d4:6d:6d:3d:ef:b8

Thank you very much!

danneboom, 2019-01-29:
update: it works using the following udev rule:

     SUBSYSTEMS=="bluetooth",MODE="0666",GROUP="plugdev"

So, I think the idProduct and idVendor are not detected correctly. The existing rule is probably applied to every bluetooth device. How can we clearly identify the wacom tablet? The ```udevadm info -a /dev/input/event14 ``` output is as follows:

```
Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.001D/input/input87/event14':
    KERNEL=="event14"
    SUBSYSTEM=="input"
    DRIVER==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.001D/input/input87':
    KERNELS=="input87"
    SUBSYSTEMS=="input"
    DRIVERS==""
    ATTRS{name}=="Wacom Intuos BT S Pen"
    ATTRS{phys}=="d4:6d:6d:3d:ef:b8"
    ATTRS{properties}=="1"
    ATTRS{uniq}=="dc:2c:26:d8:8c:4d"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256/0005:056A:0377.001D':
    KERNELS=="0005:056A:0377.001D"
    SUBSYSTEMS=="hid"
    DRIVERS=="wacom"
    ATTRS{country}=="21"
    ATTRS{speed}=="0"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0/hci0:256':
    KERNELS=="hci0:256"
    SUBSYSTEMS=="bluetooth"
    DRIVERS==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0/bluetooth/hci0':
    KERNELS=="hci0"
    SUBSYSTEMS=="bluetooth"
    DRIVERS==""

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14/1-14:1.0':
    KERNELS=="1-14:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="btusb"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="e0"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="01"
    ATTRS{bInterfaceSubClass}=="01"
    ATTRS{bNumEndpoints}=="03"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1/1-14':
    KERNELS=="1-14"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="e0"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bDeviceSubClass}=="01"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="100mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 2"
    ATTRS{bcdDevice}=="0002"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="6"
    ATTRS{devpath}=="14"
    ATTRS{idProduct}=="0aaa"
    ATTRS{idVendor}=="8087"
    ATTRS{ltm_capable}=="no"
    ATTRS{maxchild}=="0"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="3270"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb1':
    KERNELS=="usb1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{authorized_default}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0415"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="1"
    ATTRS{devpath}=="0"
    ATTRS{idProduct}=="0002"
    ATTRS{idVendor}=="1d6b"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Linux 4.15.0-43-generic xhci-hcd"
    ATTRS{maxchild}=="16"
    ATTRS{product}=="xHCI Host Controller"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{serial}=="0000:00:14.0"
    ATTRS{speed}=="480"
    ATTRS{urbnum}=="7519"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0':
    KERNELS=="0000:00:14.0"
    SUBSYSTEMS=="pci"
    DRIVERS=="xhci_hcd"
    ATTRS{broken_parity_status}=="0"
    ATTRS{class}=="0x0c0330"
    ATTRS{consistent_dma_mask_bits}=="64"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{dbc}=="disabled"
    ATTRS{device}=="0xa36d"
    ATTRS{dma_mask_bits}=="64"
    ATTRS{driver_override}=="(null)"
    ATTRS{enable}=="1"
    ATTRS{index}=="3"
    ATTRS{irq}=="124"
    ATTRS{label}=="Onboard - Other"
    ATTRS{local_cpulist}=="0-11"
    ATTRS{local_cpus}=="fff"
    ATTRS{msi_bus}=="1"
    ATTRS{numa_node}=="-1"
    ATTRS{revision}=="0x10"
    ATTRS{subsystem_device}=="0x1214"
    ATTRS{subsystem_vendor}=="0x1462"
    ATTRS{vendor}=="0x8086"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""
```

danneboom, 2019-02-25:
A better version is 
``` 
DRIVERS=="wacom",MODE="0666",GROUP="plugdev"
```
If that's no error (that idVendor and idProduct are not detected), you can close this issue.

jigpu, 2019-03-04:
Interesting! It looks like Bluetooth devices require a match rule with a slightly different format. See [here](https://github.com/linuxwacom/libwacom/blob/master/tools/generate-udev-rules.c#L82) for an example from libwacom. Looking at the generated 65-libwacom.rules file, the following should hopefully work:
~~~
ENV{PRODUCT}=="5/56a/377/*",MODE="0666",GROUP="plugdev"
~~~
I'm not a udev expert, but I don't believe there's any particular problem with using the `DRIVERS` match instead if you want the same permissions to be granted to any Wacom device attached to the system.

closed 2019-03-04 by jigpu