linuxwacom/input-wacom #68: Wacom Intuos not recognized in xsetwacom. Xorg log says driver not specified.

allensutton, 2019-02-01:
I have a Wacom Intuos and the stylus nor the tablet itself are being seen on my Debian 9.0 Linux machine with respect to X.  When I issue an "lsusb" command, I see the Wacom device:

`allen@tornado:~$ lsusb | grep -i wacom
Bus 001 Device 018: ID 056a:0374 Wacom Co., Ltd`

The Wacom module is there:

`allen@tornado:~$ lsmod | grep -i wacom
wacom                  94208  0
hid                   122880  3 hid_generic,usbhid,wacom
`
But xinput doesn't see the device:

 `allen@tornado:~$ xsetwacom --list devices
allen@tornado:~$`

Xorg logs are complaining about a missing driver:

`allen@tornado:~/scripts/lysi/eosMain$ tail -f /var/log/Xorg.0.log
[681513.093] (II) systemd-logind: releasing fd for 13:65
[681513.113] (II) config/udev: Adding input device Wacom Co.,Ltd. Intuos S Pen (/dev/input/event4)
[681513.113] (II) No input driver specified, ignoring this device.
[681513.113] (II) This device may have been added with another device file.
[774540.889] (II) config/udev: Adding input device Wacom Co.,Ltd. Intuos S Pen (/dev/input/event4)
[774541.072] (II) No input driver specified, ignoring this device.
[774541.072] (II) This device may have been added with another device file.
`

But I know that all the necessary libraries are there because I installed them:

`allen@tornado:~$ sudo dpkg -s xserver-xorg-input-wacom | grep Status
Status: install ok installed
allen@tornado:~$ sudo dpkg -s libwacom2 | grep Status
Status: install ok installed
allen@tornado:~$ sudo dpkg -s libwacom-common | grep Status
Status: install ok installed
`
I am assuming that the two lines in the "lsmod" command have something to do with it (I'm guessing they're 'fighting' over it or something to that effect, I'm not sure).  I have Googled around for quite a while but can't seem to find an answer.  I did stumble upon your sysinfo.sh script, which I ran, and I have attached the output, as I am not sure what I am really looking for.  I realize this may be an issue on my end, but I'm at a loss with the sysinfo stuff, so any help would be greatly appreciated.

[sysinfo.zip](https://github.com/linuxwacom/input-wacom/files/2823444/sysinfo.zip)

jigpu, 2019-02-08:
One possibility is that you don't have the necessary udev rules installed for X to recognize your tablet as such. These rules should be installed as part of the "libwacom-common" package and can be found at `/lib/udev/rules.d/65-libwacom.rules`. It looks like Debian 9 uses version 0.22 of libwacom which isn't new enough to have the rules for your particular tablet. You could try installing the version of [libwacom-common for Debian 10](https://packages.debian.org/buster/libwacom-common) which should be new enough to support your device. The libwacom-common package just consists of tablet information files and rules so should be very safe to update to the Debian 10 version.

allensutton, 2019-02-09:
@jigpu I figured it out - when I first tried setting this up, I installed the version of input-wacom from the Debian repo.  That didn't seem to work, so I compiled and installed the latest version from source.  However, the old module was still loaded even though I uninstalled it via aptitude.  This link helped me:
[(https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source#troubleshooting)]

closed 2019-02-09 by allensutton