linuxwacom/input-wacom #69: Everything but eraser works on Nuvision Solo 10 Draw

ShapeShifter499, 2019-02-18:
I don't know much about kennel code. But I'm hoping someone here can help me out.  @jigpu I saw that you made quite a few updates and changes to the kennel driver.  @whot mentioned it could be a driver issue. That whole discussion is here: https://github.com/linuxwacom/libwacom/issues/70

This is the only major issue left for this tablet, everything else works pretty good under Linux.

ShapeShifter499, 2019-02-22:
Any updates on this?

whot, 2019-02-27:
As mentioned in https://github.com/linuxwacom/libwacom/issues/70, the driver used here is `hid-generic`, not the wacom driver. So the input-wacom project is kinda the wrong project here since it focuses exclusively on wacom tablets, sorry.

closed 2019-02-27 by whot

ShapeShifter499, 2019-02-27:
@whot wait, this is a "hid-generic" issue?