linuxwacom/input-wacom #70: CTL-672 not working

IdanWolnnec, 2019-03-04:
I have the Wacom CTL-672 tablet.
my PC is Lenovo IdeaPad G70.
when I used Linux mint 19 my tablet worked, after upgrading to linux mint 19.1 cinnamon 64-bit kernel version 4.15.0.45 it does not work.
when I plug my tablet it says "No tablet detected", even after installing the driver.
does anyone have a solution or a different installation guide?

jigpu, 2019-03-05:
Does the pen work at all, or is this just an issue with the control panel? The control panel will show "no tablet detected" if there are problems finding the device in its database, even if the hardware itself works fine.

Could you also please run our [sysinfo.sh script](https://github.com/linuxwacom/wacom-hid-descriptors/blob/master/scripts/sysinfo.sh) and attach the output to this bug? This will give us many of the relevant logs and package information that will help us pinpoint the issue.

IdanWolnnec, 2019-03-05:
> Does the pen work at all, or is this just an issue with the control panel? The control panel will show "no tablet detected" if there are problems finding the device in its database, even if the [...]

1. Pen does not work (but when the tablet is plugged, indicator LED is working properly)
2. I'm not sure I ran it properly but it saved all the data from it in a .tar.gz file which I will
 [link](https://drive.google.com/open?id=1JOEZzhxWOb15Hk6YJacGmIf2vjuwsd_U)

jigpu, 2019-03-07:
I should have caught this from your original message, but the issue is that your 4.15 kernel does not natively support your tablet. You'll need to install the latest version of this input-wacom driver ([instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source)) to get it working. Note that the driver is tied to the kernel it was compiled with, so it will need to be re-installed any time the kernel is updated (which is likely why it stopped working after your upgrade).

IdanWolnnec, 2019-03-07:
> I should have caught this from your original message, but the issue is that your 4.15 kernel does not natively support your tablet. You'll need to install the latest version of this input-wacom [...]

it does not work, I get no errors during installation but the tablet still does not work

jigpu, 2019-03-07:
Odd... Could you please re-run the sysinfo.sh script and upload its new tarball?

IdanWolnnec, 2019-03-08:
> Odd... Could you please re-run the sysinfo.sh script and upload its new tarball?

[Link](https://drive.google.com/open?id=1JOEZzhxWOb15Hk6YJacGmIf2vjuwsd_U)

jigpu, 2019-03-12:
Thanks. The tablet isn't listed in `lsusb` -- did you have it connected when running sysinfo? It shouldn't make a difference for my interpretation of the logs, but I was just wondering.

The input-wacom driver appears to be loading but the kernel is still failing to set up the device for some reason. I'm away from my debugging workstation for the next several days but I can try to see what happens when I plug a CTL-672 into my own system.

IdanWolnnec, 2019-03-12:
> Thanks. The tablet isn't listed in `lsusb` -- did you have it connected when running sysinfo? It shouldn't make a difference for my interpretation of the logs, but I was just wondering.
>
> The [...]

The tablet was not connected, I tried the script again With the tablet connected to see if there's any different [Link](https://drive.google.com/open?id=1JOEZzhxWOb15Hk6YJacGmIf2vjuwsd_U)

jigpu, 2019-03-22:
Thanks. Your kernel's dmesg log printed out the following message shortly after connecting the tablet:
~~~
Lockdown: Loading of unsigned modules is restricted; see man kernel_lockdown.7
~~~
If the kernel was trying to load the stock kernel driver (without support for the CTL-672) originally and now is trying to load input-wacom but fails because it isn't signed, that could possibly explain the issue. Can you try turning off secure boot in the BIOS? That should disable the kernel's lockdown mode and let the driver loading process get further. If we're lucky the tablet will start working.

IdanWolnnec, 2019-03-24:
> Thanks. Your kernel's dmesg log printed out the following message shortly after connecting the tablet:
>
> ```
> Lockdown: Loading of unsigned modules is restricted; see man kernel_lockdown.7
> ```
>
> If the [...]

that worked, is there any way to re-enable secure boot that still allows the tablet to work?

jigpu, 2019-03-25:
I've just written a guide for our wiki which should explain how to configure your system for [secure boot](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot). Let me know if it makes sense and works for you.

The new module signing code isn't in the current 0.42 release, so once your system is set up for secure boot, you'll want to clone our repository with the following command and then follow the [revised instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source) (which mention how to provide the path to the signing key and cert if necessary).

~~~
$ git clone https://github.com/linuxwacom/input-wacom.git
~~~

jigpu, 2019-04-11:
Did the instructions above help?

IdanWolnnec, 2019-04-11:
> Did the instructions above help?

yes they did, thank you

closed 2019-04-11 by jigpu