linuxwacom/input-wacom #72: Disable touchpad pth451 (intuos pro s) fedora29 i3wm

ahloiscreamo, 2019-03-19:
Hi there,
I'm looking for a way to disable the touchpad on my device (pth451)
I can list all the devices but there is no touch output to disable,
the command above didn't work on my system
terminal output-
```

~ ❯❯❯ xsetwacom list devices
Wacom Intuos Pro S Pen stylus           id: 9   type: STYLUS    
Wacom Intuos Pro S Pad pad              id: 10  type: PAD       
Wacom Intuos Pro S Pen eraser           id: 16  type: ERASER    
Wacom Intuos Pro S Pen cursor           id: 17  type: CURSOR    
~ ❯❯❯ xsetwacom --set "Wacom Intuos Pro S Pad pad" touch off
~ ❯❯❯ 
 
```
Picture for reference 

[image]

`Linux fedora-ahloi 4.20.16-200.fc29.x86_64 #1 SMP Thu Mar 14 15:10:22 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

Thanks

Pinglinux, 2019-03-19:
PAD is for those buttons on the tablet, which we call expresskeys. Touch should be TOUCH, which is not listed in your output. We need to figure out why it's not listed.

Do you see "Finger" or "TOUCH" from dmesg command after disconnect and reconnect the tablet?

ahloiscreamo, 2019-03-20:
From dmesg i can see both input for touch and finger

```
[    2.083293] usb 1-1.3: Product: Intuos5 touch S
[    2.083295] usb 1-1.3: Manufacturer: Wacom Co.,Ltd.
[    2.086775] hid-generic 0003:056A:0314.0002: hiddev96,hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos5 touch S] on usb-0000:00:1a.0-1.3/input0
[    2.087460] hid-generic 0003:056A:0314.0003: hiddev97,hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos5 touch S] on usb-0000:00:1a.0-1.3/input1
[    2.088382] input: Wacom Co.,Ltd. Intuos5 touch S as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.2/0003:056A:0314.0004/input/input3
[    2.088476] hid-generic 0003:056A:0314.0004: input,hidraw3: USB HID v1.10 Mouse [Wacom Co.,Ltd. Intuos5 touch S] on usb-0000:00:1a.0-1.3/input2
[    2.100699] input: Wacom Intuos Pro S Pen as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:056A:0314.0002/input/input4
[    2.100819] input: Wacom Intuos Pro S Pad as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:056A:0314.0002/input/input6
[    2.100982] wacom 0003:056A:0314.0002: hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos5 touch S] on usb-0000:00:1a.0-1.3/input0
[    2.101184] input: Wacom Intuos Pro S Finger as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.3/1-1.3:1.1/0003:056A:0314.0003/input/input8
[    2.101277] wacom 0003:056A:0314.0003: hidraw2: USB HID v1.10 Device [Wacom Co.,Ltd. Intuos5 touch S] on usb-0000:00:1a.0-1.3
```

Pinglinux, 2019-03-20:
I updated my system with kernel 4.20 and tested PTH-451. I see touch. I can disable it by (xsetwacom set "Wacom Intuos Pro S Finger touch" touch off").

From the output of dmesg on my system, I do not see those three lines started with "hid-generic ". Your system is definitely running something different from the upstream kernel at kernel.org.

@bentiss Since it's on Fedora, Benjamin may know something I don't...

bentiss, 2019-03-21:
re hid-generic: that's because by default, fedora doesn't load the wacom driver. So when the tablet gets plugged in, hid-generic catches it first, and then udev loads wacom.ko, which makes hid-generic release the tablet so wacom.ko can handle it.

I have the feeling that the touch part of the tablet is not handled through the xorg-wacom driver, but libinput. I can't be sure, but this would make the more sense. AFAIU, you want to disable the touch device, so that means that it is handled, and so I guess xorg-libinput is handling it.

So I think @whot would know better on how to disable the touch, because whether you are running wayland, or X there will be differences, and I am sure he knows off hand how to solve this :)

(in any cases, this doesn't seem to be a kernel issue, it's either xorg-wacom or libwacom issue IMO)

whot, 2019-03-21:
`xsetwacom` only lists devices handled by the wacom driver, use `xinput list` to list all of them. Most likely, the issue is as benjamin said, it's the libinput driver that handles the touch device. You can either force-assign the wacom driver to it with a custom xorg.conf.d snippet or run `xinput disable <device name>` to disable the touch part.

ahloiscreamo, 2019-03-21:
Thank you very much for the help/guide @whot  @bentiss  @Pinglinux  
I have disabled the touchpad successfully with the command above 

[image]

I will close the issue.

closed 2019-03-21 by ahloiscreamo