linuxwacom/input-wacom #74: CTL-4100WL / Intuos BT S relative pointer movement

ruabmbua, 2019-03-21:
The CTL-4100WL / Intuos BT S works fully via USB, but when connected via bluetooth, the relative pointer movement does not work at all. I am not sure what is going on, sometimes it always moves into the same corner, sometimes it does only not work, when release the pen, and put it onto another screen position.

skomra, 2019-06-10:
Hi ruabmbua,

Sorry for the delay in responding.

First is the absolute pointer movement operating correctly?

Additionally please attach the information from the following script:

https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

Best,
Aaron

skomra, 2019-06-10:
I've reproduced this GNOME bug on Ubuntu 18.04.02 (Gnome 3.28.2). This issue can also be seen on the Intuos Pro 2nd generation (PTH-660).

ruabmbua, 2019-06-11:
Here is my sysinfo: [sysinfo.srbISjs4mq.tar.gz](https://github.com/linuxwacom/input-wacom/files/3277787/sysinfo.srbISjs4mq.tar.gz)

About absolute movement, it seems to work well.
Some more information about my scenario:

I think the pad somehow interprets the distance between the lift of point on the pad to the next drop down point. This movement is the applied to the last position of the cursor. It gets even weirder with multiple monitors (I can not really see a pattern in that case).

TOTON95, 2019-07-01:
@ruabmbua how did you make it work with BT in the first place, I've been installing the latest releases and no success.

ruabmbua, 2019-07-01:
Plugged in, and it worked? Bluetooth was a little bit more challenging, I had to enable the bluetooth daemon,

TOTON95, 2019-07-01:
> Plugged in, and it worked? Bluetooth was a little bit more challenging, I had to enable the bluetooth daemon,

I agree with the USB option, it works even if the Wacom features are not fully displayed, what did you have to enable at the Bluetooth daemon?

ruabmbua, 2019-07-02:
Just bluetoothd. `systemctl start bluetooth; systemctl enable bluetooth;`

I use archlinux, which has a pretty up to date bluetooth stack (bluez 5). Maybe that is your issue?

TOTON95, 2019-07-02:
> I use archlinux, which has a pretty up to date bluetooth stack (bluez 5).

`bluez is already the newest version (5.48-0ubuntu3.1).`

Well at least, I know that is possible, i will work on it once I have some free time, thanks!

jigpu, 2019-12-12:
Should be mostly fixed with xf86-input-wacom 0.38.0. See https://github.com/linuxwacom/xf86-input-wacom/pull/73 and https://github.com/linuxwacom/input-wacom/issues/54 for more information. Jumps may still occur before the pen first touches the tablet, but afterwards they should not happen again.

closed 2019-12-12 by jigpu