linuxwacom/input-wacom #75: CTL-472 not working in gentoo, device detected and functions but driver fails

oscitrus, 2019-03-29:
I've been trying to get my One CTL-472 working for about two straight days and I've all but run out of things to try. I am running Gentoo linux with xfce4. My INPUT_DEVICES in make.conf are evdev and wacom. I am using Portage's existing libinput, evdev, libwacom, xf86-input-wacom, and xf86-input-libinput packages. The device should be compiled correctly, as it is picked up in evemu-record, lsusb, everywhere else. It did work briefly before installing any of the linux wacom packages but X seemed to have considered it a mouse and I need the capabilites provided by xsetwacom. The device does not provide cursor movement and I get no results when running xsetwacom --list devices. The LED does still work correctly.

My tablet seems to not have a correct "type" to apply to it and the module associated with it ends up unloading, and due to failure at this step it can't proceed to run through libinput, X, etc properly. I will post some relevant code results below. Any help would be appreciated.

/var/log/Xorg.0.log:
```
[    10.904] (II) config/udev: Adding input device Wacom Co.,Ltd. CTL-472 Mouse (/dev/input/event13)
[    10.904] (**) Wacom Co.,Ltd. CTL-472 Mouse: Applying InputClass "evdev pointer catchall"
[    10.904] (**) Wacom Co.,Ltd. CTL-472 Mouse: Applying InputClass "evdev tablet catchall"
[    10.904] (**) Wacom Co.,Ltd. CTL-472 Mouse: Applying InputClass "Wacom USB tablet class"
[    10.904] (**) Wacom Co.,Ltd. CTL-472 Mouse: Applying InputClass "Wacom tablet class"
[    10.904] (II) LoadModule: "wacom"
[    10.904] (II) Loading /usr/lib64/xorg/modules/input/wacom_drv.so
[    10.905] (II) Module wacom: vendor="X.Org Foundation"
[    10.905]    compiled for 1.20.3, module version = 0.36.1
[    10.905]    Module class: X.Org XInput Driver
[    10.905]    ABI class: X.Org XInput driver, version 24.1
[    10.905] (II) Build version: 0.36.1
[    10.905] (II) wacom: Driver for Wacom graphics tablets: PenPartner, Graphire, {truncated}
[    10.907] (II) Using input driver 'wacom' for 'Wacom Co.,Ltd. CTL-472 Mouse'
[    10.907] (**) Wacom Co.,Ltd. CTL-472 Mouse: always reports core events
[    10.907] (**) Option "Device" "/dev/input/event13"
[    10.959] (EE) Wacom Co.,Ltd. CTL-472 Mouse: Invalid type 'stylus' for this device.
[    10.959] (EE) Wacom Co.,Ltd. CTL-472 Mouse: Invalid type 'eraser' for this device.
[    10.959] (EE) Wacom Co.,Ltd. CTL-472 Mouse: Invalid type 'cursor' for this device.
[    10.959] (EE) Wacom Co.,Ltd. CTL-472 Mouse: Invalid type 'touch' for this device.
[    10.959] (EE) Wacom Co.,Ltd. CTL-472 Mouse: Invalid type 'pad' for this device.
[    10.959] (EE) Wacom Co.,Ltd. CTL-472 Mouse: No type specified
[    10.973] (EE) PreInit returned 8 for "Wacom Co.,Ltd. CTL-472 Mouse"
[    10.973] (II) UnloadModule: "wacom"
```

lsusb returns one device, always
```
Bus 005 Device 003: ID 056a:037a Wacom Co., Ltd CTL-472 [One by Wacom (S)]
```

cat /proc/bus/input/devices | grep -i wacom
```
N: Name="Wacom Co.,Ltd. CTL-472 Mouse"
```

dmesg
```
[    1.857595] usb 5-1.3: New USB device found, idVendor=056a, idProduct=037a, bcdDevice= 1.00
[    1.857599] usb 5-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.857601] usb 5-1.3: Product: CTL-472
[    1.857604] usb 5-1.3: Manufacturer: Wacom Co.,Ltd.
[    1.857606] usb 5-1.3: SerialNumber: 8BE00L1002242
[    1.866643] input: Wacom Co.,Ltd. CTL-472 Mouse as /devices/pci0000:00/0000:00:1c.3/0000:05:00.0/usb5/5-1/5-1.3/5-1.3:1.0/0003:056A:037A.0001/input/input13
[    1.866885] hid-generic 0003:056A:037A.0001: input,hiddev96,hidraw0: USB HID v1.10 Mouse [Wacom Co.,Ltd. CTL-472] on usb-0000:05:00.0-1.3/input0
[    1.871118] hid-generic 0003:056A:037A.0002: hiddev97,hidraw1: USB HID v1.10 Device [Wacom Co.,Ltd. CTL-472] on usb-0000:05:00.0-1.3/input1
```

libinput list-devices
```
event13 - Wacom Co.,Ltd. CTL-472 Mouse: libinput bug: missing tablet capabilities: xy pen btn-stylus resolution. Ignoring this device.
```

grep "" /sys/module/wacom*/version
```
zsh: no matches found: /sys/module/wacom*/version
```

modinfo wacom | grep version
```
modinfo: ERROR: Module wacom not found.
```

For the previous two this might just be because of the module unload? I ran both while the tablet was plugged in. lsmod also returns nothing wacom-related.

jigpu, 2019-03-29:
Can you ensure that you've included the Wacom driver in your kernel configuration? If you have a /proc/config.gz file, try running `zgrep CONFIG_HID_WACOM /proc/config.gz`

If nothing is printed out you'll need to enable the driver by running `make menuconfig` inside the kernel sources and navigating to "Device Drivers --> HID support --> Special HID drivers" and then setting the "Wacom Intuos/Graphire tablet support (USB)" to either be a part of the kernel or a module. Rebuild/reinstall the kernel, and if you decided to have the Wacom driver be in a module, be sure that the wacom.ko module gets installed.

If "CONFIG_HID_WACOM=m" is printed out, you will need to make sure that the wacom.ko module has been installed alongside the other kernel modules.

If "CONFIG_HID_WACOM=y" is printed out, we'll probably need to do a bit more debugging since the driver should be built into the kernel and thus no reason for it to use the "hid-generic" driver instead.

oscitrus, 2019-03-29:
Thanks so much for the quick response and sorry for taking so long to get back to you. I do have that kernel option compiled and not as a module so it should definitely be getting loaded... I recompiled and no change. I followed the kernel options from the Handbook (https://wiki.gentoo.org/wiki/Wacom). I used to have all of the options enabled but I currently only have the one you mentioned. 

Interestingly, wacom and wacom_w8001 appear as options in modprobe. "modprobe wacom" prints nothing and "rmmod wacom" says that the module is builtin and can't be removed. "modprobe wacom_w8001" says "could not insert: Device or resource busy" while "rmmod wacom_w8001" says "not currently loaded"... very strange.

zgrep CONFIG_HID_WACOM /proc/config.gz just says that I have no such file or directory, it might not exist in Gentoo and/or my installed packages.

I'll see if I can't mess with the packages a bit, removing something would cause X to break and I would need to fix it by re-emerging from the black terminal tty. The main packages of concern are "xf86_input_wacom," "libwacom," "libinput," and "xf86-input-libinput." I don't think the libinput packages should be important; I originally had libinput in my make.conf before switching to evdev thinking that that might do something. The xf86-input-libinput package might be completely useless since I can remove it freely and it doesn't seem to do anything. I'll try to remove libinput altogether, I think it's getting pulled in by some dependencies.

oscitrus, 2019-03-30:
Another update: the Wacom works again with none of the linux wacom drivers or packages applied. It as treated as a mouse as expected. lsusb returns one device as before. The results from rmmod and modprobe for wacom and wacom_w8001 are the same...

```
[    12.060] (II) config/udev: Adding input device Wacom Co.,Ltd. CTL-472 Mouse (/dev/input/event13)
[    12.060] (**) Wacom Co.,Ltd. CTL-472 Mouse: Applying InputClass "evdev pointer catchall"
[    12.060] (II) Using input driver 'evdev' for 'Wacom Co.,Ltd. CTL-472 Mouse'
[    12.060] (**) Wacom Co.,Ltd. CTL-472 Mouse: always reports core events
[    12.060] (**) evdev: Wacom Co.,Ltd. CTL-472 Mouse: Device: "/dev/input/event13"
[    12.112] (--) evdev: Wacom Co.,Ltd. CTL-472 Mouse: Vendor 0x56a Product 0x37a
[    12.112] (--) evdev: Wacom Co.,Ltd. CTL-472 Mouse: Found 9 mouse buttons
[    12.112] (--) evdev: Wacom Co.,Ltd. CTL-472 Mouse: Found relative axes
[    12.112] (--) evdev: Wacom Co.,Ltd. CTL-472 Mouse: Found x and y relative axes
[    12.112] (II) evdev: Wacom Co.,Ltd. CTL-472 Mouse: Configuring as mouse
[    12.112] (**) evdev: Wacom Co.,Ltd. CTL-472 Mouse: YAxisMapping: buttons 4 and 5
[    12.112] (**) evdev: Wacom Co.,Ltd. CTL-472 Mouse: EmulateWheelButton: 4, EmulateWheelInertia: 10, EmulateWheelTimeout: 200
[    12.112] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:1c.3/0000:05:00.0/usb5/5-1/5-1.3/5-1.3:1.0/0003:0$
[    12.112] (II) XINPUT: Adding extended input device "Wacom Co.,Ltd. CTL-472 Mouse" (type: MOUSE, id 8)
[    12.112] (II) evdev: Wacom Co.,Ltd. CTL-472 Mouse: initialized for relative axes.
[    12.112] (**) Wacom Co.,Ltd. CTL-472 Mouse: (accel) keeping acceleration scheme 1
[    12.112] (**) Wacom Co.,Ltd. CTL-472 Mouse: (accel) acceleration profile 0
[    12.112] (**) Wacom Co.,Ltd. CTL-472 Mouse: (accel) acceleration factor: 2.000
[    12.112] (**) Wacom Co.,Ltd. CTL-472 Mouse: (accel) acceleration threshold: 4
```

oscitrus, 2019-03-30:
> Can you ensure that you've included the Wacom driver in your kernel configuration? If you have a /proc/config.gz file, try running `zgrep CONFIG_HID_WACOM /proc/config.gz`
>
> If nothing is printed out [...]

Deleted my previous two comments since they were useless. I recompiled my kernel to include all of the Wacom-related options as per the Gentoo Handbook. I made the main "Intuos/Graphire" part module instead of built-in but the result is the same as before. The device is detected in lsusb and "wacom" appears as a selectable object in modprobe so it should be compiled correctly...? I don't have a /proc/config.gz file.

libwacom-list-local-devices displays the device but xsetwacom does not and never has. Evemu-record has always worked. 

Whenever I try to "modprobe wacom" I get the error below and can not reactive.
```
[  325.941285] wacom: Unknown symbol devm_of_led_classdev_register (err -2)
[  325.941341] wacom: Unknown symbol devm_led_trigger_register (err -2)
[  325.941364] wacom: Unknown symbol led_trigger_event (err -2)
```

I honestly have no idea what to do from here.

oscitrus, 2019-03-31:
Fixed. I am not sure if any of the things I messed around with did anything but uname -v revealed that all of my new kernels with wacom compiled were never being run. A simple copy from /boot to /boot/efi fixed it...

closed 2019-03-31 by oscitrus

jigpu, 2019-04-01:
Ah, glad to hear it! I was trying to come up with ways the device wouldn't show up with an built-in driver, and the options weren't pretty :D

A couple of things I should mention for future reference:
* Enabling the kernel option [described here](https://superuser.com/a/538110) will enable the `/proc/config.gz` that shows the kernel's compile-time config. Very handy for sanity-checking that the running kernel is configured how you think.
* The "xf86-input-libinput" X11 driver (and the "libinput" library it relies on) is a successor to the older "xf86-input-evdev" X11 driver. Both should work fine for keyboards/mice/etc, but I'm not sure if the evdev driver is still being developed.
* The "libwacom" library isn't strictly necessary for a tablet to move the pointer around. It provides additional device information to control panels, but isn't used by any of the drivers (at least, not yet).

oscitrus, 2019-04-01:
Yeah I ended up enabling `/proc/config.gz`, seeing that I didn't have it after recompiling my kernel was how I figured out that my kernel wasn't updating correctly. I can't remember if xf86-input-libinput was ever getting pulled in (I don't think it was). I don't remember when I got xf86-input-evdev but I do have it so maybe evdev pulled it in. Portage seems to always pull in libwacom when setting `INPUT_DEVICES=wacom` and updating @world along with xf86-input-wacom.

fluentpwn, 2022-12-24:
Hello! I have the same problem, but I am on Artix Linux and not gentoo:

`sudo zgrep WACOM /proc/config.gz` shows
```
CONFIG_TABLET_SERIAL_WACOM4=m
CONFIG_TOUCHSCREEN_WACOM_W8001=m
CONFIG_TOUCHSCREEN_WACOM_I2C=m
CONFIG_HID_WACOM=m
```
And the module seems to not be loaded on startup. I don't really know where to go from here. Thanks for any help!