linuxwacom/input-wacom #76: CTL-672 Ubuntu 16.04 freezing/kernel panic

4l1fe, 2019-03-31:
I've bought one recently and hoped everything will go well =)

Yesterday i installed drivers on my laptop as decribed here  [installing](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source)

There were no problems for first time. But today i got ones:
1. Ubuntu loading freeze
2. Ubuntu freeze after plugging in a tablet if it was unplugged
3. Ubuntu freeze after touching a tablet by a pen if plugging in was successful
All freeze times capslock button flashed(it means kernel panic i think)

What can i do to solve it? 
What debugging info should i provide here?

skomra, 2019-06-10:
Hi 4l1fe, 

Sorry for the delay in responding.

Please run the sysinfo script[1] and upload a link to the resulting file here.

[1] https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

Best,
Aaron

rajvinoo-ctrl, 2019-09-05:
Even i am facing the same issue with CTL-672 , input-wacom-0.42/0.43, kernel 3.10.0-957.21.3.el7.x86_64.. module being used is hid_wacom. 
During reboot, if the wacom pad is connected it results in kernel panic. But otherwise it is working fine.

skomra, 2019-09-05:
Hi rajvinoo-ctrl

Please run the sysinfo script[1] and upload (or link to) the resulting file here.

[1] https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

rajvinoo-ctrl, 2019-09-06:
[sysinfo.f2SEMSQ1Hw_af_inp_wcm_43.tar.gz](https://github.com/linuxwacom/input-wacom/files/3582992/sysinfo.f2SEMSQ1Hw_af_inp_wcm_43.tar.gz)
[sysinfo.p40lJOlZu2_bef_apply_inp_wcm.tar.gz](https://github.com/linuxwacom/input-wacom/files/3582993/sysinfo.p40lJOlZu2_bef_apply_inp_wcm.tar.gz)

One is before installing input-wacom and another is after. Also one more thing to be noted is, after i configure and make install the same, after every restart i had to "modprobe -r hid_wacom ; modprobe hid_wacom, and it works well. To avoid doing this after every reboot if i give "dracut -f" it is resulting in kernel panic, if the tablet is connected during reboot.Your help is much appreciated on this. Thanks much.

skomra, 2019-09-06:
Let me repeat what you've said, just to make sure I've got everything.

1. Centos 7.6 has no out of box support for the CTL-672.
2. You install input-wacom and start it with modprobe. The tablet works.
3. You restart the machine, the driver is no longer loaded.
4. You use dracut to avoid having to use modprobe after each reboot.
5. You get a kernel panic with the dracut modified kernel on reboot.

Did I leave anything out?

rajvinoo-ctrl, 2019-09-06:
yes.. exactly....

skomra, 2019-09-06:
Have you used dracut for any other modules (besides wacom) on your system?

rajvinoo-ctrl, 2019-09-06:
yes.. tried that.. but only this is causing panic as per my knowledge..

rajvinoo-ctrl, 2019-09-06:
And if my memory serves right, same issue we had got for EL7.4 and only version 37 was stable(all upstream version resulting in panic during reboot), but unfortunately that is popping compile errors for EL7.6, tried the patch suggested by jigpu for EL7.5(https://github.com/linuxwacom/input-wacom/issues/23) and it was unsuccessful..

skomra, 2019-09-06:
[image]

skomra, 2019-09-06:
HI rajvinoo-ctrl,

I've reproduced the issue and I'm working on it. It is tricky to work on, though, so it may take some time.

rajvinoo-ctrl, 2019-09-07:
Thanks much Skomra... Greatly appreciated..

rajvinoo-ctrl, 2019-09-12:
Hi Skomra, 
Also tried the below for testing..
1. Built an RPM with ver 43
2. Installed and rebooted. Started working as expected. (Anyways reboot resulted in panic)
3. To avoid panic, blacklisted the module and observed no panic.(But the device was not working)
4. Setup a cron to insmod @reboot, to make the device to work.
5. Sadly, 4 out of 7 reboots still lands in panic. But kernel seems to load till log in screen and when the cron is invoked it panics :(

Didn't mean to urge you but is there any workaround you can think of ? Thanks..

4l1fe, 2019-09-13:
> Hi 4l1fe,
> 
> Sorry for the delay in responding.
> 
> Please run the sysinfo script[1] and upload a link to the resulting file here.
> 
> [1] https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh
> 
> Best,
> Aaron

4l1fe, 2019-09-13:
sry closed casually. I'll send info soon

4l1fe, 2019-09-15:
@skomra 
here you go [sysinfo.DtkaUHm5nB.tar.gz](https://github.com/linuxwacom/input-wacom/files/3613707/sysinfo.DtkaUHm5nB.tar.gz)

skomra, 2019-09-16:
Hi 4l1fe,

I see from the libwacom logs that it lists your PC as a tablet PC with a Wacom sensor built-in (Name=Wacom Serial Tablet WACf004). I just want to confirm with you that your device does/does not have a built in pen sensor. That might be a factor in troubleshooting your issue. 

Are you able to catch any logs or crash information with the tablet attached? I don't see "672" in any of the logs.

4l1fe, 2019-09-18:
>I see from the libwacom logs that it lists your PC as a tablet PC with a Wacom sensor built-in (Name=Wacom Serial Tablet WACf004).

I'll try to find out what it is

>I just want to confirm with you that your device does/does not have a built in pen sensor

my laptop hasn't any extra sensors. It has a touchpad only

>Are you able to catch any logs or crash information with the tablet attached? I don't see "672" in any of the logs.

I got info with an unplugged in tablet coz it can freeze my OS. If i have to do it anyway let me know.
And i don't have an idea where logs are placed. I'll search it

KishoreVignesh, 2019-10-10:
I get the same issue, bought a Wacom CTL-4762 yesterday.

$ uname -a
Linux kishore-xxxx 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

With default drivers, the device dint work. 

so i installed Kernel Driver,X Driver and libwacom from the latest releases.

if I connect the tablet system freezes with capslock blinking. 

I tried to fetch the Linux kernel logs. since it crashes I am not able to get the complete output. I will attach what i was able to see in the screen

[image]

KishoreVignesh, 2019-10-10:
output of [1] 
https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh
[sysinfo.1O52vlXyYQ.tar.gz](https://github.com/linuxwacom/input-wacom/files/3711733/sysinfo.1O52vlXyYQ.tar.gz)

[sysinfo.1O52vlXyYQ.tar.gz](https://github.com/linuxwacom/input-wacom/files/3711733/sysinfo.1O52vlXyYQ.tar.gz)

User18172, 2019-10-18:
Hello! Same issue. If I connect tablet (CTL-672) with loaded module system with a high probability freezes. Or when I do modprobe wacom when module not loaded and tablet connected. Is it possible to fix this error? Because the kernel cannot be updated due to many dependencies. Thanks.

user@laptop:/tmp$ uname -a
Linux laptop 3.19.0-80-generic #88~14.04.1-Ubuntu SMP Fri Jan 13 14:55:00 UTC 2017 i686 i686 i686 GNU/Linux

[sysinfo.B07FMOeoNx.tar.gz](https://github.com/linuxwacom/input-wacom/files/3745794/sysinfo.B07FMOeoNx.tar.gz)

skomra, 2019-10-22:
[image]

Keeping notes here - Attaching the tablet after system boot gives the above trace.

jigpu, 2019-10-23:
I'm not sure why *both* `wacom_init_work` and `wacom_reset_resume` are in the stack trace, but I notice that both functions do basically the same thing except that **one** of the two grabs a mutex lock. I wonder if adding the same mutex locks to the other function would solve this...

~~~
diff --git a/3.17/wacom_sys.c b/3.17/wacom_sys.c
index 8aaaa7e6e5a9..539a7cfd6448 100644
--- a/3.17/wacom_sys.c
+++ b/3.17/wacom_sys.c
@@ -1487,8 +1487,12 @@ static void wacom_init_work(struct work_struct *work)
 {
 	struct wacom *wacom = container_of(work, struct wacom, init_work.work);
 
+        mutex_lock(&wacom->lock);
+
 	_wacom_query_tablet_data(wacom);
 	wacom_led_control(wacom);
+
+        mutex_unlock(&wacom->lock);
 }
 
 static void wacom_query_tablet_data(struct wacom *wacom)
~~~

skomra, 2019-10-23:
I was hopeful, but that didn't seem to work...

PeganovAnton, 2019-11-12:
Hello!

I have two laptops with Ubuntu 16.04 and I installed the driver on them slightly differently. 

On the first laptop, I followed instructions from [stackoverflow](https://stackoverflow.com/questions/51878250/wacom-one-pen-tablet-driver-in-ubuntu-18-04)  but  replaced the last command with
```
sudo -s
apt-get update && apt-get install xserver-xorg-input-wacom
```
Such the installation caused my system freezing after tablet plugin, sometimes CapsLock blinking.
The issue was fixed with the removal of the file `/usr/share/unity-wacom-panel.desktop`.

On the second laptop, I installed the driver solely with
```
sudo apt-get install linux-headers-$(uname -r) build-essential 
sudo apt-get install autoconf  
if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"
```
and there were no issues. The tablet works correctly.

skomra, 2019-11-25:
More notes: 

Upstream commit 6ecfe51b4082e7cff12609f97daa052ae04b61fd ("Input: refuse to register absolute devices without absinfo" introduced in v4.11) could be relevant for this issue and
https://github.com/linuxwacom/input-wacom/issues/138 where the device is not working, but does not crash the kernel.

skomra, 2019-12-04:
Hi @User18172  @4l1fe @rajvinoo-ctrl @KishoreVignesh can you test the following patch to make sure I've isolated the issue. Note that this is intended for newer Centos 7 (they have two wacom drivers) or kernels 3.17 and later.

```
git clone https://github.com/linuxwacom/input-wacom.git
cd input-wacom
git remote add bugfix https://github.com/skomra/input-wacom.git
git checkout -b bugfix_wacom_one bugfix/bugfix_wacom_one
```

Then build and install the driver according to the directions on our wiki:
https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

Then reboot and test with the tablet. Note: for some reason I had to reboot twice for the system to use the updated driver.

User18172, 2019-12-05:
Hi Skomra. I compiled the module from source, then plugged module in, then plugged and unplugged the tablet about 5 times. Then when the tablet is connected, about 15 times unloaded and loaded the module and everything works fine. Thanks!

jigpu, 2019-12-09:
It looks like the `wacom_parse_and_register` function fails in the `/* pen only Bamboo neither support touch nor pad */` check for one of this tablet's interfaces. When we `goto fail_quirks` to clean up after ourselves (and free the various resources we've created along the way), we never cancel the delayed work that was started with the call to `wacom_query_tablet_data`.

The following change fixes the issue for me, although I haven't tested it super thoroughly. I'm still not yet sure why the crash only occurs on boot and not when plugging in later...
~~~
diff --git a/3.17/wacom_sys.c b/3.17/wacom_sys.c
index 8aaaa7e..06832e7 100644
--- a/3.17/wacom_sys.c
+++ b/3.17/wacom_sys.c
@@ -2272,6 +2272,7 @@ static int wacom_parse_and_register(struct wacom *wacom, bool wireless)
        return 0;
 
 fail_quirks:
+       cancel_delayed_work_sync(&wacom->init_work);
        hid_hw_stop(hdev);
 fail_hw_start:
 fail_remote:
~~~

For future reference, installing the "kernel-debug" package on CentOS 7 and running the installed debug kernel showed the following WARNING at the moment the kernel normally crashes (and even continues to boot too!). Might be a good tool to keep in our back pocket...
~~~
[    9.120287] ------------[ cut here ]------------
[    9.120294] WARNING: CPU: 2 PID: 2026 at lib/debugobjects.c:286 debug_print_object+0x86/0xb0
[    9.120299] ODEBUG: free active (active state 0) object type: timer_list hint: delayed_work_timer_fn+0x0/0x20
[    9.120301] Modules linked in: cfg80211 btusb hid_wacom(OE+) btrtl btbcm uvcvideo btintel rtsx_pci_ms bluetooth memstick videobuf2_vmalloc videobuf2_memops videobuf2_core videodev snd_hda_codec_realtek wmi snd_hda_codec_generic snd_hda_codec_hdmi thinkpad_acpi rfkill snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_seq snd_seq_device snd_pcm pcc_cpufreq snd_timer mei_me snd lpc_ich mei i2c_i801 soundcore ip_tables xfs libcrc32c sd_mod crc_t10dif crct10dif_generic rtsx_pci_sdmmc mmc_core i915 crct10dif_pclmul crct10dif_common crc32c_intel serio_raw rtsx_pci i2c_algo_bit iosf_mbi drm_kms_helper ahci syscopyarea libahci sysfillrect sysimgblt fb_sys_fops drm libata e1000e video ptp pps_core drm_panel_orientation_quirks dm_mirror dm_region_hash dm_log dm_mod
[    9.120364] CPU: 2 PID: 2026 Comm: systemd-udevd Tainted: G           OE  ------------   3.10.0-1062.9.1.el7.x86_64.debug #1
[    9.120367] Hardware name: LENOVO 20B6006DUS/20B6006DUS, BIOS GJET94WW (2.44 ) 09/14/2017
[    9.120368] Call Trace:
[    9.120375]  [<ffffffffafc704ae>] dump_stack+0x19/0x1b
[    9.120380]  [<ffffffffaf4ac8d8>] __warn+0xd8/0x100
[    9.120385]  [<ffffffffaf4ac95f>] warn_slowpath_fmt+0x5f/0x80
[    9.120391]  [<ffffffffaf829126>] debug_print_object+0x86/0xb0
[    9.120394]  [<ffffffffaf4d7990>] ? work_on_cpu+0xc0/0xc0
[    9.120398]  [<ffffffffaf82a58b>] debug_check_no_obj_freed+0x21b/0x270
[    9.120406]  [<ffffffffaf94e0b7>] ? release_nodes+0x1d7/0x220
[    9.120410]  [<ffffffffaf67841b>] kfree+0xdb/0x2d0
[    9.120415]  [<ffffffffaf94e0b7>] release_nodes+0x1d7/0x220
[    9.120421]  [<ffffffffaf94e20c>] devres_release_all+0x3c/0x60
[    9.120425]  [<ffffffffaf949e48>] driver_probe_device+0xf8/0x3e0
[    9.120429]  [<ffffffffaf94a213>] __driver_attach+0x93/0xa0
[    9.120433]  [<ffffffffaf94a180>] ? __device_attach+0x50/0x50
[    9.120437]  [<ffffffffaf9478c5>] bus_for_each_dev+0x75/0xc0
[    9.120442]  [<ffffffffaf94976e>] driver_attach+0x1e/0x20
[    9.120445]  [<ffffffffaf9491f8>] bus_add_driver+0x208/0x2e0
[    9.120450]  [<ffffffffc0b5e000>] ? 0xffffffffc0b5dfff
[    9.120453]  [<ffffffffaf94a8c4>] driver_register+0x64/0xf0
[    9.120458]  [<ffffffffafa75c28>] __hid_register_driver+0x68/0xb0
[    9.120466]  [<ffffffffc0b5e01e>] wacom_driver_init+0x1e/0x1000 [hid_wacom]
[    9.120470]  [<ffffffffaf40210a>] do_one_initcall+0xba/0x240
[    9.120476]  [<ffffffffaf553900>] load_module+0x19c0/0x1f10
[    9.120479]  [<ffffffffaf839bb0>] ? ddebug_proc_write+0x110/0x110
[    9.120487]  [<ffffffffaf54e903>] ? copy_module_from_fd.isra.43+0x53/0x150
[    9.120494]  [<ffffffffaf554056>] SyS_finit_module+0xa6/0xd0
[    9.120502]  [<ffffffffafc87a9e>] system_call_fastpath+0x25/0x2a
[    9.120509] ---[ end trace f3e58965b158962d ]---
~~~

jigpu, 2019-12-12:
Fix should now be in the master branch of input-wacom. A new release with the fix should be coming Very Soon™.

closed 2019-12-12 by jigpu

4l1fe, 2020-03-19:
> Hi @User18172 @4l1fe @rajvinoo-ctrl @KishoreVignesh can you test the following patch to make sure I've isolated the issue. Note that this is intended for newer Centos 7 (they have two wacom drivers) [...]

sry for a late reply. I've updated a kernel version to 5.3. That is my workaround.