linuxwacom/input-wacom #77: Wacom module will not load (CTL-472)

Aretno, 2019-04-03:
Hi, 

I have a Wacom One (CTL-472) and I've been trying to install the input-wacom drivers (last version, 0.42) in Ubuntu 18.04. Apparently the installation went fine, no errors when using: `$ if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed" `. However, the tablet is not recognized, neither after installation nor after rebooting.

So, to check that the module was loaded i tried:`$ grep "" /sys/module/wacom*/version` which showed that there was not such a directory. However, `modinfo wacom | grep version` works with the following output:
```
version:        v2.00-0.42.0
srcversion:     0EB0B6DCD9EBBD5F87C11F3
```
In both cases I have the tablet connected to the computer.

Finally, I tried to force the reload of the modules and while the modprobe of wacom_8001 works just fine, the wacom one (`sudo modprobe wacom`) gives the following error:

`modprobe: ERROR: could not insert 'wacom': Required key not available`

Any help would be highly appreciated, thanks!

jigpu, 2019-04-03:
It sounds like your system may have secure boot enabled, which will prevent the module from being loaded until it is signed by a trusted key. If you run `dmesg | grep lockdown` do you see the following message: "Lockdown: Loading of unsigned modules is restricted; see man kernel_lockdown.7"? Our [Secure Boot wiki article](https://github.com/linuxwacom/input-wacom/wiki/Secure-Boot) should be of use if this is the case.

Note that the current release of our driver (0.42) doesn't support module signing, so you'll need to build the development version of our driver. Run `git clone https://github.com/linuxwacom/input-wacom.git` to get a copy and then follow the same [build instructions](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source#build--install).

Aretno, 2019-04-05:
Ok, that's what it was, it worked like a charm, thanks!!

closed 2019-04-05 by Aretno

ArnabGh16May, 2023-01-27:
> 

With the same problem mentioned above, I'm getting this message after typing `dmesg | grep lockdown` : `dmesg: read kernel buffer failed: Operation not permitted. 
`
Should I have to follow that article or I have to do something else?

theGeekyLad, 2023-11-08:
Try running that with `sudo`.