linuxwacom/input-wacom #78: Pen on Dell 7390 2-1 keeps disconnecting

lukyluket, 2019-04-11:
I am working with Ubuntu 18.04. My wacom setup is
Wacom HID 4841 Pen stylus       	id: 11	type: STYLUS    
Wacom HID 4841 Finger touch     	id: 12	type: TOUCH     
Wacom HID 4841 Pen eraser       	id: 21	type: ERASER   

Whenever I write, the pen stops working after a while. The only way to get it working again, is by removing the battery and putting it back in.

Sometimes touching with the fingers does the job.
Could you have a look?
Which information should I provide?
Regards

LaserTron, 2020-09-18:
I have the exact same problem!!! I wrote about it here:
https://forum.manjaro.org/t/pen-on-pc-tablet-works-unreliably/24639/2

Pinglinux, 2022-08-05:
This is a system integration issue, which is beyond input-wacom's scope.

closed 2022-08-05 by Pinglinux