linuxwacom/input-wacom #81: OLED brightness on intuos4

PrzemoF, 2019-04-21:
I'm not sure if this is the right place to report the problem - please close it if it's not.

OLEDs display luminance/brightness on Intuos4 no longer works as expected. There is no way to set the brightness to above "barely visible" level. I was tweaking the value directly using:

`echo 15 >
/sys/bus/hid/drivers/wacom/0003\:056A\:00BC.0001/wacom_led/buttons_lumi
nance`

15 is max, 16 goes to black again, so the problem is probably in the way kernel is handling the values.

jigpu, 2019-04-22:
I don't seem to have the issue over here. Setting the brightness to 15 results in an image that seems to be as bright as I remember it ever being. Has this recently changed for you? What distribution and kernel version are you using, and what version of input-wacom are you using (if any -- it shouldn't necessary for the Intuos4 :wink:)

PrzemoF, 2019-04-22:
Hmmm.. It's fedora 29
`$ uname -a
Linux localhost.localdomain 5.0.7-200.fc29.x86_64 #1 SMP Mon Apr 8 15:40:59 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
$ rpm  -qi libwacom | grep Vers
Version     : 0.32
`

I'll bring my work intuos4 to make sure the device is OK. I didn't use it for a while.

PrzemoF, 2019-04-23:
I tested 2 identical devices connected over usb and bluetooth and I'm getting inconsistent results. Please close the issue for now. If I can prove there is problem I'll reopen it.

closed 2019-04-23 by PrzemoF