linuxwacom/input-wacom #82: I don't receive any answer to this request

Simo190, 2019-04-27:
When i give this line "grep "" /sys/module/wacom*/version" i receive this ansewer: sys/module/wacom* directory doesn't exist.
In my folder /sys/module I can see the directory wacom_w8000, but this appear after the reboot and when I type the command sudo modprobe wacom_w8001
Could someone help me?

jigpu, 2019-04-29:
Please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and then attach the generated tarball by dragging it into the comment box here.

Also, could you describe what you are trying to do? Do you have a USB-connected Wacom tablet that you are trying to get working? Or are you trying to get a (very) old tablet PC working as suggested by the mention of "wacom_w8001"?

Simo190, 2019-04-29:
I was trying to install the Wacom tablet on xubuntu 18, I remove this OS and install Ubuntu 16.04. After all gone ok!

closed 2019-04-29 by Simo190