linuxwacom/input-wacom #83: Are all of the tablet pc listed on this site use usb-hid touchscreen?

Jamesxxx1997, 2019-05-18:
Hi , thank you for your detailed list about tablet PC.
I'm curious whether all the tablet pc list on this site is usb-hid touchscreen or not(i2c-touchscreen?).
Thank you!!! 
[image]

jigpu, 2019-05-18:
The tablets in your screenshot (0x90, 0x93, 0x97, 0x9a, 0x9f, and 0xe2) are all USB connected. Not all of the devices in the entire "Tablet PCs" list are USB connected, however: some use an I2C connection instead. The https://github.com/linuxwacom/wacom-hid-descriptors/ repository has more detailed information on various tablet PCs, including the bus they connect over.

For instance, as can be seen at https://github.com/linuxwacom/wacom-hid-descriptors/tree/master/HP%20EliteBook%20x360%201030%20G2/sysinfo.IK9qVzgtmZ, the complete device ID for the "HP EliteBook x360 1030 G2" is "0018:056A:481A". The "0018" at the beginning means "I2C" ("0003" would mean "USB").

Also note that a few of the devices in the wacom-hid-descriptors repository don't actually use Wacom sensors. We keep the information anyway for reference, however. Wacom devices will have a vendor ID of 0x056A or 0x2D1F.

Jamesxxx1997, 2019-05-19:
Wow!!! Sincerely very very thankful to you for this detailed explaination !!!

If the tablet pc is usb-hid touchscreen , with stylus support

- without install any touchscreen driver , can the touchscreen sense single touch and stylus , act like a mouse when I use operating system other than win10?

- Some tablet pc has multiple Wacom Device ID (like thinkpad x1 yoga first gen) , does it makes any difference ?

Thank you very ~ very much!!!
James

jigpu, 2019-05-20:
Most Linux distributions include the "wacom.ko" kernel driver which provides support for pen and touch input (as well as the associated "xf86-input-wacom" userspace driver). If you were to install Ubuntu 19.04, for example, both the pen and touchscreen for most devices with Wacom sensors should work. Multi-touch is also supported, but using it for more than gestures (right-click, scroll, zoom) requires some configuration.

Oftentimes a single tablet PC model will have several minor variations. For instance, a manufacturer may get equivalent displays from multiple sources or offer a slightly different screen (e.g. higher-resolution, or OLED instead of LCD) as an upgrade option. Each variation usually gets its own device ID. The multiple IDs do not matter to the driver, however, since it just uses the HID descriptor. The "libwacom" library, on the other hand, **does** do device ID matching and so may prevent the control panel from allowing you to configure the device. Adding new device IDs to libwacom is very straightforward though, so just let us know if you have control panel issues.

Jamesxxx1997, 2019-05-20:
Ok ~~~ I got it~
Thank you for so detailed reply XD.

closed 2019-05-20 by jigpu