linuxwacom/input-wacom #84: mouse wheel don't work with qt program

moebius1, 2019-06-07:
Bonjour,
Mouse wheel of my graphire 2 usually work but I suspect it's recognized like a usual mouse, not a wacom one.
When using frescobaldi, very nice frontend for lilypond score editor, mouse wheel don't work.
ok, some outputs :
**cat /var/log/Xorg.0.log**
 9.331] (II) config/udev: Adding input device PC Speaker (/dev/input/event4)
[     9.332] (II) No input driver specified, ignoring this device.
[     9.332] (II) This device may have been added with another device file.
[     9.337] (**) Wacom Graphire2 4x5 Pen eraser: Applying InputClass "libinput tablet catchall"
[     9.337] (**) Wacom Graphire2 4x5 Pen eraser: Applying InputClass "Wacom USB tablet class"
[     9.337] (**) Wacom Graphire2 4x5 Pen eraser: Applying InputClass "Wacom tablet class"
[     9.337] (II) Using input driver 'wacom' for 'Wacom Graphire2 4x5 Pen eraser'
[     9.337] (**) Wacom Graphire2 4x5 Pen eraser: always reports core events
[     9.337] (**) Option "Device" "/dev/input/event3"
[     9.337] (**) Option "Type" "eraser"
[     9.337] (--) Wacom Graphire2 4x5 Pen eraser: maxX=10206 maxY=7422 maxZ=511 resX=80000 resY=80000  tilt=disabled
[     9.365] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:02.0/usb5/5-2/5-2:1.0/0003:056A:0011.0001/input/input5/event3"
[     9.365] (II) XINPUT: Adding extended input device "Wacom Graphire2 4x5 Pen eraser" (type: ERASER, id 10)
[     9.366] (**) Wacom Graphire2 4x5 Pen eraser: (accel) keeping acceleration scheme 1
[     9.366] (**) Wacom Graphire2 4x5 Pen eraser: (accel) acceleration profile 0
[     9.366] (**) Wacom Graphire2 4x5 Pen eraser: (accel) acceleration factor: 2.000
[     9.366] (**) Wacom Graphire2 4x5 Pen eraser: (accel) acceleration threshold: 4
[     9.366] (**) Wacom Graphire2 4x5 Pen cursor: Applying InputClass "libinput tablet catchall"
[     9.366] (**) Wacom Graphire2 4x5 Pen cursor: Applying InputClass "Wacom USB tablet class"
[     9.366] (**) Wacom Graphire2 4x5 Pen cursor: Applying InputClass "Wacom tablet class"
[     9.366] (II) Using input driver 'wacom' for 'Wacom Graphire2 4x5 Pen cursor'
[     9.367] (**) Wacom Graphire2 4x5 Pen cursor: always reports core events
[     9.367] (**) Option "Device" "/dev/input/event3"
[     9.367] (**) Option "Type" "cursor"
[     9.388] (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:02.0/usb5/5-2/5-2:1.0/0003:056A:0011.0001/input/input5/event3"
[     9.388] (II) XINPUT: Adding extended input device "Wacom Graphire2 4x5 Pen cursor" (type: CURSOR, id 11)
[     9.390] (**) Wacom Graphire2 4x5 Pen cursor: (accel) keeping acceleration scheme 1
[     9.390] (**) Wacom Graphire2 4x5 Pen cursor: (accel) acceleration profile 0
[     9.390] (**) Wacom Graphire2 4x5 Pen cursor: (accel) acceleration factor: 2.000
[     9.390] (**) Wacom Graphire2 4x5 Pen cursor: (accel) acceleration threshold: 4

libwacom-list-local-devices
[Device]
Name=Wacom Graphire2 4x5
DeviceMatch=usb:056a:0011;
Class=Graphire
Width=5
Height=4
IntegratedIn=
Styli=0xfffff;0xffffe;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=false
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=0
--------------

So, if some have an idea for a workaround...
cordialement,

moebius1, 2019-06-07:
Just want to add that frescobaldi is a python/qt program

skomra, 2019-06-07:
Hi moebius1, 

It would help us to know which environment (Operating System/Kernel/Wayland or Xorg?) you were previously using where the wheel did work, and the environment where the wheel no longer works. Also does the wheel work with non QT programs?

Additionally please attach the information from the following script:

https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

moebius1, 2019-06-07:
Bonsoir,

Thank you for your answer :

I'm running Debian Buster with xorg.

Previously, some months ago , I was running ubuntu 18.04 and everything 
worked fine at this time.

Since this Debian (Mate), everything work fine except with frescobaldi 
where wacom mouse wheel doen't work ; but, with a real usb mouse, wheel 
is working in frescobaldi.

The tablet is an old graphire2 with mouse and pen.

I've attached the script result run in root mode for modinfo access.

cordialement,

skomra, 2019-06-07:
Hi moebius1,

Thanks for the system info. Does the wheel work with non QT programs? For example, will it scroll a gnome-terminal window (that is full output).

Best,
Aaron

moebius1, 2019-06-08:
Bonjour,

I'm running MATE and terminal is a fork of gnome one : Mate terminal
Wheel scroll is OK in terminal ; the wheel problem appears only with frescobaldi but, as I have a big use of frescobaldi these times, it's really boring...
So :
- Everything work fine except with frescobaldi
- but, with a real usb mouse, wheel is working well in frescobaldi.

cordialement,

skomra, 2019-06-11:
I've reproduced the bug in this QT app as well as a few others that state that they are written in QT. I reproduced using the Graphire as well as with the Intuos Pro first generation.

This is most likely a bug in QT.

moebius1, 2019-06-11:
Thank you for your effort,
Do you think you could do something ? If you think that it's a QT problem, could you report the problem to the QT team ? You are, for sure, more competent than me to do this, believe me ! 

cordialement,

moebius1, 2019-06-14:
Bonjour,
I read again your former answer and something look strange to me : a real mouse wheel work well in frescobaldi ; so, if only a wacom wheel don't work, it's perhaps because the wacom wheel message sent to the system is not the same that the message sent by a real mouse wheel, isn't it ?
But I have to confess that my knowledge of that kind of things is very low...
cordialement,

jigpu, 2019-06-18:
Looks like several others have reported similar issues with Qt: see [QTBUG-47727](https://bugreports.qt.io/browse/QTBUG-47727). Searching their bug tracker for "wacom" shows that the toolkit seems to have a lot of problems with the tablet mouse, probably because it doesn't appear to the system exactly like a pen (e.g., its normally in relative mode and doesn't have the "stylus" or "eraser" type associated with it).

You can join in on the bug to report your own experiences, but there doesn't seem to have been any kind of response from Qt in the three years its been open...

closed 2019-06-18 by jigpu