linuxwacom/input-wacom #85: links to protocol descriptions are broken

reuvenpo, 2019-06-08:
The links under:

https://github.com/linuxwacom/input-wacom/wiki/Wacom-Protocols

... lead to removed pages in the old sourceforge wiki. I can't find any equivalent information online.

Can you please update those links if the relevant information is still available?

skomra, 2019-06-10:
Hi reuvenpo,

There is an archive of the final state of the old wiki available at: https://github.com/jigpu/linuxwacom-wiki-archive/tree/master/wiki . Those pages contain very old information so I did not transfer them over when I migrated the wiki.

I have updated the broken links that you point out with their equivalent page from the archive.

Best,
Aaron

closed 2019-06-10 by skomra

reuvenpo, 2019-06-10:
Oh alright, i see. Thanks :)