linuxwacom/input-wacom #89: Bamboo only hovers

polbrian, 2019-06-19:
My Wacom Bamboo 16fg 4x5 tablet is recognized, touch, pen and eraser move the cursor on teh display, but it only hovers, does not draw or tap, and stylus buttons have no effect.

[sysinfo.cdU5mECorR.tar.gz](https://github.com/linuxwacom/input-wacom/files/3305248/sysinfo.cdU5mECorR.tar.gz)

jigpu, 2019-06-19:
Please run the [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh). If the script is missing any dependencies, it should let you know what to install. Otherwise, choose your stylus device from the list, try to draw a few lines when the recording starts, and then press CTRL+C to stop recording. Afterwards, drag-and-drop the generated tarball onto the comment box here to attach it for review.

polbrian, 2019-06-19:
Done. Had to care of other businesses in the meanwhile, and to download a few dependencies.
[record_1560968058.tar.gz](https://github.com/linuxwacom/input-wacom/files/3307420/record_1560968058.tar.gz)

skomra, 2019-06-20:
I wasn't able to reproduce this issue on a clean install of OpenSuse 15.1 with the CTH-470. I tried with both Wayland and Xorg.

jigpu, 2019-06-20:
Buttons are present in the kernel log, but not the X log, so whatever is the problem is definitely in userspace somewhere. Could you try running `xsetwacom set <stylus id> threshold 26` to reset the click threshold to default?

polbrian, 2019-06-21:
Tablet name is CTH-470.
`~> xsetwacom set "Wacom Bamboo 16FG 4x5 Pen stylus" Threshold 26`
to no avail. Tried values as low as 1 with no effect.

polbrian, 2019-06-21:
> I wasn't able to reproduce this issue on a clean install of OpenSuse 15.1 with the CTH-470
I updated from Leap 15.0 which was a clean install.

polbrian, 2019-06-22:
Sorry about the inconvenience, I opened a graphical interface as root, the display default 640x480 instead of 1920x1080, but the tablet is OK. I have no idea of where the problem comes from, but clearly it's not from the kernel.

Thank you for your advice and help.

closed 2019-06-22 by polbrian