linuxwacom/input-wacom #92: CTL-472 not working

krishnshyam, 2019-06-21:
I've been trying to get my Wacom One to work on my Ubuntu 14.04 unsuccessfully for a couple of days now. When I installed input-wacom this time and rebooted, I got an error message on reboot saying the driver failed installing. The tablet doesn't work - I can't control the cursor with the tab.

`modinfo wacom | grep version` returns:
```
version:        v1.53-0.43.0
srcversion:     03C6A57E7A78532D898690A
vermagic:       3.13.0-170-generic SMP mod_unload modversions retpoline
``` 
And `grep "" /sys/module/wacom*/version` returns: `v1.53-0.43.0`

However, `xsetwacom --list devices` returns nothing.

I ran the `sysinfo.sh` and have attached the tar.gz file here.

[sysinfo.76o6PwJ6IB.tar.gz](https://github.com/linuxwacom/input-wacom/files/3315316/sysinfo.76o6PwJ6IB.tar.gz)

What should I do?

Thanks!

skomra, 2019-06-21:
Hello,

I tested on a new 14.04.06 system (that I installed kernel 3.13 on) with input-wacom-0.43 and the device worked. 

Clean 14.04.06 ships with kernel 4.4 so it looks like you have a 14.04/14.04.01 system that you've upgraded through each of the point releases. I've never upgraded a system through all of its point releases so I can only guess as to what might be the issue here. 

From your sysinfo files I see that the kernel driver has been updated but that xsetwacom doesn't see the device. Something may have gone wrong with your installed xf86-input-wacom driver package. 

Did you use apt to install any xserver-xorg-input-wacom* packages or install xf86-input-wacom from source?

Could you issue the command 'apt list --installed | grep wacom' and paste the results here?

Also running the [capture script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh) will confirm that is an X driver issue.

Best,
Aaron

krishnshyam, 2019-06-22:
Yeah so I think I went a little overboard with the installs, I'd first installed xf86-input-wacom from source, at that time, grep "" /sys/module/wacom*/version returned nothing, though the modinfo command did. I'd then installed xf86-input-wacom-trusty from apt.

These are the packages from apt list:
libwacom-common/trusty,now 0.8-1 all [installed]
libwacom2/trusty,now 0.8-1 amd64 [installed]
wacom-dkms/trusty,now 0.30.0-1 all [installed]
xserver-xorg-input-wacom/trusty,now 1:0.31.0-4 amd64 [installed]
xserver-xorg-input-wacom-lts-trusty/trusty-updates,now 3:6 amd64 [installed]

The capture scripts returns this:
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech Unifying Device. Wireless PID:4055       id=8    [slave  pointer  (2)]
⎜   ↳ Logitech Unifying Device. Wireless PID:400d       id=9    [slave  pointer  (2)]
⎜   ↳ PS/2 Generic Mouse                        id=11   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=10   [slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                    id=12   [slave  keyboard (3)]

ID number of the device to record:
```

Thanks,
Shyam

skomra, 2019-06-24:
First let me say that since we're not sure about all of the things that you may or may have not installed, the quickest way to solve this issue might be to start with a clean install and then install input-wacom-0.43 from source via the instructions here:

https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

That said, if you really want to try to fix this system:

It looks like wacom-dkms might be another version of the wacom kernel driver, it may also install an X driver, I don't know what's in that package. I would start by removing wacom-dkms since it looks like you already have input-wacom-0.43 installed. I don't think you'll need an updated xf86-input-wacom driver for this tablet.

Also, the above mentioned capture script will let us see if where the events aren't being propagated.

krishnshyam, 2019-06-27:
Ok, it's working now and I'm not sure how.

First I ran make clean on both the input-wacom as well as the xf86 driver.

Then I realised I'd added a usbhid:quirks to the device in the grub config in the beginning since some youtube video had suggested it, I removed that.

Then I uninstalled wacom-dkms, but also went ahead and uninstalled libwacom2, which removed the gnome-desktop and left me locked out unable to login until I figured out what went wrong.

After reinstalling gnome-desktop, and simply plugging in the tab, it started working like nothing had happened!

I'm not sure what made things work, but things seem sorted out for now :)

Thank you!

jigpu, 2019-06-27:
Could you please re-run sysinfo.sh and attach its output?

krishnshyam, 2019-06-30:
Hi,

I did a bit more sleuthing around.

So what seems to happen is when I install the xserver-xorg-input-wacom from apt-get, the pad stops working. This is the only breaking portion of the whole thing. I've attached the sysinfo output both when the xorg package is installed:
[sysinfo.U7t7iRLEs5-with_xorg.tar.gz](https://github.com/linuxwacom/input-wacom/files/3342711/sysinfo.U7t7iRLEs5-with_xorg.tar.gz)
and when it's not (tablet works fine):
[sysinfo.nWD5CbBPlm-without_xorg.tar.gz](https://github.com/linuxwacom/input-wacom/files/3342712/sysinfo.nWD5CbBPlm-without_xorg.tar.gz)

I've currently built and installed all three components of the linuxwacom packages, and the pad works, but "xsetwacom list" returns nothing. The sysinfo output is currently:
[sysinfo.NgdMxqM3YR_final.tar.gz](https://github.com/linuxwacom/input-wacom/files/3342715/sysinfo.NgdMxqM3YR_final.tar.gz)

However, currently the xinput output is:
```
⎡ Virtual core pointer                          id=2    [master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer                id=4    [slave  pointer  (2)]
⎜   ↳ Logitech Unifying Device. Wireless PID:4055       id=8    [slave  pointer  (2)]
⎜   ↳ Logitech Unifying Device. Wireless PID:400d       id=9    [slave  pointer  (2)]
⎜   ↳ Wacom Co.,Ltd. CTL-472                    id=10   [slave  pointer  (2)]
⎜   ↳ PS/2 Generic Mouse                        id=12   [slave  pointer  (2)]
⎣ Virtual core keyboard                         id=3    [master keyboard (2)]
    ↳ Virtual core XTEST keyboard               id=5    [slave  keyboard (3)]
    ↳ Power Button                              id=6    [slave  keyboard (3)]
    ↳ Video Bus                                 id=7    [slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard              id=11   [slave  keyboard (3)]
    ↳ ThinkPad Extra Buttons                    id=13   [slave  keyboard (3)]
```

and running "xinput list-props 10" outputs:
```
Device 'Wacom Co.,Ltd. CTL-472':
        Device Enabled (135):   1
        Coordinate Transformation Matrix (137): 1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
        Device Accel Profile (263):     0
        Device Accel Constant Deceleration (264):       1.000000
        Device Accel Adaptive Deceleration (265):       1.000000
        Device Accel Velocity Scaling (266):    10.000000
        Device Product ID (252):        1386, 890
        Device Node (253):      "/dev/input/event5"
        Evdev Axis Inversion (267):     0, 0
        Evdev Axes Swap (269):  0
        Axis Labels (270):      "Rel X" (145), "Rel Y" (146)
        Button Labels (271):    "Button Left" (138), "Button Middle" (139), "Button Right" (140), "Button Wheel Up" (141), "Button Wheel Down" (142), "Button Horiz Wheel Left" (143), "Button Horiz Wheel Right" (144), "Button Side" (256), "Button Extra" (257)
        Evdev Middle Button Emulation (272):    0
        Evdev Middle Button Timeout (273):      50
        Evdev Third Button Emulation (274):     0
        Evdev Third Button Emulation Timeout (275):     1000
        Evdev Third Button Emulation Button (276):      3
        Evdev Third Button Emulation Threshold (277):   20
        Evdev Wheel Emulation (278):    0
        Evdev Wheel Emulation Axes (279):       0, 0, 4, 5
        Evdev Wheel Emulation Inertia (280):    10
        Evdev Wheel Emulation Timeout (281):    200
        Evdev Wheel Emulation Button (282):     4
        Evdev Drag Lock Buttons (283):  0
```

Thanks!

jigpu, 2019-07-01:
Thanks for the updated output :) It looks like what's going on is that your CTL-472 is being handled by the kernel's "hid-generic" driver rather than the necessary "wacom" driver. This results in a device that isn't compatible with xf86-input-wacom (which Ubuntu calls xserver-xorg-input-wacom). Removing xserver-xorg-input-wacom allows a generic Xorg driver to take over. The generic driver is good enough to move the pointer but xsetwacom and the GNOME Control Center won't work.

You previously had the "input-wacom" driver installed, but it seems to be missing now. That's probably why the "hid-generic" driver is being used at the moment. Installing it should fix the kernel issue that's going on and allow you to configure the tablet through xsetwacom/GNOME. Here's a quick link to the instructions: https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

krishnshyam, 2019-07-08:
Ok things got better then worse now everything seems kinda fine.

Installing the input-wacom seemed to do nothing. I then uninstalled all three linux-wacom components. I uninstalled gimp, which removed a lot of dependencies and reinstalled a new version by snap. Reinstalling the driver after this somehow made this work, I saw the stylus and eraser in the xsetwacom list.

However, the mode seems to be set to Absolute by default, previously, whenever the cursor worked it would be in Relative. I wanted to change it to Relative and tried modifying the 70-wacom.conf. I wasn't very sure how to modify it, and I added the Option "Mode" "Relative" line to all the device categories listed there (this was under /etc/X11/. There is another of the same under /usr/share/X11 which I didn't modify). Right After doing this and restarting, the tablet started crashing the system the moment I brought the stylus to the pad. The only thing I can do when this happens is to force reboot, which is a pain. I looked around and apparently this has happened to many people before (https://www.linuxquestions.org/questions/linux-hardware-18/touching-wacom-tablet-crashes-linux-4175571470/). I was thinking to patch my kernel with [this](https://www.linuxquestions.org/questions/blog/tpalmerstudios-1027157/kernel-patch-for-wacom-tablet-crashes-system-36891/) which seemed to be the solution everyone suggested. However I found a workaround where I just let the tablet sit after connecting, without bringing the stylus near, and if I use the pad after a while it seems to work OK. I also change the Mode using

Currently this is what I'm doing. Is the patch the proper way to deal with this issue or is there any other way to deal with it? I'm also a little scared to change things now, since everything's sorta working OK.

I've attached the current sysinfo output for good measure.
[sysinfo.T4G2R6LsCu.tar.gz](https://github.com/linuxwacom/input-wacom/files/3368563/sysinfo.T4G2R6LsCu.tar.gz)

Thanks!

jigpu, 2019-08-16:
My guess is that the patch is probably the correct solution, but you're right that there is a chance applying it could cause further problems. Given that Ubuntu 14.04 is past its end-of-life and unsupported (at least without paying Canonical), my recommendation would be to either stick with your current workaround or to install a newer version of Ubuntu. If you install Ubuntu 18.10 or 19.04 it should have a new enough kernel to support your particular tablet out of the box.

closed 2019-08-16 by jigpu