linuxwacom/input-wacom #93: Lenovo Yoga 720 - Pen working Finger not - both devices found

lu-kors, 2019-06-27:
Hello everybody. 
I have tried several weeks now to get the (finger) touchscreen working but with no luck. I hope you can help me:  
I have a Lenovo Yoga 720 which was shipped with Windows. Pen + Finger worked fine there. Then I moved to Ubuntu - only Pen worked out of the box. Now I am using fedora and have the same issue (as ubuntu) here. 

Some information for you: 
```
$ modinfo wacom | grep version
version:        v2.00
srcversion:     ADC0B7EDD3A6C0630B4C57F
```
```
$ xsetwacom --list devices
Wacom HID 50FD Pen stylus       	id: 15	type: STYLUS    
Wacom HID 50FD Finger touch     	id: 16	type: TOUCH     
Wacom HID 50FD Pen eraser       	id: 21	type: ERASER    

```
```
$ xinput test 16
motion a[0]=5920 a[1]=2612 

```
The output above was there about an hour later, where I touched the touchscreen several times in between. I cannot reproduce it - it stays blank most of the time. 
If I do the same with id 16 there is a lot output in a few seconds after I wrote some stuff with the pen. 

```
$ libwacom-list-local-devices 
[Device]
Name=Wacom ISDv4 50fd
DeviceMatch=i2c:056a:50fd;
Class=ISDV4
Width=14
Height=8
IntegratedIn=Display;System;
Styli=0xfffff;0xffffe;

[Features]
Reversible=false
Stylus=true
Ring=false
Ring2=false
Touch=true
TouchSwitch=false
StatusLEDs=
NumStrips=0
Buttons=0
---------------------------------------------------------------
```
```
$ ls -l /dev/input/by-path | grep i2c
lrwxrwxrwx. 1 root root 10 27. Jun 17:15 pci-0000:00:15.1-platform-i2c_designware.1-event -> ../event13
lrwxrwxrwx. 1 root root 10 27. Jun 17:15 pci-0000:00:15.1-platform-i2c_designware.1-event-mouse -> ../event12
lrwxrwxrwx. 1 root root  9 27. Jun 17:15 pci-0000:00:15.1-platform-i2c_designware.1-mouse -> ../mouse2
lrwxrwxrwx. 1 root root 10 27. Jun 17:15 pci-0000:00:15.2-platform-i2c_designware.2-event-mouse -> ../event11
lrwxrwxrwx. 1 root root  9 27. Jun 17:15 pci-0000:00:15.2-platform-i2c_designware.2-mouse -> ../mouse1

```
```
$ sudo evtest /dev/input/event13
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x56a product 0x50fd version 0x100
Input device name: "Wacom HID 50FD Finger"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max    13820
      Resolution      40
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     7772
      Resolution      40
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max    13820
      Resolution      40
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     7772
      Resolution      40
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
```
(no output during touching the screen with event12 (PEN) there is again pagewise)

Please let me know if you need more information. I am really desperate right now - any good idea or a approach is more then welcome!

jigpu, 2019-06-28:
Could you please run our [sysinfo.sh script](https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh) and attach the generated tarball to this issue? (Just drag+drop it onto the comment box).

Also, please run my [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/af36c0a0fc2c019750472443369915ad7ea97737/capture.sh) to collect a synchronized set of hardware, kernel, and X events. Choose your touch device and then drag your finger across the screen two or three times. Press CTRL+C when done and attach the generated tarball to this bug as well.

Also, please run `sudo dnf install hid-replay` to install the HID recorder software and then run my [capture.sh script](https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh) to record a synchronized set of hardware, kernel, and X events from the touch device. The script should let you know if there are any missing tools besides the HID recorder. After

lu-kors, 2019-06-29:
Thanks @jigpu the response.
sysinfo:
[sysinfo.B0HTlTZQyI.zip](https://github.com/linuxwacom/input-wacom/files/3341697/sysinfo.B0HTlTZQyI.zip) 
Some small note: I replaced my real username with a pseudonym in journalctl.log for privacy reasons. (and had to use .zip because github did not allow .tar.xz files)
caputure
[record_1561798396.tar.gz](https://github.com/linuxwacom/input-wacom/files/3341698/record_1561798396.tar.gz)

I hope it helps.

jigpu, 2019-07-01:
Thanks! It looks like the hardware isn't sending any events, which is unfortunately going to be a bit difficult to debug. A few things to try:

* If you have Windows installed on this machine, I would recommend rebooting into it temporarily both to check if the hardware works and because sometimes the Windows driver can tickle the hardware into working within Linux.
* If your Yoga has an "emergency reset hole" on the bottom, try shutting down the computer and then pressing the button inside. This is equivalent to unplugging a desktop or temporarily removing the battery from a laptop and should ensure all the hardware is fully reset on the next boot.
* Try switching between UEFI and Legacy boot if you have the option.
* If BIOS updates are available for your system, you might try applying them and seeing if they have an effect.

lu-kors, 2019-07-08:
Thanks for your reply. Sorry I answered a bit late. 
To your suggestions: 

- Windows: Yes I have right now (new installation) but also there it is not working right now (prob. because I installed default windows drivers instead of the ones from the manufacturer. I tried with installing the manufacturer driver afterward but with no success. The phone support told me they would also 'only' reinstall the windows system (as first try).
- Emergency reset hole: I think I found it - I will post a update as soon as I tried it. 
- I had difficulties to boot up Linux outside of Legacy iirc (as long as the first time windows was there) afterward I had not tried it I think. 
- Bios is on the newest Version. I did the Update on Windows a few weeks ago. 
Thank you very much for your support @jigpu - I will keep you updated

Pinglinux, 2022-08-05:
No news is good news?

closed 2022-08-05 by Pinglinux

lukas-staab, 2022-08-10:
No news is no news - it did not work, but I did not try deeper, I arranged myself with a non-working touch screen.

Pinglinux, 2022-08-11:
I googled "Lenovo yoga 720 touch screen not working". I got quite a few links. Since we don't have a Lenovo Yoga 720 to test with, it hard for us to figure out the root cause. If you have time, you may try the latest Fedora, which is at 36 now to see if it makes a difference.

Serathos, 2023-06-02:
I still have this problem on Ubuntu 20.04 and 22.04.

- Tried some of the procedures found on askubuntu.com [(HWE kernel](https://askubuntu.com/questions/1303753/ubuntu-20-04-touchscreen-doesnt-work-on-lenovo-ideapad-flex-5) and [Input Wacom)](https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source), but nothing worked.
- The emergency hole only brought up a dialog giving a selection between normal startup, boot menu, bios and system recovery; I also held this button and the power button for ~20sec, because I remembered that working on a different laptop for a full reset once, but I didn't notice any difference.
- Touchscreen worked on Windows the last time I had that installed. Replaced it with 22.04, didn't work -> went back to 20.04, also doesn't work (every time on a fresh install, so I didn't change anything else, which may cause this).

Is there still hope, should I create a new issue or do I just have to switch to Windows or accept a non-functioning screen?

lukas-staab, 2023-06-04:
@Serathos as the issue creator i would reopen the issue for you if i could. 
If you have enough motivation an knowlede to dig through  @Pinglinux or someone else here might be able to help you. I was lacking the motivation - and now I dont have a Yoga anymore (it droped and broke the screen) - so i switched to a thinkpad ...

Gusfit1, 2024-10-29:
Hey I see this is an old issue, just dropping by while digging into this on my fresh install. 

My YOGA 720-15IKB has touch screen finger support out of the box with Pop! OS and EndeavourOS, but I've just noticed with EndeavourOS that only tapping to click and dragging the cursor around works. No hold to right click, and I can't do things like drag windows around - it's a one or the other sort of deal. I've also tried a tap and hold combination but that doesn't help. I never had Pop! installed long enough to test past tapping, but I might boot back into the live environment to check the rest of it.

I've only just started looking around so maybe my answer is out there, but I'll drop my system info here since it seems relevant:

[sysinfo.OuwxgcY1Xv.tar.gz](https://github.com/user-attachments/files/17557818/sysinfo.OuwxgcY1Xv.tar.gz)