linuxwacom/input-wacom #94: CTL-472 not detected

Sodium-Aluminate, 2019-06-28:
# Useless story that you can skip
I bought a CTL-472, then link it to my laptop, then I found that it doesn't work.
After Google I use `isusb`(return Wacom) `xsetwacom --list devices`(return nothing)
Then I went to [wiki](https://github.com/linuxwacom/xf86-input-wacom/wiki/xorg.conf.d), copy the [file](https://github.com/linuxwacom/xf86-input-wacom/blob/master/conf/70-wacom.conf) to `/etc/X11/95-wacom.conf`.
it **works**, and I enjoy [it](https://github.com/ppy/osu/) for weeks.

# The main idea what I want to say
But after an update, my CTL-472 was not detected **again**. idk why or if it is because of this update.
And I have nothing to do with this.
so,
*What can I do* or ***what kinds of logs should I give***?

# something maybe helpful

```
sodiumaluminate@ThinkPad-E570:~$ xsetwacom --list devices
sodiumaluminate@ThinkPad-E570:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 5986:2109 Acer, Inc 
Bus 001 Device 006: ID 2717:ff88  
Bus 001 Device 002: ID 056a:037a Wacom Co., Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
sodiumaluminate@ThinkPad-E570:~$ 
```

Sodium-Aluminate, 2019-06-28:
By the way, i found `E: 仓库 “http://ppa.launchpad.net/doctormo/wacom-plus/ubuntu bionic Release” 没有 Release 文件。` when apt update.
I'm not sure what it means (both the chinese and English words).

skomra, 2019-06-28:
Hi Sodium-Aluminate,

Please run the sysinfo and capture scripts so that we can take a look.

https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh

Best,
Aaron

Sodium-Aluminate, 2019-06-28:
> Hi Sodium-Aluminate,
>
> Please run the sysinfo and capture scripts so that we can take a look.
>
> https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh [...]

```
sodiumaluminate@ThinkPad-E570:~/Downloads/Cache$ sudo ./sysinfo.sh
Gathering system and tablet information. This may take a few seconds.
  * General host information...
  * Kernel driver information...
  * Kernel device information...
     - /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:056A:037A.0003...
     - /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:056A:037A.0004...
     - udev...
  * Unbinding devices...
  * Rebinding devices...
  * Userspace driver information...
  * Userspace device information...
  * Device display information...
  * System logs...
  * System config files...
  * Tarball generation...
Finished. Data available in '/tmp/sysinfo.lSPPbXvuEc.tar.gz'
sodiumaluminate@ThinkPad-E570:~/Downloads/Cache$ sudo ./capture.sh
The following utilities are missing and must be installed to capture logs:
  * evemu-record [Try: `apt-get install evemu-tools` or `yum install evemu-record`]
  * hid-recorder [Try: https://bentiss.github.io/hid-replay-docs/ ]
sodiumaluminate@ThinkPad-E570:~/Downloads/Cache$ sudo apt install evemu-tools
```

......

```
sodiumaluminate@ThinkPad-E570:~/Downloads/Cache$ sudo ./capture.sh
The following utilities are missing and must be installed to capture logs:
  * hid-recorder [Try: https://bentiss.github.io/hid-replay-docs/ ]
```

May I install hid-play on Ubuntu 18.04?
it said *hid-replay may (will) harm your kernel* ... *you need to have a at least a 3.6 kernel* but **I don't know anything about kernel**, so sorry.

Sodium-Aluminate, 2019-06-28:
[sysinfo.lSPPbXvuEc.tar.gz](https://github.com/linuxwacom/input-wacom/files/3341081/sysinfo.lSPPbXvuEc.tar.gz)
sorry for my Chinses-translated system, this translations looks terrible. 
xd

skomra, 2019-07-01:
Hi Sodium-Aluminate,

The kernel version required to run each tablet is listed in this table:

https://github.com/linuxwacom/input-wacom/wiki/Device-IDs

Since CTL-472 requires 4.16 and you are running kernel 4.15, you will need to install the input-wacom kernel driver:

https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

Please let me know if this resolves your issue.

Best,
Aaron

Sodium-Aluminate, 2019-07-04:
> Hi Sodium-Aluminate,
>
> The kernel version required to run each tablet is listed in this table:
>
> https://github.com/linuxwacom/input-wacom/wiki/Device-IDs
>
> Since CTL-472 requires 4.16 and you are [...]

thanks! I will try later. :)
(my computer is not around)

Sodium-Aluminate, 2019-07-15:
Thanks a lot!
I downloaded the [latest release](https://github.com/linuxwacom/input-wacom/releases/tag/input-wacom-0.43.0), `tar xvjf ` it, `./configure && make && sudo make install`, and now it works!

closed 2019-07-15 by Sodium-Aluminate