linuxwacom/input-wacom #95: Wacom device randomly doesn't send click events

queglay, 2019-07-09:
I will intermittently observe that I cannot send click events with the wacom or the mouse, but I can type freely without issue, and the system is otherwise fine.  the problem wll last a few minutes, and then go away.

I've observed this problem last year on another machine using Kubuntu 16 and an older intuos tablet, but getting a similar problem on my current workstation RHEL 7.6 with a Wacom Intuos Pro Large PTH-860.

when I provided a sosreport to red hat, they responded that the log contained these related error messages-

```
 47.211] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Finger' [ 47.211] (**) Wacom Intuos Pro L Finger: always reports core events [ 47.262] (EE) Wacom Intuos Pro L Finger: Invalid type 'stylus' for this device. [ 47.262] (EE) Wacom Intuos Pro L Finger: Invalid type 'eraser' for this device. [ 47.262] (EE) Wacom Intuos Pro L Finger: Invalid type 'cursor' for this device. [ 47.262] (II) Wacom Intuos Pro L Finger: type not specified, assuming 'touch'. [ 47.262] (II) Wacom Intuos Pro L Finger: other types will be automatically added. [ 47.262] (**) Wacom Intuos Pro L Finger touch: panscroll is 0 [ 47.262] (**) Wacom Intuos Pro L Finger touch: panscroll modified to 13 [ 47.262] (--) Wacom Intuos Pro L Finger touch: maxX=12400 maxY=8640 maxZ=0 resX=40000 resY=40000 [ 47.262] (II) Wacom Intuos Pro L Finger touch: hotplugging dependent devices. [ 47.262] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'stylus' for this device. [ 47.262] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'eraser' for this device. [ 47.262] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'cursor' for this device. [ 47.262] (EE) Wacom Intuos Pro L Finger touch: Invalid type 'pad' for this device. [ 47.262] (II) Wacom Intuos Pro L Finger touch: hotplugging completed. 

>> Once the extended input devices such as "Wacom Intuos Pro L Finger touch" , "Wacom Intuos Pro L Pen eraser" , "Wacom Intuos Pro L Pen cursor" the wacom works fine without an error. 

[ 47.290] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Finger touch" (type: TOUCH, id 12) [ 47.290] (**) Wacom Intuos Pro L Finger touch: (accel) keeping acceleration scheme 1 [ 47.290] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration profile 0 [ 47.290] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration factor: 2.000 [ 47.290] (**) Wacom Intuos Pro L Finger touch: (accel) acceleration threshold: 4 [ 47.342] (II) config/udev: Adding input device Wacom Intuos Pro L Finger (/dev/input/mouse2) [ 47.356] (**) Wacom Intuos Pro L Pen eraser: Applying InputClass "evdev tablet catchall" [ 47.356] (**) Wacom Intuos Pro L Pen eraser: Applying InputClass "Wacom USB tablet class" [ 47.356] (**) Wacom Intuos Pro L Pen eraser: Applying InputClass "Wacom tablet class" [ 47.356] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Pen eraser' [ 47.356] (**) Wacom Intuos Pro L Pen eraser: always reports core events [ 47.356] (**) Wacom Intuos Pro L Pen eraser: panscroll is 2600 [ 47.356] (**) Wacom Intuos Pro L Pen eraser: panscroll modified to 2600 [ 47.356] (--) Wacom Intuos Pro L Pen eraser: maxX=62200 maxY=43200 maxZ=8191 resX=200000 resY=200000 tilt=enabled [ 47.363] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Pen eraser" (type: ERASER, id 14) [ 47.363] (**) Wacom Intuos Pro L Pen eraser: (accel) keeping acceleration scheme 1 [ 47.363] (**) Wacom Intuos Pro L Pen eraser: (accel) acceleration profile 0 [ 47.363] (**) Wacom Intuos Pro L Pen eraser: (accel) acceleration factor: 2.000 [ 47.363] (**) Wacom Intuos Pro L Pen eraser: (accel) acceleration threshold: 4 [ 47.363] (**) Wacom Intuos Pro L Pen cursor: Applying InputClass "evdev tablet catchall" [ 47.363] (**) Wacom Intuos Pro L Pen cursor: Applying InputClass "Wacom USB tablet class" [ 47.363] (**) Wacom Intuos Pro L Pen cursor: Applying InputClass "Wacom tablet class" [ 47.363] (II) Using input driver 'wacom' for 'Wacom Intuos Pro L Pen cursor' [ 47.363] (**) Wacom Intuos Pro L Pen cursor: always reports core events [ 47.363] (**) Wacom Intuos Pro L Pen cursor: panscroll is 2600 [ 47.363] (**) Wacom Intuos Pro L Pen cursor: panscroll modified to 2600 [ 47.372] (II) XINPUT: Adding extended input device "Wacom Intuos Pro L Pen cursor" (type: CURSOR, id 15) [ 47.372] (**) Wacom Intuos Pro L Pen cursor: (accel) keeping acceleration scheme 1 [ 47.372] (**) Wacom Intuos Pro L Pen cursor: (accel) acceleration profile 0 [ 47.372] (**) Wacom Intuos Pro L Pen cursor: (accel) acceleration factor: 2.000 [ 47.372] (**) Wacom Intuos Pro L Pen cursor: (accel) acceleration threshold: 4

 >> This is probably because the system works fine when the "Wacom Intuos Pro L" is setup completely. 

>> In the message logs i encountered the errors with OLED. Can you confirm what is the OLED device you use ? 

Jul 5 13:40:58 workstation journal: Could not set OLED icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg Jul 5 23:04:37 workstation journal: invalid unclassed pointer in cast to 'CcWacomPanel'
```

skomra, 2019-07-15:
Hi Queglay,

Could you provide a link to the bug you filed with Red Hat? Also could you run the sysinfo script and attach the result here?

https://raw.githubusercontent.com/linuxwacom/wacom-hid-descriptors/master/scripts/sysinfo.sh

Best,
Aaron

queglay, 2019-07-18:
Here is a link to the ticket number - https://access.redhat.com/support/cases/#/case/02416450
results of the script to follow...

queglay, 2019-07-18:
Here is a link the the result of the sysinfo script - https://drive.google.com/open?id=19pJJ3jWscdEaFyT0lFbhUbZXkZqHy5X3
question, does this contain any sensitive information I need to know about?

skomra, 2019-07-18:
Hi Queglay,

My opinion is that there is no personally identifiable information in the sysinfo logs, but you should be able to open them in any text reader to examine them for yourself.

Unfortunately I get a 404 when I try to see a your ticket. Perhaps you could attach a screenshot or save the whole page from your browser ("File > Save page as" in Firefox). 

The error you see in the Xorg.log above is not a true error is the "Finger" device should not have a stylus, eraser, or cursor. That is, that error is always present in the log and it is misleading. 

Since you experience this issue with both the mouse and tablet there is a good change that this is not a tablet specific issue. Is it possible to see the issue when the tablet is not connected to the system? 

Also, does the cursor still move when the clicks are disabled?

Best,
Aaron

queglay, 2019-07-19:
Its definitely not an issue when there is no tablet connected.  But when it occurs neither mouse nor tablet can send click events to say drag select text in a shell or select say multiple files on a desktop.  The pointer moves freely though, and Windows can still be dragged around.  I've repeatedly seen this problem on multiple systems over years in different countries.  Centos and rhel always though.

queglay, 2019-07-19:
https://drive.google.com/file/d/1sfzrz0SFpyDToerfZKPJoDVwyaHx5BS-/view?usp=sharing

skomra, 2019-07-19:
Thanks for the copy of the report and your comments, I understand the issue better now. 

Could you capture a sos report and run the following script when you are experiencing the issue, and then attach them here?

https://gist.githubusercontent.com/jigpu/b7b47e4bede584cec8562f666fd84af4/raw/1050d9aab9514ca0658c3cb415ccb11af5e4d6d6/capture.sh

Also you say Centos and RHEL always, but above you mention Kubuntu. If you're also experiencing the issue on KDE that would be helpful information.

queglay, 2019-07-24:
can you provide an email address to send the sosreport to?

skomra, 2019-07-24:
my email is aaron dot <my github name> at wacom dot com

queglay, 2019-07-27:
I sent the sosreport your way.

skomra, 2019-07-29:
Hi thanks for the report. Can you confirm for me that you experienced the issue during the boot in which you took the report.

Also, since I am unable to reproduce the issue myself, it will be important to get capture while you are experiencing the issue. The link for the capture script is above.

$ sudo chmod +x capture.sh 
$ sudo ./capture.sh

queglay, 2019-08-06:
apologies for my late response, I can confirm that sosreport was taken when the event occured.  

Red hat recently replied with this-
```
Most recent comment: On 2019-08-06 15:42:01, Pippin, Carroline commented:
"Hello Andrew,

On checking this sosreport , I constantly see this error about OLED icon.  (*must be sosreport-workstation-2019-07-20-gycyhcl.tar.xz (14.9 MB) )

It could be that the device doesn't support that settings Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg.

Jul  3 15:26:49 workstation journal: using plugins: altos, amt, colorhug, csr, dfu, ebitdo, nitrokey, steelseries, thunderbolt, udev, unifying, upower, wacomhid, thunderbolt_power, uefi
Jul  3 16:57:44 workstation journal: Could not set OLED icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg
Jul  4 16:14:23 workstation journal: Could not set OLED icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg
Jul  4 23:45:12 workstation journal: Could not set OLED icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg
Jul  5 00:09:41 workstation journal: Could not set OLED icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg
Jul  5 13:40:58 workstation journal: Could not set OLED icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg
Jul  5 23:04:37 workstation journal: invalid unclassed pointer in cast to 'CcWacomPanel'
Jul  6 21:46:34 workstation kernel: usb 1-6.1: Product: Wacom Intuos Pro L
Jul  6 21:46:34 workstation kernel: usb 1-6.1: Manufacturer: Wacom Co.,Ltd.

I recommend you check the OLED icon settings.
```

Though I'm not sure how to check those settings.

queglay, 2019-08-06:
I also cannot install the depends to run this script.  see below:

```

[deadlineuser@workstation Downloads]$ sudo ./capture.sh 
[sudo] password for deadlineuser: 
The following utilities are missing and must be installed to capture logs:
  * evemu-record [Try: `apt-get install evemu-tools` or `yum install evemu-record`]
[deadlineuser@workstation Downloads]$ sudo yum install evemu-record
Loaded plugins: langpacks, product-id, search-disabled-repos, subscription-manager
code                                                                                                                                                                         | 2.9 kB  00:00:00     
epel/x86_64/metalink                                                                                                                                                         | 4.7 kB  00:00:00     
epel                                                                                                                                                                         | 5.4 kB  00:00:00     
rhel-7-workstation-rpms                                                                                                                                                      | 3.5 kB  00:00:00     
rpmfusion-free-updates                                                                                                                                                       | 3.0 kB  00:00:00     
rpmfusion-nonfree-updates                                                                                                                                                    | 3.0 kB  00:00:00     
teamviewer                                                                                                                                                                   | 2.5 kB  00:00:00     
virtualbox/7Workstation/x86_64/signature                                                                                                                                     |  181 B  00:00:00     
virtualbox/7Workstation/x86_64/signature                                                                                                                                     | 2.9 kB  00:00:00 !!! 
(1/2): epel/x86_64/updateinfo                                                                                                                                                | 996 kB  00:00:00     
(2/2): epel/x86_64/primary_db                                                                                                                                                | 6.8 MB  00:00:00     
No package evemu-record available.
Error: Nothing to do
[deadlineuser@workstation Downloads]$ 

```

skomra, 2019-08-06:
That error is coming from gnome-settings-dameon's gsd-wacom-oled-helper. (https://gitlab.gnome.org/GNOME/gnome-settings-daemon/blob/master/plugins/wacom/gsd-wacom-oled-helper.c) That is a separate command with its own main(), I'm not sure why it's being run for the PTH-860 which does not have OLEDs.

The sosreport that you sent me was called "sosreport-workstation-2019-07-26-jjfvrro", the clipping from the log above is from an sosreport named "sosreport-workstation-2019-07-20-gycyhcl.tar.xz " I was not able to find the OLED error in the 7/26 report that you sent me.

--------

Maybe the package is just called evemu? (sudo yum install evemu)

https://rpmfind.net/linux/rpm2html/search.php?query=evemu

queglay, 2019-08-12:
```
sudo yum install evemu 
```
worked!  I will run capture again next time the problem presents.  for feedback it would be worth updating that notification in the capture.sh script.

I also emailed the sosreport from the redhat ticket your way Aaron.

skomra, 2019-08-13:
I don't see 

"Could not set " icon for '/dev/input/event7': Failed to open filename: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6.1/1-6.1:1.0/wacom_led/button0_rawimg"

in the sos report you sent me. Could you ask the redhat support person which specific file those messages come from?

Also reading back over the thread, you should also ask your support person how to "check the OLED icon settings" that they recommend. I'm not sure what they're referring to.

Pinglinux, 2022-08-03:
Not a Wacom kernel driver issue

closed 2022-08-03 by Pinglinux