linuxwacom/input-wacom #104: Cintiq 27QHD Pen Doesn't Work...

wwderw, 2019-08-14:
On initial booting (or restart or restarting X via ctrl-alt-backspace).  Wasn't an issue with 16.04, but it appears to be in 18.04.

It does seem to be the same issue as reported in #34 and I debated on to continue that thread or start a new one, but decided to start a new one since it is a different model.  

I was unable to get the script workaround to actually work after setting it up as a startup script (unless this shouldn't be run as a startup script, but just the first thing that I run after everything is up manually) in Neon.  I did replace "00fa" with "032a" in the script.  Was there anything else that I needed to do for my specific model in that script?

Any thoughts as to a solution or at least a temporary work around in the meantime?

jigpu, 2019-08-16:
Can you confirm that the 27 still works properly in 16.04 if you boot it up in a live environment?

If your problem is the same as #34 I would expect the edit you described to be sufficient. Its possible that it needs to be run later in the startup process, and easy to check by just manually running the script once you reach the desktop. I don't, however, expect the result to change...

Could you run our [sysinfo.sh script](https://github.com/linuxwacom/wacom-hid-descriptors/raw/master/scripts/sysinfo.sh) and attach the generated tarball to this bug? If you are able to see the device working in Ubuntu 16.04 once again, it would be good to also run sysinfo there for comparison.

wwderw, 2019-08-17:
Well, the odd thing is that it would work in Mint, but not in KDE using the same kernel (18.04).

So I went with 19.04 and it apparently doesn't recognize touch in the xsetwacom --list command.  It still does the input on screen, just doesn't recognize it in the driver (or KDE's gui frontend).  All I get now is pad, stylus, and eraser.  It still has the same startup quirk, but I managed to come up with an unbind/bind script that replaced having to physically finagle with the cable every time.

jigpu, 2019-08-19:
That _is_ really weird... I guess there could be something in userspace that is causing a problem. I wonder if e.g. fwupd could be to blame?

The missing touch device is a quirk specific to Ubuntu and its derivatives. See https://bugs.launchpad.net/ubuntu/+source/xf86-input-wacom/+bug/1774242. It can be worked around if you need to configure the touchscreen through xsetwacom or KDE's GUI.

Would you mind sharing your unbind/bind script with others if its notably different from the version for the 24?

wwderw, 2019-08-19:
> The missing touch device is a quirk specific to Ubuntu and its derivatives. See https://bugs.launchpad.net/ubuntu/+source/xf86-input-wacom/+bug/1774242. It can be worked around if you need to configure the touchscreen through xsetwacom or KDE's GUI.

That did it for the touch.  Just made those 2 sections live again and then ran my bash script and it worked like a charm, like it did before.

> Would you mind sharing your unbind/bind script with others if its notably different from the version for the 24?

May not be the best thing, but here is what I managed to get working:

`port="3-10.3" # as shown by lsusb -t: {bus}-{port}(.{subport})

bind_usb() {
  echo "$1" >/sys/bus/usb/drivers/usb/bind
}

unbind_usb() {
  echo "$1" >/sys/bus/usb/drivers/usb/unbind
}

unbind_usb "$port"
sleep 1 # enable delay here
bind_usb "$port"`

jigpu, 2019-09-03:
Thanks for the script! I'll have to give that a shot on my 24HD if/when it stops working again to see if your bind/unbind can work for both devices.

wwderw, 2019-09-04:
It should be device agnostic.  If you wanted to use it for any USB device, it should work.  The only thing to watch out for is bus port subport.  That may change (for a variety of reasons).

Since this seems to always happen at reboot, I set it up as a cron job and then just run my normal startup scripts as usual.

I did look closer at Mint and Kubuntu (at least as of 18.04.3 with regard to Kubuntu, I don't think Mint 19.2 is to that point release, maybe in December when 19.3 comes out) and it appears that Mint is running the 4.15 kernel and the aforementioned Kubuntu is running 5.0.0, so while there is still the same rough base of 18.04, there is still some differences there.  Wonder if an update in the kernel is causing this?

jigpu, 2019-11-11:
Just got [another report](https://twitter.com/artredj/status/1193511701888589825) of this issue affecting a 27QHD user with a 5.x kernel. I've also been able to confirm that using the bind/unbind method on the USB port also works for my 24HDT.

The below script should save people from having to figure out the port name -- just edit the device name and it should do the rest.

~~~
#!/bin/sh

DEVICE="Wacom Cintiq 27QHD touch Pen stylus"

#### DO NOT EDIT BELOW THIS LINE ####

get_port() {
  EVENT_NODE=$(xinput list-props "$1" | grep -Eo '/dev/input/[^"]+')
  EVENT_SYSFS=$(udevadm info -q path -n "${EVENT_NODE}")
  PORT_SYSFS=$(udevadm info -q path -p "${EVENT_SYSFS}/../../../../../")
  basename "${PORT_SYSFS}"
}

bind_usb() {
  echo "Binding port $1..."
  echo "$1" >/sys/bus/usb/drivers/usb/bind
}

unbind_usb() {
  echo "Unbinding port $1..."
  echo "$1" >/sys/bus/usb/drivers/usb/unbind
}

if test "$(id -u)" -ne 0; then
  echo "This command must be run as root."
  exit 1
fi

PORT=$(get_port "${DEVICE}")
unbind_usb "${PORT}" && sleep 1 && bind_usb "${PORT}"
~~~

jigpu, 2019-11-14:
Ran `git bisect` on a Fedora 27 machine and got e04a0442d33b8cf183bba38646447b891bb02123 as a commit of interest.

~~~
$ git bisect log
git bisect start
# good: [569dbb88e80deb68974ef6fdd6a13edb9d686261] Linux 4.13
git bisect good 569dbb88e80deb68974ef6fdd6a13edb9d686261
# good: [bebc6082da0a9f5d47a1ea2edc099bf671058bd4] Linux 4.14
git bisect good bebc6082da0a9f5d47a1ea2edc099bf671058bd4
# good: [d8a5b80568a9cb66810e75b182018e9edb68e8ff] Linux 4.15
git bisect good d8a5b80568a9cb66810e75b182018e9edb68e8ff
# bad: [0adb32858b0bddf4ada5f364a84ed60b196dbcda] Linux 4.16
git bisect bad 0adb32858b0bddf4ada5f364a84ed60b196dbcda
# bad: [c14376de3a1befa70d9811ca2872d47367b48767] printk: Wake klogd when passing console_lock owner
git bisect bad c14376de3a1befa70d9811ca2872d47367b48767
# bad: [a103950e0dd2058df5e8a8d4a915707bdcf205f0] Merge branch 'linus' of git://git.kernel.org/pub/scm/linux/kernel/git/herbert/crypto-2.6
git bisect bad a103950e0dd2058df5e8a8d4a915707bdcf205f0
# good: [d8b91dde38f4c43bd0bbbf17a90f735b16aaff2c] Merge branch 'perf-core-for-linus' of git://git.kernel.org/pub/scm/linux/kernel/git/tip/tip
git bisect good d8b91dde38f4c43bd0bbbf17a90f735b16aaff2c
# good: [28bc6fb9596fe1e577d09fc17ee6e1bb051c6ba3] Merge tag 'scsi-misc' of git://git.kernel.org/pub/scm/linux/kernel/git/jejb/scsi
git bisect good 28bc6fb9596fe1e577d09fc17ee6e1bb051c6ba3
# good: [50081e437872e68300750068754f21d0faac5d86] Merge tag 'leds_for_4.16-rc1' of git://git.kernel.org/pub/scm/linux/kernel/git/j.anaszewski/linux-leds
git bisect good 50081e437872e68300750068754f21d0faac5d86
# bad: [3c29548f87f9545f2f3c1cd1a784fae8ad2d53ba] Merge branch 'next-integrity' of git://git.kernel.org/pub/scm/linux/kernel/git/jmorris/linux-security
git bisect bad 3c29548f87f9545f2f3c1cd1a784fae8ad2d53ba
# bad: [3c29548f87f9545f2f3c1cd1a784fae8ad2d53ba] Merge branch 'next-integrity' of git://git.kernel.org/pub/scm/linux/kernel/git/jmorris/linux-security
git bisect bad 3c29548f87f9545f2f3c1cd1a784fae8ad2d53ba
# good: [2046362cf11d3dcb0c6904f7f3209c568d7858b3] gpio: Fix a documentation spelling mistake
git bisect good 2046362cf11d3dcb0c6904f7f3209c568d7858b3
# good: [2046362cf11d3dcb0c6904f7f3209c568d7858b3] gpio: Fix a documentation spelling mistake
git bisect good 2046362cf11d3dcb0c6904f7f3209c568d7858b3
# good: [4c6790c49a8aa044b7b4c275ed556ef512866092] Merge tag 'for-linus-4.16-1' of git://github.com/cminyard/linux-ipmi
git bisect good 4c6790c49a8aa044b7b4c275ed556ef512866092
# good: [4c6790c49a8aa044b7b4c275ed556ef512866092] Merge tag 'for-linus-4.16-1' of git://github.com/cminyard/linux-ipmi
git bisect good 4c6790c49a8aa044b7b4c275ed556ef512866092
# bad: [a44f3ec625002a3631f95ea317a9c25f70578fe7] Merge branch 'for-4.16/elo' into for-linus
git bisect bad a44f3ec625002a3631f95ea317a9c25f70578fe7
# bad: [00720277a517e6dcc4773fb413711fe0131ee9bd] HID: hid-multitouch: support fine-grain orientation reporting
git bisect bad 00720277a517e6dcc4773fb413711fe0131ee9bd
# bad: [d5158e020c45936d0ea5a1ce1e49a45da6e76f44] HID: Ignore Jabra HID interface based on firmware version
git bisect bad d5158e020c45936d0ea5a1ce1e49a45da6e76f44
# good: [6e65d9d5492f370dd0e5418bdd38265b2ca74f88] HID: quirks: move the list of special devices into a quirk
git bisect good 6e65d9d5492f370dd0e5418bdd38265b2ca74f88
# bad: [e04a0442d33b8cf183bba38646447b891bb02123] HID: core: remove the absolute need of hid_have_special_driver[]
git bisect bad e04a0442d33b8cf183bba38646447b891bb02123
# good: [f745d162f469a4b1e805779a8b0d9157100c813c] HID: core: move the list of ignored devices in hid-quirks.c
git bisect good f745d162f469a4b1e805779a8b0d9157100c813c
# first bad commit: [e04a0442d33b8cf183bba38646447b891bb02123] HID: core: remove the absolute need of hid_have_special_driver[]
~~~

jigpu, 2019-11-14:
Also: I'd noticed early on in debugging that running `rmmod wacom` and then `modprobe wacom` would cause the device to disconnect (as expected) but never come back until I either disconnected and reconnected the tablet or ran the fix script. Even with a "working" version of the kernel (e.g. stock Fedora 27) this seems to be the case.

Given that e04a0442d33b8cf183bba38646447b891bb02123 may be trying to unbind and rebind the device, perhaps there's more to the problem than *just* this commit may indicate...

bentiss, 2019-11-15:
> Also: I'd noticed early on in debugging that running rmmod wacom and then modprobe wacom would cause the device to disconnect (as expected) but never come back until I either disconnected and reconnected the tablet or ran the fix script.

this rings a bell, but I think it was related to fwupd.

Could you provide kernel logs when you do the operation and when the device is not coming back?
The unbind should be transparent for the tablet (we should just be opening/closing the usb endpoints), but the rebind should initiate the tablet mode switch, which might not be accepted by the firmware.

Anyway, logs on an 5.3 kernel would be appreciated.

jigpu, 2019-11-15:
Here's the logs from that same Fedora 27 system with the upstream 5.3 kernel installed (I can, of course, gather from a stock Fedora 31 if you would prefer). I don't see anything suspcicious, and haven't yet found anything to explain the issue. I tried disabling fwupd but did not observe a difference.

I would have expected to see more of a difference in the logs at step 2 and step 4 below, but they're quite similar.

### Log steps:
1. Boot system
2. Plug in Cintiq
    1. First gets bound to hid-generic.
    2. Next gets re-bound to wacom. Doesn't work!
3. Unplug Cintiq
4. Plug in Cintiq a 2nd time
    1. Binds immediately to wacom. Start seeing USB events from the tablet. Works!
5. Run `rmmod wacom`
    1. Re-binds to hid-generic. Doesn't work, as expected.
6. Run `modprobe wacom`
    1. Re-binds to wacom. Doesn't work!

### Log
[dmesg-annotated.log](https://github.com/linuxwacom/input-wacom/files/3852984/dmesg-annotated.log)

jigpu, 2020-04-27:
@bentiss any thoughts? Same issue seems to affect the Cintiq 22 (https://github.com/linuxwacom/input-wacom/issues/115) and Cintiq 24 (https://github.com/linuxwacom/input-wacom/issues/34)

n3xxt, 2020-05-21:
Hello,

Thank you for supporting Wacom for Linux!

Worked great in Debian Jessie 8*, Stretch 9*, not working since Buster 10*.

# MAIN ISSUE

Now trying in MX Linux 19.1 xfce [based on Debian Buster 10, systemd-free] and it doesn't work (Pen Pro stylus/eraser doesn't produce any events).

As requested: [sysinfo.atpZghcO76.tar.gz](https://github.com/linuxwacom/input-wacom/files/4662914/sysinfo.atpZghcO76.tar.gz) 

Commented out  `systemd` in **/lib/udev/rules.d/69-wacom.rules**

```
ACTION!="add|change", GOTO="wacom_end"

# Match all serial wacom tablets with a serial ID starting with WACf
# Notes: We assign NAME though we shouldn't, but currently the server requires it
#        We assign the lot to subsystem pnp too because server reads NAME from
#        the parent device. Once all that's fixed, as simple SUBSYSTEM="tty"
#        will do and the ENV{NAME} can be removed.
SUBSYSTEM=="tty|pnp", SUBSYSTEMS=="pnp", ATTRS{id}=="WACf*", ENV{ID_MODEL}="Serial Wacom Tablet $attr{id}", ENV{ID_INPUT}="1", ENV{ID_INPUT_TABLET}="1", ENV{NAME}="Serial Wacom Tablet $attr{id}"
SUBSYSTEM=="tty|pnp", SUBSYSTEMS=="pnp", ATTRS{id}=="FUJ*", ENV{ID_MODEL}="Serial Wacom Tablet $attr{id}", ENV{ID_INPUT}="1", ENV{ID_INPUT_TABLET}="1", ENV{NAME}="Serial Wacom Tablet $attr{id}"

# comment out the next line if your distribution does not provide systemd
# If a /dev/ttySx device with the WACf name is detected, start the
# wacom-inputattach service with the kernel name as parameter
#SUBSYSTEM=="tty|pnp", KERNEL=="ttyS[0-9]*", ATTRS{id}=="WACf*", TAG+="systemd", ENV{SYSTEMD_WANTS}+="wacom-inputattach@%k.service"

LABEL="wacom_end"
```
tried listening to events streams

`sudo cat /dev/input/mouse0`

...nothing...

`sudo cat /dev/input/event7`

...nothing...

**Wacom Pad** circle scroller **works**.

# HACK01

After boot, login - **switch off 27QHD, wait 5s, switch on** - pen/eraser events are being read.

This script allows to reset USB without physically turning off the device (thanks @mnlipp https://github.com/linuxwacom/input-wacom/issues/34).

DISCLAIMER: it's a "hack", not a solution. (c)

```
#!/bin/bash

VENDOR_ID="056a"works
DISPLAY_ID="DisplayPort-1"

for d in /sys/bus/usb/devices/*; do 
  if [ -r "$d/idVendor" ]; then
    idVendor=`cat $d/idVendor` 
    idProduct=`cat $d/idProduct` 

    if [ "$idVendor" = $VENDOR_ID ]; then
        echo 0 > "$d/authorized"
        echo "$d $idVendor:$idProduct turned OFF/n"
        echo 1 > "$d/authorized"
        echo "$d $idVendor:$idProduct turned ON/n"

    fi
[... 390 lines cut ...]
[   457.043] (II) AMDGPU(0): Modeline "1280x720"x0.0   74.25  1280 1390 1430 1650  720 725 730 750 +hsync +vsync (45.0 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "720x480"x0.0   27.00  720 736 798 858  480 489 495 525 -hsync -vsync (31.5 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "1152x864"x60.0   81.62  1152 1216 1336 1520  864 865 868 895 -hsync +vsync (53.7 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "1280x1024"x0.0  108.00  1280 1328 1440 1688  1024 1025 1028 1066 +hsync +vsync (64.0 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "1600x900"x60.0  119.00  1600 1696 1864 2128  900 901 904 932 -hsync +vsync (55.9 kHz e)
[   457.043] (II) AMDGPU(0): Modeline "1280x800"x0.0   83.50  1280 1352 1480 1680  800 803 809 831 -hsync +vsync (49.7 kHz e)
```

Seems like smth is up with "evdev" module, as this Xorg event "[ 456.340] (II) UnloadModule: "evdev"" triggers all these errors.

Thank you for paying attention.

Please let me know if you may need me to perform more testing.

# similar

https://github.com/linuxwacom/input-wacom/issues/34
https://github.com/linuxwacom/input-wacom/issues/115

MX Linux community discussion

https://forum.mxlinux.org/viewtopic.php?p=576078#p576078

n3xxt, 2020-05-21:
PS JFYI MX Linux 19.1 is **not** WayLand-based as Debian Buster is.

jigpu, 2020-05-21:
Thanks for the sysinfo @n3xxt. The more puzzle pieces for understanding this mystery the better :)

> Commented out `systemd` in **/lib/udev/rules.d/69-wacom.rules**

Not necessary in your particular case, since the Cintiq 27 doesn't run these particular rules (they're only used by old sensors that use a serial port connection). As an aside, I also have to wonder if commenting out the line is actually necessary on distros without systemd (it looks like it just sets up an additional tag and environment variable), but I don't have the necessary hardware to check... Still, no harm in doing what it says :+1: 

> tried listening to events streams
> 
> `sudo cat /dev/input/mouse0`
> 
> ...nothing...
> 
> `sudo cat /dev/input/event7`
> 
> ...nothing...
>
> **Wacom Pad** circle scroller **works**.

The lack of input events matches up with what others experience. I don't think I've heard anyone comment on wether the various pad controls work, however. The ExpressKey Remote that you're using is a separate USB device, so it working doesn't say too much, but would seem to further point at this issue being specific to the EMR device. I'll have to check if the pad on my 24HD still works the next time it bugs out since its events are reported through the EMR device...

jigpu, 2020-05-21:
@n3xxt To keep this issue focused, would you mind opening a new bug for "Issue 2" with a link back here for reference? You can remove the issue from your post once that's been done.

As for the "Other Errors" you mentioned, you can file a bug for those too if you want more information, but they're either expected (the "not supported by xorg.conf" and "does not exist on device" warnings) or a result of MX Linux 19.1 having an out-of-date version of the driver (the "69984 lines of errors" was reported in [SourceForge bug 337](https://sourceforge.net/p/linuxwacom/bugs/337/) and fixed in xf86-input-wacom 0.36.0). As with the second issue, please go ahead and also remove them from your reply since they're unfortunately not relevant...

n3xxt, 2020-05-22:
Hey @jigpu thanks for response!

>To keep this issue focused, would you mind opening a new bug for "Issue 2" with a link back here for reference? You can remove the issue from your post once that's been done.

Done!

https://github.com/linuxwacom/input-wacom/issues/186

>...or a result of MX Linux 19.1 having an out-of-date version of the driver (the "69984 lines of errors" was reported in SourceForge bug 337 and fixed in xf86-input-wacom 0.36.0)

So I have created this script (was saddened you don't have one on your repos)

```
#!/bin/bash

INPUT_WACOM="https://github.com/linuxwacom/input-wacom.git"
INPUT_WACOM_V="0.46.0"
XF86_INPUT_WACOM="https://github.com/linuxwacom/xf86-input-wacom.git"
XF86_INPUT_WACOM_V="0.39.0"
LIBWACOM="https://github.com/linuxwacom/libwacom.git"

mkdir wacom
cd wacom

echo "\n\n######################### input-wacom\n\n\n"
# https://github.com/linuxwacom/input-wacom/wiki/Installing-input-wacom-from-source

sudo apt-get install -y build-essential autoconf linux-headers-$(uname -r)

git clone --branch "input-wacom-${INPUT_WACOM_V}" $INPUT_WACOM

cd input-wacom

if test -x ./autogen.sh; then ./autogen.sh; else ./configure; fi && make && sudo make install || echo "Build Failed"

cd ..

echo "\n\n######################### xf86-input-wacom\n\n\n"

# https://github.com/linuxwacom/xf86-input-wacom/wiki/Building-The-Driver
sudo sh -c "apt-get update && apt-get install xserver-xorg-input-wacom$(dpkg -S $(which Xorg) | grep -Eo -- "-hwe-[^:]*")"

# https://github.com/linuxwacom/xf86-input-wacom/wiki/Dependencies
sudo apt-get install -y xserver-xorg-dev xutils-dev libxi-dev libxrandr-dev libxext-dev libx11-dev libxinerama-dev libudev-dev

sudo apt-get install -y autoconf pkg-config make xutils-dev libtool xserver-xorg-dev$(dpkg -S $(which Xorg) | grep -Eo -- "-hwe-[^:]*") libx11-dev libxi-dev libxrandr-dev libxinerama-dev libudev-dev

git clone --branch "xf86-input-wacom-${XF86_INPUT_WACOM_V}" $XF86_INPUT_WACOM

cd xf86-input-wacom

set -- --prefix="/usr" --libdir="$(readlink -e $(ls -d /usr/lib*/xorg/modules/input/../../../ | head -n1))"
if test -x ./autogen.sh; then ./autogen.sh "$@"; else ./configure "$@"; fi && make && sudo make install || echo "Build Failed"

cd ..

echo "\n\n######################### libwacom\n\n\n"
# https://github.com/linuxwacom/libwacom/wiki

sudo apt-get install -y build-essential libgudev-1.0-dev libxml++2.6-dev

git clone $LIBWACOM

cd libwacom

./autogen.sh --prefix=/usr --libdir=/lib64
make
sudo make install

cd ..
[... 74 lines cut ...]
$ `xsetwacom --list devices`

```
Wacom Cintiq 27QHD Pen stylus   	id: 14	type: STYLUS    
Wacom Cintiq 27QHD Pen eraser   	id: 15	type: ERASER    
Wacom Express Key Remote Pad pad	id: 21	type: PAD   
```

$ `xsetwacom get 14 all`
$ `xsetwacom get 15 all`
$ `xsetwacom get 15 all`

```
**...same results skipped...**
```

>As for the "Other Errors" you mentioned, you can file a bug for those too if you want more information, but they're either expected (the "not supported by xorg.conf" and "does not exist on device" warnings)

Yes, I would appreciate more details on that one - Thanks!

Created https://github.com/linuxwacom/input-wacom/issues/187

The mouse pointer is still jumping (ISSUE 02).

Thanks!

n3xxt, 2020-05-26:
# Same issue on Debian 10.4

[sysinfo.CSUk49KvBs.tar.gz](https://github.com/linuxwacom/input-wacom/files/4683034/sysinfo.CSUk49KvBs.tar.gz)

Hack01 - doesn't work;
Cursor doesn't jump after switching Off/On the Wacom device

$ `uname -a`
```
Linux x 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2 (2020-04-29) x86_64 GNU/Linux
```
$ `cat /etc/os-release`
```
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

$ `cat /etc/debian_version `
`10.4`

wwderw, 2020-05-26:
n3xxt did you try my original script?  I have yet to have that one not work, but I really only use Ubuntu based distros and Manjaro just in case Debian is still significantly different from what Canonical did with their offshoot of it).  Only downside is that if you move the connection to a different port, have to adjust accordingly.

n3xxt, 2020-05-26:
> n3xxt did you try my original script? I have yet to have that one not work, but I really only use Ubuntu based distros and Manjaro just in case Debian is still significantly different from what [...]

Hey @wwderw ! Thanks for response, yes your script 

https://github.com/linuxwacom/input-wacom/issues/104#issuecomment-522722965

```
#!/bin/bash

port="5-3.3" # as shown by lsusb -t: {bus}-{port}(.{subport})

bind_usb() {
  echo "$1" >/sys/bus/usb/drivers/usb/bind
}

unbind_usb() {
  echo "$1" >/sys/bus/usb/drivers/usb/unbind
}

unbind_usb "$port"
sleep 1 # enable delay here
bind_usb "$port"
```

as well as a lazier version by @jigpu 

https://github.com/linuxwacom/input-wacom/issues/104#issuecomment-552527880

```
#!/bin/sh

DEVICE="Wacom Cintiq 27QHD Pen stylus"

#### DO NOT EDIT BELOW THIS LINE ####

get_port() {
  EVENT_NODE=$(xinput list-props "$1" | grep -Eo '/dev/input/[^"]+')
  EVENT_SYSFS=$(udevadm info -q path -n "${EVENT_NODE}")
  PORT_SYSFS=$(udevadm info -q path -p "${EVENT_SYSFS}/../../../../../")
  basename "${PORT_SYSFS}"
}

bind_usb() {
  echo "Binding port $1..."
  echo "$1" >/sys/bus/usb/drivers/usb/bind
}

unbind_usb() {
  echo "Unbinding port $1..."
  echo "$1" >/sys/bus/usb/drivers/usb/unbind
}

if test "$(id -u)" -ne 0; then
  echo "This command must be run as root."
  exit 1
fi

PORT=$(get_port "${DEVICE}")
unbind_usb "${PORT}" && sleep 1 && bind_usb "${PORT}"

```

worked as expected!

Just need to install xinput in Debian first

`sudo apt-get install -y xinput`

Thanks for taking care!

n3xxt, 2020-05-26:
Actually "my" script also worked, just tripple checked now - the reason it didn't work because of a typo - it had this line which I have corrected now

'VENDOR_ID="056a"works'

```
#!/bin/bash

VENDOR_ID="056a"
DISPLAY_ID="DisplayPort-1"

for d in /sys/bus/usb/devices/*; do 
  if [ -r "$d/idVendor" ]; then
    idVendor=`cat $d/idVendor` 
    idProduct=`cat $d/idProduct` 

    if [ "$idVendor" = $VENDOR_ID ]; then
        echo 0 > "$d/authorized"
        echo "$d $idVendor:$idProduct turned OFF/n"
        echo 1 > "$d/authorized"
        echo "$d $idVendor:$idProduct turned ON/n"

    fi
  fi
done

echo "waiting.../n"
sleep 5s

for x in $(xsetwacom --list devices | grep -o -P "id: \d+" | cut -f2 -d' '); do
    xsetwacom set $x MapToOutput $DISPLAY_ID
    echo "$x ok/n"
done

```

n3xxt, 2020-05-26:
It's great to have multiple approaches to the same "solution"

Pinglinux, 2022-08-10:
This sounds like a kernel 5.3 USB subsystem regression issue. Did you try an newer kernel? Do you still see the issue?

Pinglinux, 2022-08-12:
The root cause of those Cintiq related issues are not in Wacom driver, especially since the issue disappear after a power/USB cycle. It should be in the kernel/system configuration for some kernels. Feel free to open a new request if systems running kernel 5.18+ still have the issue.

closed 2022-08-12 by Pinglinux

JoernLoviscach, 2022-08-19:
I can indeed confirm the issue with a kernel as recent as 5.19, running openSUSE Tumbleweed. The hack by @n3xxt still works.

wwderw, 2022-08-19:
It still happens on Arch, I still just run the little script that I came up with and that still seems to work.  

I don't necessarily know exactly what you man when  you say "disappear after a power/USB cycle".   

When one does a traditional power cycle (or atleast what I consider traditional, as in full power cycle of the entire computer) the issue shows itself again.  When one manually disconnects and reconnects the USB port that has the Wacom connection, without power cycling the entire computer, that fixes the issue.  Once one power cycles the entire computer though, it reverts and have to disconnect and reconnect the USB port that the Wacom device uses.

Pinglinux, 2022-08-19:
By "power/USB cycle", I meant "power cycle the tablet" or "disconnect/reconnect the tablet through USB", not "power cycle the system". So, it seems not a kernel issue either. Since Cintiq Pros, which technology is newer than Cintiqs, don't have the issue, I suspect it is a hardware/technology limitation for the Cintiqs.

wwderw, 2022-08-19:
Power cycling the tablet doesn't do anything either.

No issue on a 21UX Cintiq that I have running the same version of Manjaro and the 21UX is older compared to the QHD.  But it could be a limitation specific to the QHD model though.

Pinglinux, 2022-08-19:
@wwderw Thanks for your correction and clarification!